f4pga/prjxray #953: Enhance IDELAY/ISERDES/OSERDES fuzzer to solve missing bits

tmichalak, 2019-07-22:
FASM for Litex is reporting following missing bits:
<pre>

# In frame 0x004015a2 49 bits were not converted.
--
{ unknown_bit = "004015a2_85_30", unknown_segment = "0x00401580", unknown_segbit = "34_2750" }
{ unknown_bit = "004015a2_93_30", unknown_segment = "0x00401580", unknown_segbit = "34_3006" }
{ unknown_bit = "004015a2_86_12", unknown_segment = "0x00401580", unknown_segbit = "34_2764" }
{ unknown_bit = "004015a2_77_8", unknown_segment = "0x00401580", unknown_segbit = "34_2472" }
{ unknown_bit = "004015a2_57_30", unknown_segment = "0x00401580", unknown_segbit = "34_1854" }
{ unknown_bit = "004015a2_61_30", unknown_segment = "0x00401580", unknown_segbit = "34_1982" }
{ unknown_bit = "004015a2_53_30", unknown_segment = "0x00401580", unknown_segbit = "34_1726" }
{ unknown_bit = "004015a2_69_30", unknown_segment = "0x00401580", unknown_segbit = "34_2238" }
{ unknown_bit = "004015a2_77_30", unknown_segment = "0x00401580", unknown_segbit = "34_2494" }
{ unknown_bit = "004015a2_81_30", unknown_segment = "0x00401580", unknown_segbit = "34_2622" }
{ unknown_bit = "004015a2_54_4", unknown_segment = "0x00401580", unknown_segbit = "34_1732" }
{ unknown_bit = "004015a2_58_4", unknown_segment = "0x00401580", unknown_segbit = "34_1860" }
{ unknown_bit = "004015a2_62_4", unknown_segment = "0x00401580", unknown_segbit = "34_1988" }
{ unknown_bit = "004015a2_70_4", unknown_segment = "0x00401580", unknown_segbit = "34_2244" }
{ unknown_bit = "004015a2_78_4", unknown_segment = "0x00401580", unknown_segbit = "34_2500" }
{ unknown_bit = "004015a2_82_4", unknown_segment = "0x00401580", unknown_segbit = "34_2628" }
{ unknown_bit = "004015a2_86_4", unknown_segment = "0x00401580", unknown_segbit = "34_2756" }
{ unknown_bit = "004015a2_94_4", unknown_segment = "0x00401580", unknown_segbit = "34_3012" }
{ unknown_bit = "004015a2_54_12", unknown_segment = "0x00401580", unknown_segbit = "34_1740" }
{ unknown_bit = "004015a2_58_12", unknown_segment = "0x00401580", unknown_segbit = "34_1868" }
{ unknown_bit = "004015a2_62_12", unknown_segment = "0x00401580", unknown_segbit = "34_1996" }
{ unknown_bit = "004015a2_70_12", unknown_segment = "0x00401580", unknown_segbit = "34_2252" }
{ unknown_bit = "004015a2_78_12", unknown_segment = "0x00401580", unknown_segbit = "34_2508" }
{ unknown_bit = "004015a2_82_12", unknown_segment = "0x00401580", unknown_segbit = "34_2636" }
{ unknown_bit = "004015a2_54_18", unknown_segment = "0x00401580", unknown_segbit = "34_1746" }
{ unknown_bit = "004015a2_58_18", unknown_segment = "0x00401580", unknown_segbit = "34_1874" }
{ unknown_bit = "004015a2_62_18", unknown_segment = "0x00401580", unknown_segbit = "34_2002" }
{ unknown_bit = "004015a2_70_18", unknown_segment = "0x00401580", unknown_segbit = "34_2258" }
{ unknown_bit = "004015a2_78_18", unknown_segment = "0x00401580", unknown_segbit = "34_2514" }
{ unknown_bit = "004015a2_82_18", unknown_segment = "0x00401580", unknown_segbit = "34_2642" }
{ unknown_bit = "004015a2_58_24", unknown_segment = "0x00401580", unknown_segbit = "34_1880" }
{ unknown_bit = "004015a2_62_24", unknown_segment = "0x00401580", unknown_segbit = "34_2008" }
{ unknown_bit = "004015a2_54_24", unknown_segment = "0x00401580", unknown_segbit = "34_1752" }
{ unknown_bit = "004015a2_70_24", unknown_segment = "0x00401580", unknown_segbit = "34_2264" }
{ unknown_bit = "004015a2_78_24", unknown_segment = "0x00401580", unknown_segbit = "34_2520" }
{ unknown_bit = "004015a2_82_24", unknown_segment = "0x00401580", unknown_segbit = "34_2648" }
{ unknown_bit = "004015a2_86_24", unknown_segment = "0x00401580", unknown_segbit = "34_2776" }
{ unknown_bit = "004015a2_94_12", unknown_segment = "0x00401580", unknown_segbit = "34_3020" }
{ unknown_bit = "004015a2_94_24", unknown_segment = "0x00401580", unknown_segbit = "34_3032" }
{ unknown_bit = "004015a2_55_8", unknown_segment = "0x00401580", unknown_segbit = "34_1768" }
{ unknown_bit = "004015a2_59_8", unknown_segment = "0x00401580", unknown_segbit = "34_1896" }
{ unknown_bit = "004015a2_67_8", unknown_segment = "0x00401580", unknown_segbit = "34_2152" }
{ unknown_bit = "004015a2_71_8", unknown_segment = "0x00401580", unknown_segbit = "34_2280" }
{ unknown_bit = "004015a2_79_8", unknown_segment = "0x00401580", unknown_segbit = "34_2536" }
{ unknown_bit = "004015a2_83_8", unknown_segment = "0x00401580", unknown_segbit = "34_2664" }
{ unknown_bit = "004015a2_91_8", unknown_segment = "0x00401580", unknown_segbit = "34_2920" }
{ unknown_bit = "004015a2_95_8", unknown_segment = "0x00401580", unknown_segbit = "34_3048" }
{ unknown_bit = "004015a2_86_18", unknown_segment = "0x00401580", unknown_segbit = "34_2770" }
{ unknown_bit = "004015a2_94_18", unknown_segment = "0x00401580", unknown_segbit = "34_3026" }

</pre>

The bits are pretty close to the bits we have solved for IDELAY. In such case the solution would be:
<pre>
LIOI3.IDELAY_Y1.34_8 34_8
LIOI3.IDELAY_Y1.34_94 34_94
LIOI3.IDELAY_Y1.34_100 34_100
LIOI3.IDELAY_Y1.34_108 34_108
LIOI3.IDELAY_Y1.34_114 34_114
LIOI3.IDELAY_Y1.34_120 34_120
LIOI3.IDELAY_Y1.35_7 35_7
LIOI3.IDELAY_Y1.35_13 35_13
LIOI3.IDELAY_Y1.35_19 35_19
LIOI3.IDELAY_Y1.35_27 35_27
LIOI3.IDELAY_Y1.35_119 35_119
</pre>

mkurc-ant, 2019-07-22:
From my findings bits related to ISERDES are in frames 26-29 with the exception of of 31_67 and 31_77. I'll try to chaining two ISERDES, maybe some additional bits will be there.

tmichalak, 2019-07-24:
Solved with #957

closed 2019-07-26 by litghost