f4pga/prjxray #955: Bits for 401580 base address and frame 26

tmichalak, 2019-07-22:
24 bits reported missing for Litex
<pre>
# In frame 0x0040159a 24 bits were not converted.
{ unknown_bit = "0040159a_68_25", unknown_segment = "0x00401580", unknown_segbit = "26_2201" }
{ unknown_bit = "0040159a_56_25", unknown_segment = "0x00401580", unknown_segbit = "26_1817" }
{ unknown_bit = "0040159a_60_25", unknown_segment = "0x00401580", unknown_segbit = "26_1945" }
{ unknown_bit = "0040159a_59_25", unknown_segment = "0x00401580", unknown_segbit = "26_1913" }
{ unknown_bit = "0040159a_72_25", unknown_segment = "0x00401580", unknown_segbit = "26_2329" }
{ unknown_bit = "0040159a_80_25", unknown_segment = "0x00401580", unknown_segbit = "26_2585" }
{ unknown_bit = "0040159a_84_25", unknown_segment = "0x00401580", unknown_segbit = "26_2713" }
{ unknown_bit = "0040159a_92_25", unknown_segment = "0x00401580", unknown_segbit = "26_2969" }
{ unknown_bit = "0040159a_96_25", unknown_segment = "0x00401580", unknown_segbit = "26_3097" }
{ unknown_bit = "0040159a_67_25", unknown_segment = "0x00401580", unknown_segbit = "26_2169" }
{ unknown_bit = "0040159a_71_25", unknown_segment = "0x00401580", unknown_segbit = "26_2297" }
{ unknown_bit = "0040159a_79_25", unknown_segment = "0x00401580", unknown_segbit = "26_2553" }
{ unknown_bit = "0040159a_83_25", unknown_segment = "0x00401580", unknown_segbit = "26_2681" }
{ unknown_bit = "0040159a_91_25", unknown_segment = "0x00401580", unknown_segbit = "26_2937" }
{ unknown_bit = "0040159a_95_25", unknown_segment = "0x00401580", unknown_segbit = "26_3065" }
{ unknown_bit = "0040159a_54_13", unknown_segment = "0x00401580", unknown_segbit = "26_1741" }
{ unknown_bit = "0040159a_58_13", unknown_segment = "0x00401580", unknown_segbit = "26_1869" }
{ unknown_bit = "0040159a_62_13", unknown_segment = "0x00401580", unknown_segbit = "26_1997" }
{ unknown_bit = "0040159a_70_13", unknown_segment = "0x00401580", unknown_segbit = "26_2253" }
{ unknown_bit = "0040159a_78_13", unknown_segment = "0x00401580", unknown_segbit = "26_2509" }
{ unknown_bit = "0040159a_82_13", unknown_segment = "0x00401580", unknown_segbit = "26_2637" }
{ unknown_bit = "0040159a_86_13", unknown_segment = "0x00401580", unknown_segbit = "26_2765" }
{ unknown_bit = "0040159a_94_13", unknown_segment = "0x00401580", unknown_segbit = "26_3021" }
{ unknown_bit = "0040159a_55_25", unknown_segment = "0x00401580", unknown_segbit = "26_1785" }
</pre>

are solved by 
<pre>
RIOB33.IOB_Y1.26_25 26_25
RIOB33.IOB_Y1.26_57 26_57
RIOB33.IOB_Y1.26_107 26_107
RIOB33.IOB_Y1.26_109 26_109
</pre>

They are very close to ISERDES bits.

tmichalak, 2019-07-23:
`26_25` and `26_57` are part of the `ISERDES.IN_USE` feature: `RIOB33.IOB_Y1.ISERDES.IN_USE 26_17 26_25 26_57 29_17 29_44 30_44`.
It seems that our fuzzer solves too many bits for it.

litghost, 2019-07-23:
> `26_25` and `26_57` are part of the `ISERDES.IN_USE` feature: `RIOB33.IOB_Y1.ISERDES.IN_USE 26_17 26_25 26_57 29_17 29_44 30_44`.
> It seems that our fuzzer solves too many bits for it.

When on an `IN_USE` feature vary, we just need to add stimulus or a bitfilter.  For example, I think some of the "IN_USE" bits for that feature might belong to IOI3 pips.

tmichalak, 2019-07-25:
Bit `RIOB33.IOB_Y1.26_109 26_109` e.g. `0040159a_54_13`
is definitely related to the ISERDES.DATA_WIDTH feature.
Based on my minitest:
<pre>
4bit iserdes
bit_0040159a_054_11
bit_0040159a_054_25

6bit iserdes
bit_0040159a_054_11
bit_0040159a_054_15
bit_0040159a_054_25

8bit iserdes
bit_0040159a_054_11
bit_0040159a_054_13
bit_0040159a_054_25

10bit iserdes
bit_0040159a_054_11
bit_0040159a_054_13
bit_0040159a_054_15
bit_0040159a_054_25

14bit iserdes
bit_0040159a_054_11
bit_0040159a_054_13
bit_0040159a_054_15
bit_0040159a_054_25
</pre>

For litex we have 8 bit everywhere which suits the above results.

mkurc-ant, 2019-07-26:
I almost have a solution for varying data width of ISERDES. I'll push it today.

mkurc-ant, 2019-07-26:
Look at https://github.com/SymbiFlow/prjxray/pull/978

kgugala, 2019-07-29:
`RIOB33.IOB_Y1.26_109 26_109` seems to be related to interface type and data rate in ISERDESE

kgugala, 2019-07-30:
`RIOB33.IOB_Y1.26_109 26_109` is solved by #991

kgugala, 2019-07-30:
but it actually says `IOB33.IOB_Y0.ISERDES.DATA_WIDTH.W8 26_109`

litghost, 2019-07-30:
All bits in this frame are solved by current or pending PR's.

closed 2019-07-31 by litghost