f4pga/prjxray #956: Bits for 401580 base address frame 28

tmichalak, 2019-07-22:
Following 45 bits reported as missing for litex:
<pre>
# In frame 0x0040159c 48 bits were not converted.
{ unknown_bit = "0040159c_0_19", unknown_segment = "0x00401580", unknown_segbit = "28_19" }
{ unknown_bit = "0040159c_16_19", unknown_segment = "0x00401580", unknown_segbit = "28_531" }
{ unknown_bit = "0040159c_40_19", unknown_segment = "0x00401580", unknown_segbit = "28_1299" }
{ unknown_bit = "0040159c_0_25", unknown_segment = "0x00401580", unknown_segbit = "28_25" }
{ unknown_bit = "0040159c_16_25", unknown_segment = "0x00401580", unknown_segbit = "28_537" }
{ unknown_bit = "0040159c_40_25", unknown_segment = "0x00401580", unknown_segbit = "28_1305" }
{ unknown_bit = "0040159c_1_1", unknown_segment = "0x00401580", unknown_segbit = "28_33" }
{ unknown_bit = "0040159c_17_1", unknown_segment = "0x00401580", unknown_segbit = "28_545" }
{ unknown_bit = "0040159c_41_1", unknown_segment = "0x00401580", unknown_segbit = "28_1313" }
{ unknown_bit = "0040159c_1_15", unknown_segment = "0x00401580", unknown_segbit = "28_47" }
{ unknown_bit = "0040159c_17_15", unknown_segment = "0x00401580", unknown_segbit = "28_559" }
{ unknown_bit = "0040159c_41_15", unknown_segment = "0x00401580", unknown_segbit = "28_1327" }
{ unknown_bit = "0040159c_65_19", unknown_segment = "0x00401580", unknown_segbit = "28_2099" }
{ unknown_bit = "0040159c_89_19", unknown_segment = "0x00401580", unknown_segbit = "28_2867" }
{ unknown_bit = "0040159c_1_23", unknown_segment = "0x00401580", unknown_segbit = "28_55" }
{ unknown_bit = "0040159c_17_23", unknown_segment = "0x00401580", unknown_segbit = "28_567" }
{ unknown_bit = "0040159c_65_25", unknown_segment = "0x00401580", unknown_segbit = "28_2105" }
{ unknown_bit = "0040159c_41_23", unknown_segment = "0x00401580", unknown_segbit = "28_1335" }
{ unknown_bit = "0040159c_89_25", unknown_segment = "0x00401580", unknown_segbit = "28_2873" }
{ unknown_bit = "0040159c_65_31", unknown_segment = "0x00401580", unknown_segbit = "28_2111" }
{ unknown_bit = "0040159c_89_31", unknown_segment = "0x00401580", unknown_segbit = "28_2879" }
{ unknown_bit = "0040159c_26_3", unknown_segment = "0x00401580", unknown_segbit = "28_835" }
{ unknown_bit = "0040159c_82_14", unknown_segment = "0x00401580", unknown_segbit = "28_2638" }
{ unknown_bit = "0040159c_66_15", unknown_segment = "0x00401580", unknown_segbit = "28_2127" }
{ unknown_bit = "0040159c_58_14", unknown_segment = "0x00401580", unknown_segbit = "28_1870" }
{ unknown_bit = "0040159c_90_15", unknown_segment = "0x00401580", unknown_segbit = "28_2895" }
{ unknown_bit = "0040159c_66_29", unknown_segment = "0x00401580", unknown_segbit = "28_2141" }
{ unknown_bit = "0040159c_90_29", unknown_segment = "0x00401580", unknown_segbit = "28_2909" }
{ unknown_bit = "0040159c_99_9", unknown_segment = "0x00401580", unknown_segbit = "28_3177" }
{ unknown_bit = "0040159c_28_19", unknown_segment = "0x00401580", unknown_segbit = "28_915" }
{ unknown_bit = "0040159c_28_25", unknown_segment = "0x00401580", unknown_segbit = "28_921" }
{ unknown_bit = "0040159c_29_1", unknown_segment = "0x00401580", unknown_segbit = "28_929" }
{ unknown_bit = "0040159c_77_11", unknown_segment = "0x00401580", unknown_segbit = "28_2475" }
{ unknown_bit = "0040159c_29_15", unknown_segment = "0x00401580", unknown_segbit = "28_943" }
{ unknown_bit = "0040159c_77_17", unknown_segment = "0x00401580", unknown_segbit = "28_2481" }
{ unknown_bit = "0040159c_77_19", unknown_segment = "0x00401580", unknown_segbit = "28_2483" }
{ unknown_bit = "0040159c_29_23", unknown_segment = "0x00401580", unknown_segbit = "28_951" }
{ unknown_bit = "0040159c_77_25", unknown_segment = "0x00401580", unknown_segbit = "28_2489" }
{ unknown_bit = "0040159c_78_1", unknown_segment = "0x00401580", unknown_segbit = "28_2497" }
{ unknown_bit = "0040159c_38_3", unknown_segment = "0x00401580", unknown_segbit = "28_1219" }
{ unknown_bit = "0040159c_54_14", unknown_segment = "0x00401580", unknown_segbit = "28_1742" }
{ unknown_bit = "0040159c_70_14", unknown_segment = "0x00401580", unknown_segbit = "28_2254" }
{ unknown_bit = "0040159c_86_14", unknown_segment = "0x00401580", unknown_segbit = "28_2766" }
{ unknown_bit = "0040159c_62_14", unknown_segment = "0x00401580", unknown_segbit = "28_1998" }
{ unknown_bit = "0040159c_78_14", unknown_segment = "0x00401580", unknown_segbit = "28_2510" }
{ unknown_bit = "0040159c_78_15", unknown_segment = "0x00401580", unknown_segbit = "28_2511" }
{ unknown_bit = "0040159c_94_14", unknown_segment = "0x00401580", unknown_segbit = "28_3022" }
{ unknown_bit = "0040159c_78_29", unknown_segment = "0x00401580", unknown_segbit = "28_2525" }
</pre>
 are solved by:
<pre>
LIOB33.IOB_Y1.28_03 28_03
LIOB33.IOB_Y1.28_83 28_83
LIOB33.IOB_Y1.28_89 28_89
LIOB33.IOB_Y1.28_95 28_95
LIOB33.IOB_Y1.28_97 28_97
LIOB33.IOB_Y1.28_110 28_110
LIOB33.IOB_Y1.28_111 28_111
LIOB33.IOB_Y1.28_119 28_119
LIOB33.IOB_Y1.28_125 28_125
</pre>
They are close to the IOB bits.

litghost, 2019-07-22:
These likely have something to do with the IOSTANDARD used by LiteX, ``SSTL135``.

litghost, 2019-07-24:
```
LIOB33.IOB_Y1.28_03 28_03
LIOB33.IOB_Y1.28_83 28_83
LIOB33.IOB_Y1.28_89 28_89
LIOB33.IOB_Y1.28_95 28_95
LIOB33.IOB_Y1.28_97 28_97
LIOB33.IOB_Y1.28_111 28_111
LIOB33.IOB_Y1.28_119 28_119
LIOB33.IOB_Y1.28_125 28_125
```

are solved by #934

Remaining bits:
```
LIOB33.IOB_Y1.28_110 28_110
```

mkurc-ant, 2019-07-26:
I found `28_110` belonging to `IOB_Y0.ISERDES.IN_USE`. Can be solved by the WIP fuzzer https://github.com/SymbiFlow/prjxray/pull/978

litghost, 2019-07-30:
All bits in this frame are solved by current or pending PR's.

closed 2019-07-31 by litghost