f4pga/prjxray #958: Bits for base address 0x41580, frame 29

tmichalak, 2019-07-23:
Following bits reported as missing for litex:

# In frame 0x0040159d 34 bits were not converted.
{ unknown_bit = "0040159d_17_28", unknown_segment = "0x00401580", unknown_segbit = "29_572" }
{ unknown_bit = "0040159d_1_28", unknown_segment = "0x00401580", unknown_segbit = "29_60" }
{ unknown_bit = "0040159d_41_28", unknown_segment = "0x00401580", unknown_segbit = "29_1340" }
{ unknown_bit = "0040159d_26_8", unknown_segment = "0x00401580", unknown_segbit = "29_840" }
{ unknown_bit = "0040159d_50_15", unknown_segment = "0x00401580", unknown_segbit = "29_1615" }
{ unknown_bit = "0040159d_26_16", unknown_segment = "0x00401580", unknown_segbit = "29_848" }
{ unknown_bit = "0040159d_50_17", unknown_segment = "0x00401580", unknown_segbit = "29_1617" }
{ unknown_bit = "0040159d_50_19", unknown_segment = "0x00401580", unknown_segbit = "29_1619" }
{ unknown_bit = "0040159d_66_22", unknown_segment = "0x00401580", unknown_segbit = "29_2134" }
{ unknown_bit = "0040159d_90_22", unknown_segment = "0x00401580", unknown_segbit = "29_2902" }
{ unknown_bit = "0040159d_26_30", unknown_segment = "0x00401580", unknown_segbit = "29_862" }
{ unknown_bit = "0040159d_50_31", unknown_segment = "0x00401580", unknown_segbit = "29_1631" }
{ unknown_bit = "0040159d_99_2", unknown_segment = "0x00401580", unknown_segbit = "29_3170" }
{ unknown_bit = "0040159d_27_6", unknown_segment = "0x00401580", unknown_segbit = "29_870" }
{ unknown_bit = "0040159d_27_12", unknown_segment = "0x00401580", unknown_segbit = "29_876" }
{ unknown_bit = "0040159d_99_16", unknown_segment = "0x00401580", unknown_segbit = "29_3184" }
{ unknown_bit = "0040159d_67_17", unknown_segment = "0x00401580", unknown_segbit = "29_2161" }
{ unknown_bit = "0040159d_83_17", unknown_segment = "0x00401580", unknown_segbit = "29_2673" }
{ unknown_bit = "0040159d_59_17", unknown_segment = "0x00401580", unknown_segbit = "29_1905" }
{ unknown_bit = "0040159d_91_17", unknown_segment = "0x00401580", unknown_segbit = "29_2929" }
{ unknown_bit = "0040159d_99_30", unknown_segment = "0x00401580", unknown_segbit = "29_3198" }
{ unknown_bit = "0040159d_100_6", unknown_segment = "0x00401580", unknown_segbit = "29_3206" }
{ unknown_bit = "0040159d_100_12", unknown_segment = "0x00401580", unknown_segbit = "29_3212" }
{ unknown_bit = "0040159d_29_28", unknown_segment = "0x00401580", unknown_segbit = "29_956" }
{ unknown_bit = "0040159d_38_8", unknown_segment = "0x00401580", unknown_segbit = "29_1224" }
{ unknown_bit = "0040159d_38_16", unknown_segment = "0x00401580", unknown_segbit = "29_1232" }
{ unknown_bit = "0040159d_78_22", unknown_segment = "0x00401580", unknown_segbit = "29_2518" }
{ unknown_bit = "0040159d_38_30", unknown_segment = "0x00401580", unknown_segbit = "29_1246" }
{ unknown_bit = "0040159d_39_6", unknown_segment = "0x00401580", unknown_segbit = "29_1254" }
{ unknown_bit = "0040159d_39_12", unknown_segment = "0x00401580", unknown_segbit = "29_1260" }
{ unknown_bit = "0040159d_55_17", unknown_segment = "0x00401580", unknown_segbit = "29_1777" }
{ unknown_bit = "0040159d_71_17", unknown_segment = "0x00401580", unknown_segbit = "29_2289" }
{ unknown_bit = "0040159d_79_17", unknown_segment = "0x00401580", unknown_segbit = "29_2545" }
{ unknown_bit = "0040159d_95_17", unknown_segment = "0x00401580", unknown_segbit = "29_3057" }

are solved by the following bits:
LIOB33.IOB_Y1.29_08 29_08
LIOB33.IOB_Y1.29_16 29_16
LIOB33.IOB_Y1.29_17 29_17
LIOB33.IOB_Y1.29_30 29_30
LIOB33.IOB_Y1.29_38 29_38
LIOB33.IOB_Y1.29_44 29_44
LIOB33.IOB_Y1.29_118 29_118

They seem to be in the ILOGIC region

tmichalak, 2019-07-23:
@litghost Bit `LIOB33.IOB_Y1.29_118 29_118` is HCLK_CMT related? #898 
The bits in hclk_cmt segdata has: ` HCLK_CMT_L.HCLK_CMT_MUX_MMCM_CLKIN1.HCLK_CMT_CCIO3 29_118 29_119`

litghost, 2019-07-24:
```
LIOB33.IOB_Y1.29_08 29_08
LIOB33.IOB_Y1.29_16 29_16
LIOB33.IOB_Y1.29_30 29_30
LIOB33.IOB_Y1.29_38 29_38
LIOB33.IOB_Y1.29_44 29_44
LIOB33.IOB_Y1.29_118 29_118
```

are solved by #934

Remaining bits:
```
LIOB33.IOB_Y1.29_17 29_17
```

mkurc-ant, 2019-07-26:
`29_17` does belong to `IOB_Y1.ISERDES`. It is common to the `IN_USE` feature and other features related to interface type, data rate and data width (https://github.com/SymbiFlow/prjxray/pull/978).

litghost, 2019-07-30:
All bits in this frame are solved by current or pending PR's.

closed 2019-07-31 by litghost