f4pga/prjxray #959: Bits for base address 0x41580, frame 30

tmichalak, 2019-07-23:
Following 147 bits reported as missing for Litex:
<pre>
# In frame 0x0040159e 147 bits were not converted.
{ unknown_bit = "0040159e_96_5", unknown_segment = "0x00401580", unknown_segbit = "30_3077" }
{ unknown_bit = "0040159e_96_6", unknown_segment = "0x00401580", unknown_segbit = "30_3078" }
{ unknown_bit = "0040159e_96_12", unknown_segment = "0x00401580", unknown_segbit = "30_3084" }
{ unknown_bit = "0040159e_0_31", unknown_segment = "0x00401580", unknown_segbit = "30_31" }
{ unknown_bit = "0040159e_32_31", unknown_segment = "0x00401580", unknown_segbit = "30_1055" }
{ unknown_bit = "0040159e_1_3", unknown_segment = "0x00401580", unknown_segbit = "30_35" }
{ unknown_bit = "0040159e_33_3", unknown_segment = "0x00401580", unknown_segbit = "30_1059" }
{ unknown_bit = "0040159e_1_28", unknown_segment = "0x00401580", unknown_segbit = "30_60" }
{ unknown_bit = "0040159e_65_31", unknown_segment = "0x00401580", unknown_segbit = "30_2111" }
{ unknown_bit = "0040159e_97_31", unknown_segment = "0x00401580", unknown_segbit = "30_3135" }
{ unknown_bit = "0040159e_66_3", unknown_segment = "0x00401580", unknown_segbit = "30_2115" }
{ unknown_bit = "0040159e_98_3", unknown_segment = "0x00401580", unknown_segbit = "30_3139" }
{ unknown_bit = "0040159e_66_22", unknown_segment = "0x00401580", unknown_segbit = "30_2134" }
{ unknown_bit = "0040159e_66_29", unknown_segment = "0x00401580", unknown_segbit = "30_2141" }
{ unknown_bit = "0040159e_2_30", unknown_segment = "0x00401580", unknown_segbit = "30_94" }
{ unknown_bit = "0040159e_34_30", unknown_segment = "0x00401580", unknown_segbit = "30_1118" }
{ unknown_bit = "0040159e_99_2", unknown_segment = "0x00401580", unknown_segbit = "30_3170" }
{ unknown_bit = "0040159e_3_5", unknown_segment = "0x00401580", unknown_segbit = "30_101" }
{ unknown_bit = "0040159e_3_6", unknown_segment = "0x00401580", unknown_segbit = "30_102" } 
{ unknown_bit = "0040159e_35_5", unknown_segment = "0x00401580", unknown_segbit = "30_1125" }
{ unknown_bit = "0040159e_35_6", unknown_segment = "0x00401580", unknown_segbit = "30_1126" }
{ unknown_bit = "0040159e_3_12", unknown_segment = "0x00401580", unknown_segbit = "30_108" }
{ unknown_bit = "0040159e_35_12", unknown_segment = "0x00401580", unknown_segbit = "30_1132" }
{ unknown_bit = "0040159e_99_16", unknown_segment = "0x00401580", unknown_segbit = "30_3184" }
{ unknown_bit = "0040159e_67_30", unknown_segment = "0x00401580", unknown_segbit = "30_2174" }
{ unknown_bit = "0040159e_99_30", unknown_segment = "0x00401580", unknown_segbit = "30_3198" }
{ unknown_bit = "0040159e_68_5", unknown_segment = "0x00401580", unknown_segbit = "30_2181" }
{ unknown_bit = "0040159e_68_6", unknown_segment = "0x00401580", unknown_segbit = "30_2182" }
{ unknown_bit = "0040159e_100_5", unknown_segment = "0x00401580", unknown_segbit = "30_3205" }
{ unknown_bit = "0040159e_100_6", unknown_segment = "0x00401580", unknown_segbit = "30_3206" }
{ unknown_bit = "0040159e_68_12", unknown_segment = "0x00401580", unknown_segbit = "30_2188" }
{ unknown_bit = "0040159e_100_12", unknown_segment = "0x00401580", unknown_segbit = "30_3212" }
{ unknown_bit = "0040159e_4_31", unknown_segment = "0x00401580", unknown_segbit = "30_159" }
{ unknown_bit = "0040159e_36_31", unknown_segment = "0x00401580", unknown_segbit = "30_1183" }
{ unknown_bit = "0040159e_5_3", unknown_segment = "0x00401580", unknown_segbit = "30_163" }
{ unknown_bit = "0040159e_37_3", unknown_segment = "0x00401580", unknown_segbit = "30_1187" }
{ unknown_bit = "0040159e_69_31", unknown_segment = "0x00401580", unknown_segbit = "30_2239" }
{ unknown_bit = "0040159e_70_3", unknown_segment = "0x00401580", unknown_segbit = "30_2243" }
{ unknown_bit = "0040159e_38_8", unknown_segment = "0x00401580", unknown_segbit = "30_1224" }
{ unknown_bit = "0040159e_38_16", unknown_segment = "0x00401580", unknown_segbit = "30_1232" }
{ unknown_bit = "0040159e_6_30", unknown_segment = "0x00401580", unknown_segbit = "30_222" }
{ unknown_bit = "0040159e_38_30", unknown_segment = "0x00401580", unknown_segbit = "30_1246" }
{ unknown_bit = "0040159e_7_5", unknown_segment = "0x00401580", unknown_segbit = "30_229" } 
{ unknown_bit = "0040159e_7_6", unknown_segment = "0x00401580", unknown_segbit = "30_230" }
{ unknown_bit = "0040159e_39_5", unknown_segment = "0x00401580", unknown_segbit = "30_1253" }
{ unknown_bit = "0040159e_39_6", unknown_segment = "0x00401580", unknown_segbit = "30_1254" }
{ unknown_bit = "0040159e_7_12", unknown_segment = "0x00401580", unknown_segbit = "30_236" }
{ unknown_bit = "0040159e_39_12", unknown_segment = "0x00401580", unknown_segbit = "30_1260" }
{ unknown_bit = "0040159e_71_30", unknown_segment = "0x00401580", unknown_segbit = "30_2302" }
{ unknown_bit = "0040159e_72_5", unknown_segment = "0x00401580", unknown_segbit = "30_2309" }
{ unknown_bit = "0040159e_72_6", unknown_segment = "0x00401580", unknown_segbit = "30_2310" }
{ unknown_bit = "0040159e_72_12", unknown_segment = "0x00401580", unknown_segbit = "30_2316" }
{ unknown_bit = "0040159e_8_31", unknown_segment = "0x00401580", unknown_segbit = "30_287" }
{ unknown_bit = "0040159e_40_31", unknown_segment = "0x00401580", unknown_segbit = "30_1311" }
{ unknown_bit = "0040159e_9_3", unknown_segment = "0x00401580", unknown_segbit = "30_291" }
{ unknown_bit = "0040159e_41_3", unknown_segment = "0x00401580", unknown_segbit = "30_1315" }
{ unknown_bit = "0040159e_41_28", unknown_segment = "0x00401580", unknown_segbit = "30_1340" }
{ unknown_bit = "0040159e_73_31", unknown_segment = "0x00401580", unknown_segbit = "30_2367" }
{ unknown_bit = "0040159e_74_3", unknown_segment = "0x00401580", unknown_segbit = "30_2371" }
{ unknown_bit = "0040159e_10_30", unknown_segment = "0x00401580", unknown_segbit = "30_350" }
{ unknown_bit = "0040159e_42_30", unknown_segment = "0x00401580", unknown_segbit = "30_1374" }
{ unknown_bit = "0040159e_11_5", unknown_segment = "0x00401580", unknown_segbit = "30_357" }
{ unknown_bit = "0040159e_11_6", unknown_segment = "0x00401580", unknown_segbit = "30_358" }
{ unknown_bit = "0040159e_43_5", unknown_segment = "0x00401580", unknown_segbit = "30_1381" }
{ unknown_bit = "0040159e_43_6", unknown_segment = "0x00401580", unknown_segbit = "30_1382" }
{ unknown_bit = "0040159e_11_12", unknown_segment = "0x00401580", unknown_segbit = "30_364" }
{ unknown_bit = "0040159e_43_12", unknown_segment = "0x00401580", unknown_segbit = "30_1388" }
[... 71 lines cut ...]
{ unknown_bit = "0040159e_61_31", unknown_segment = "0x00401580", unknown_segbit = "30_1983" }
{ unknown_bit = "0040159e_93_31", unknown_segment = "0x00401580", unknown_segbit = "30_3007" }
{ unknown_bit = "0040159e_62_3", unknown_segment = "0x00401580", unknown_segbit = "30_1987" }
{ unknown_bit = "0040159e_94_3", unknown_segment = "0x00401580", unknown_segbit = "30_3011" }
{ unknown_bit = "0040159e_30_30", unknown_segment = "0x00401580", unknown_segbit = "30_990" }
{ unknown_bit = "0040159e_31_5", unknown_segment = "0x00401580", unknown_segbit = "30_997" }
{ unknown_bit = "0040159e_31_6", unknown_segment = "0x00401580", unknown_segbit = "30_998" }
{ unknown_bit = "0040159e_31_12", unknown_segment = "0x00401580", unknown_segbit = "30_1004" }
{ unknown_bit = "0040159e_95_30", unknown_segment = "0x00401580", unknown_segbit = "30_3070" }
</pre>
are solved by the following bits:
<pre>
LIOB33.IOB_Y1.30_08 30_08
LIOB33.IOB_Y1.30_16 30_16
LIOB33.IOB_Y1.30_30 30_30
LIOB33.IOB_Y1.30_37 30_37
LIOB33.IOB_Y1.30_38 30_38
LIOB33.IOB_Y1.30_44 30_44
LIOB33.IOB_Y1.30_95 30_95
LIOB33.IOB_Y1.30_99 30_99
LIOB33.IOB_Y1.30_118 30_118
LIOB33.IOB_Y1.30_125 30_125
</pre>

The seem to be OLOGIC related.

acomodi, 2019-07-23:
So, I have checked and `037-ioi-pips` solves the following:
```
30_08
30_16
30_118
```

litghost, 2019-07-24:
```
LIOB33.IOB_Y1.30_37 30_37
LIOB33.IOB_Y1.30_125 30_125
```

are solved by https://github.com/SymbiFlow/prjxray/pull/971

Remaining bits are:
```
LIOB33.IOB_Y1.30_30 30_30
LIOB33.IOB_Y1.30_38 30_38
LIOB33.IOB_Y1.30_44 30_44
LIOB33.IOB_Y1.30_95 30_95
LIOB33.IOB_Y1.30_99 30_99
```

closed 2019-07-31 by litghost