f4pga/prjxray #960: Bits for base address 0x41580, frame 31

tmichalak, 2019-07-23:
Following 167 bits reported as missing for Litex:
<pre>
# In frame 0x0040159f 167 bits were not converted.
{ unknown_bit = "0040159f_96_0", unknown_segment = "0x00401580", unknown_segbit = "31_3072" }
{ unknown_bit = "0040159f_0_19", unknown_segment = "0x00401580", unknown_segbit = "31_19" }
{ unknown_bit = "0040159f_32_19", unknown_segment = "0x00401580", unknown_segbit = "31_1043" }
{ unknown_bit = "0040159f_0_25", unknown_segment = "0x00401580", unknown_segbit = "31_25" }
{ unknown_bit = "0040159f_0_26", unknown_segment = "0x00401580", unknown_segbit = "31_26" }
{ unknown_bit = "0040159f_32_25", unknown_segment = "0x00401580", unknown_segbit = "31_1049" }
{ unknown_bit = "0040159f_32_26", unknown_segment = "0x00401580", unknown_segbit = "31_1050" }
{ unknown_bit = "0040159f_1_1", unknown_segment = "0x00401580", unknown_segbit = "31_33" }
{ unknown_bit = "0040159f_33_1", unknown_segment = "0x00401580", unknown_segbit = "31_1057" }
{ unknown_bit = "0040159f_1_15", unknown_segment = "0x00401580", unknown_segbit = "31_47" }
{ unknown_bit = "0040159f_65_19", unknown_segment = "0x00401580", unknown_segbit = "31_2099" }
{ unknown_bit = "0040159f_97_19", unknown_segment = "0x00401580", unknown_segbit = "31_3123" }
{ unknown_bit = "0040159f_1_23", unknown_segment = "0x00401580", unknown_segbit = "31_55" }
{ unknown_bit = "0040159f_65_25", unknown_segment = "0x00401580", unknown_segbit = "31_2105" }
{ unknown_bit = "0040159f_65_26", unknown_segment = "0x00401580", unknown_segbit = "31_2106" }
{ unknown_bit = "0040159f_97_25", unknown_segment = "0x00401580", unknown_segbit = "31_3129" }
{ unknown_bit = "0040159f_97_26", unknown_segment = "0x00401580", unknown_segbit = "31_3130" }
{ unknown_bit = "0040159f_65_31", unknown_segment = "0x00401580", unknown_segbit = "31_2111" }
{ unknown_bit = "0040159f_98_1", unknown_segment = "0x00401580", unknown_segbit = "31_3137" }
{ unknown_bit = "0040159f_66_6", unknown_segment = "0x00401580", unknown_segbit = "31_2118" }
{ unknown_bit = "0040159f_66_14", unknown_segment = "0x00401580", unknown_segbit = "31_2126" }
{ unknown_bit = "0040159f_66_15", unknown_segment = "0x00401580", unknown_segbit = "31_2127" }
{ unknown_bit = "0040159f_66_18", unknown_segment = "0x00401580", unknown_segbit = "31_2130" }
{ unknown_bit = "0040159f_2_28", unknown_segment = "0x00401580", unknown_segbit = "31_92" }
{ unknown_bit = "0040159f_34_28", unknown_segment = "0x00401580", unknown_segbit = "31_1116" }
{ unknown_bit = "0040159f_66_29", unknown_segment = "0x00401580", unknown_segbit = "31_2141" }
{ unknown_bit = "0040159f_3_0", unknown_segment = "0x00401580", unknown_segbit = "31_96" }
{ unknown_bit = "0040159f_35_0", unknown_segment = "0x00401580", unknown_segbit = "31_1120" }
{ unknown_bit = "0040159f_99_9", unknown_segment = "0x00401580", unknown_segbit = "31_3177" }
{ unknown_bit = "0040159f_67_28", unknown_segment = "0x00401580", unknown_segbit = "31_2172" }
{ unknown_bit = "0040159f_99_28", unknown_segment = "0x00401580", unknown_segbit = "31_3196" }
{ unknown_bit = "0040159f_68_0", unknown_segment = "0x00401580", unknown_segbit = "31_2176" }
{ unknown_bit = "0040159f_100_0", unknown_segment = "0x00401580", unknown_segbit = "31_3200" }
{ unknown_bit = "0040159f_4_19", unknown_segment = "0x00401580", unknown_segbit = "31_147" }
{ unknown_bit = "0040159f_36_19", unknown_segment = "0x00401580", unknown_segbit = "31_1171" }
{ unknown_bit = "0040159f_4_25", unknown_segment = "0x00401580", unknown_segbit = "31_153" }
{ unknown_bit = "0040159f_4_26", unknown_segment = "0x00401580", unknown_segbit = "31_154" }
{ unknown_bit = "0040159f_36_25", unknown_segment = "0x00401580", unknown_segbit = "31_1177" }
{ unknown_bit = "0040159f_36_26", unknown_segment = "0x00401580", unknown_segbit = "31_1178" }
{ unknown_bit = "0040159f_5_1", unknown_segment = "0x00401580", unknown_segbit = "31_161" }
{ unknown_bit = "0040159f_37_1", unknown_segment = "0x00401580", unknown_segbit = "31_1185" }
{ unknown_bit = "0040159f_69_19", unknown_segment = "0x00401580", unknown_segbit = "31_2227" }
{ unknown_bit = "0040159f_69_25", unknown_segment = "0x00401580", unknown_segbit = "31_2233" }
{ unknown_bit = "0040159f_69_26", unknown_segment = "0x00401580", unknown_segbit = "31_2234" }
{ unknown_bit = "0040159f_70_1", unknown_segment = "0x00401580", unknown_segbit = "31_2241" }
{ unknown_bit = "0040159f_38_3", unknown_segment = "0x00401580", unknown_segbit = "31_1219" }
{ unknown_bit = "0040159f_6_28", unknown_segment = "0x00401580", unknown_segbit = "31_220" }
{ unknown_bit = "0040159f_38_28", unknown_segment = "0x00401580", unknown_segbit = "31_1244" }
{ unknown_bit = "0040159f_7_0", unknown_segment = "0x00401580", unknown_segbit = "31_224" }
{ unknown_bit = "0040159f_39_0", unknown_segment = "0x00401580", unknown_segbit = "31_1248" }
{ unknown_bit = "0040159f_71_28", unknown_segment = "0x00401580", unknown_segbit = "31_2300" }
{ unknown_bit = "0040159f_72_0", unknown_segment = "0x00401580", unknown_segbit = "31_2304" }
{ unknown_bit = "0040159f_8_19", unknown_segment = "0x00401580", unknown_segbit = "31_275" }
{ unknown_bit = "0040159f_40_19", unknown_segment = "0x00401580", unknown_segbit = "31_1299" }
{ unknown_bit = "0040159f_8_25", unknown_segment = "0x00401580", unknown_segbit = "31_281" }
{ unknown_bit = "0040159f_8_26", unknown_segment = "0x00401580", unknown_segbit = "31_282" }
{ unknown_bit = "0040159f_40_25", unknown_segment = "0x00401580", unknown_segbit = "31_1305" }
{ unknown_bit = "0040159f_40_26", unknown_segment = "0x00401580", unknown_segbit = "31_1306" }
{ unknown_bit = "0040159f_9_1", unknown_segment = "0x00401580", unknown_segbit = "31_289" }
{ unknown_bit = "0040159f_41_1", unknown_segment = "0x00401580", unknown_segbit = "31_1313" }
{ unknown_bit = "0040159f_41_15", unknown_segment = "0x00401580", unknown_segbit = "31_1327" }
{ unknown_bit = "0040159f_73_19", unknown_segment = "0x00401580", unknown_segbit = "31_2355" }
{ unknown_bit = "0040159f_41_23", unknown_segment = "0x00401580", unknown_segbit = "31_1335" }
{ unknown_bit = "0040159f_73_25", unknown_segment = "0x00401580", unknown_segbit = "31_2361" }
{ unknown_bit = "0040159f_73_26", unknown_segment = "0x00401580", unknown_segbit = "31_2362" }
{ unknown_bit = "0040159f_74_1", unknown_segment = "0x00401580", unknown_segbit = "31_2369" }
{ unknown_bit = "0040159f_10_28", unknown_segment = "0x00401580", unknown_segbit = "31_348" }
[... 94 lines cut ...]
{ unknown_bit = "0040159f_61_26", unknown_segment = "0x00401580", unknown_segbit = "31_1978" }
{ unknown_bit = "0040159f_62_1", unknown_segment = "0x00401580", unknown_segbit = "31_1985" }
{ unknown_bit = "0040159f_94_1", unknown_segment = "0x00401580", unknown_segbit = "31_3009" }
{ unknown_bit = "0040159f_30_28", unknown_segment = "0x00401580", unknown_segbit = "31_988" }
{ unknown_bit = "0040159f_31_0", unknown_segment = "0x00401580", unknown_segbit = "31_992" }
{ unknown_bit = "0040159f_95_28", unknown_segment = "0x00401580", unknown_segbit = "31_3068" }
</pre>
are solved by the following bits:
<pre>
LIOB33.IOB_Y1.31_28 31_28
LIOB33.IOB_Y1.31_32 31_32
LIOB33.IOB_Y1.31_83 31_83
LIOB33.IOB_Y1.31_89 31_89
LIOB33.IOB_Y1.31_90 31_90
LIOB33.IOB_Y1.31_95 31_95
LIOB33.IOB_Y1.31_97 31_97
LIOB33.IOB_Y1.31_102 31_102
LIOB33.IOB_Y1.31_110 31_110
LIOB33.IOB_Y1.31_111 31_111
LIOB33.IOB_Y1.31_114 31_114
LIOB33.IOB_Y1.31_119 31_119
LIOB33.IOB_Y1.31_125 31_125
</pre>

The seem to be OLOGIC/ILOGIC related

acomodi, 2019-07-23:
I have checked and `037-ioi-pips` fuzzer solves the following:
```
31_111
31_119
31_125
```

litghost, 2019-07-24:
```
LIOB33.IOB_Y1.31_90 31_90
LIOB33.IOB_Y1.31_102 31_102
LIOB33.IOB_Y1.31_110 31_110
LIOB33.IOB_Y1.31_114 31_114
```

are solved by #971

Remaining bits:
```
LIOB33.IOB_Y1.31_28 31_28
LIOB33.IOB_Y1.31_32 31_32
LIOB33.IOB_Y1.31_83 31_83
LIOB33.IOB_Y1.31_89 31_89
LIOB33.IOB_Y1.31_95 31_95
LIOB33.IOB_Y1.31_97 31_97
```

acomodi, 2019-07-29:
`LIOB33.IOB_Y1.31_95 31_95` is most probably related to the differential output mode.

closed 2019-07-31 by litghost