f4pga/prjxray #961: Bits for base address 0x41580, frame 32

tmichalak, 2019-07-23:
Following 234 bits reported as missing for Litex:
<pre>
# In frame 0x004015a0 234 bits were not converted.
{ unknown_bit = "004015a0_80_0", unknown_segment = "0x00401580", unknown_segbit = "32_2560" }
{ unknown_bit = "004015a0_96_0", unknown_segment = "0x00401580", unknown_segbit = "32_3072" }
{ unknown_bit = "004015a0_0_8", unknown_segment = "0x00401580", unknown_segbit = "32_08" }
{ unknown_bit = "004015a0_16_8", unknown_segment = "0x00401580", unknown_segbit = "32_520" }
{ unknown_bit = "004015a0_32_8", unknown_segment = "0x00401580", unknown_segbit = "32_1032" }
{ unknown_bit = "004015a0_48_8", unknown_segment = "0x00401580", unknown_segbit = "32_1544" }
{ unknown_bit = "004015a0_80_14", unknown_segment = "0x00401580", unknown_segbit = "32_2574" }
{ unknown_bit = "004015a0_96_14", unknown_segment = "0x00401580", unknown_segbit = "32_3086" }
{ unknown_bit = "004015a0_0_18", unknown_segment = "0x00401580", unknown_segbit = "32_18" }
{ unknown_bit = "004015a0_16_18", unknown_segment = "0x00401580", unknown_segbit = "32_530" }
{ unknown_bit = "004015a0_32_18", unknown_segment = "0x00401580", unknown_segbit = "32_1042" }
{ unknown_bit = "004015a0_48_18", unknown_segment = "0x00401580", unknown_segbit = "32_1554" }
{ unknown_bit = "004015a0_80_20", unknown_segment = "0x00401580", unknown_segbit = "32_2580" }
{ unknown_bit = "004015a0_80_22", unknown_segment = "0x00401580", unknown_segbit = "32_2582" }
{ unknown_bit = "004015a0_96_20", unknown_segment = "0x00401580", unknown_segbit = "32_3092" }
{ unknown_bit = "004015a0_96_22", unknown_segment = "0x00401580", unknown_segbit = "32_3094" }
{ unknown_bit = "004015a0_0_30", unknown_segment = "0x00401580", unknown_segbit = "32_30" }
{ unknown_bit = "004015a0_16_30", unknown_segment = "0x00401580", unknown_segbit = "32_542" }
{ unknown_bit = "004015a0_32_30", unknown_segment = "0x00401580", unknown_segbit = "32_1054" }
{ unknown_bit = "004015a0_48_30", unknown_segment = "0x00401580", unknown_segbit = "32_1566" }
{ unknown_bit = "004015a0_65_8", unknown_segment = "0x00401580", unknown_segbit = "32_2088" }
{ unknown_bit = "004015a0_81_8", unknown_segment = "0x00401580", unknown_segbit = "32_2600" }
{ unknown_bit = "004015a0_81_9", unknown_segment = "0x00401580", unknown_segbit = "32_2601" }
{ unknown_bit = "004015a0_97_8", unknown_segment = "0x00401580", unknown_segbit = "32_3112" }
{ unknown_bit = "004015a0_1_12", unknown_segment = "0x00401580", unknown_segbit = "32_44" }
{ unknown_bit = "004015a0_17_12", unknown_segment = "0x00401580", unknown_segbit = "32_556" }
{ unknown_bit = "004015a0_33_12", unknown_segment = "0x00401580", unknown_segbit = "32_1068" }
{ unknown_bit = "004015a0_49_12", unknown_segment = "0x00401580", unknown_segbit = "32_1580" }
{ unknown_bit = "004015a0_1_16", unknown_segment = "0x00401580", unknown_segbit = "32_48" }
{ unknown_bit = "004015a0_17_16", unknown_segment = "0x00401580", unknown_segbit = "32_560" }
{ unknown_bit = "004015a0_33_16", unknown_segment = "0x00401580", unknown_segbit = "32_1072" }
{ unknown_bit = "004015a0_49_16", unknown_segment = "0x00401580", unknown_segbit = "32_1584" }
{ unknown_bit = "004015a0_65_18", unknown_segment = "0x00401580", unknown_segbit = "32_2098" }
{ unknown_bit = "004015a0_81_18", unknown_segment = "0x00401580", unknown_segbit = "32_2610" }
{ unknown_bit = "004015a0_97_18", unknown_segment = "0x00401580", unknown_segbit = "32_3122" }
{ unknown_bit = "004015a0_65_30", unknown_segment = "0x00401580", unknown_segbit = "32_2110" }
{ unknown_bit = "004015a0_81_30", unknown_segment = "0x00401580", unknown_segbit = "32_2622" }
{ unknown_bit = "004015a0_97_30", unknown_segment = "0x00401580", unknown_segbit = "32_3134" }
{ unknown_bit = "004015a0_66_12", unknown_segment = "0x00401580", unknown_segbit = "32_2124" }
{ unknown_bit = "004015a0_82_12", unknown_segment = "0x00401580", unknown_segbit = "32_2636" }
{ unknown_bit = "004015a0_66_16", unknown_segment = "0x00401580", unknown_segbit = "32_2128" }
{ unknown_bit = "004015a0_82_16", unknown_segment = "0x00401580", unknown_segbit = "32_2640" }
{ unknown_bit = "004015a0_98_12", unknown_segment = "0x00401580", unknown_segbit = "32_3148" }
{ unknown_bit = "004015a0_98_16", unknown_segment = "0x00401580", unknown_segbit = "32_3152" }
{ unknown_bit = "004015a0_2_20", unknown_segment = "0x00401580", unknown_segbit = "32_84" }
{ unknown_bit = "004015a0_18_20", unknown_segment = "0x00401580", unknown_segbit = "32_596" }
{ unknown_bit = "004015a0_34_20", unknown_segment = "0x00401580", unknown_segbit = "32_1108" }
{ unknown_bit = "004015a0_3_0", unknown_segment = "0x00401580", unknown_segbit = "32_96" }
{ unknown_bit = "004015a0_19_0", unknown_segment = "0x00401580", unknown_segbit = "32_608" }
{ unknown_bit = "004015a0_35_0", unknown_segment = "0x00401580", unknown_segbit = "32_1120" }
{ unknown_bit = "004015a0_3_14", unknown_segment = "0x00401580", unknown_segbit = "32_110" }
{ unknown_bit = "004015a0_19_14", unknown_segment = "0x00401580", unknown_segbit = "32_622" }
{ unknown_bit = "004015a0_35_14", unknown_segment = "0x00401580", unknown_segbit = "32_1134" }
{ unknown_bit = "004015a0_3_20", unknown_segment = "0x00401580", unknown_segbit = "32_116" }
{ unknown_bit = "004015a0_19_20", unknown_segment = "0x00401580", unknown_segbit = "32_628" }
{ unknown_bit = "004015a0_3_22", unknown_segment = "0x00401580", unknown_segbit = "32_118" }
{ unknown_bit = "004015a0_19_22", unknown_segment = "0x00401580", unknown_segbit = "32_630" }
{ unknown_bit = "004015a0_35_20", unknown_segment = "0x00401580", unknown_segbit = "32_1140" }
{ unknown_bit = "004015a0_35_22", unknown_segment = "0x00401580", unknown_segbit = "32_1142" }
{ unknown_bit = "004015a0_67_20", unknown_segment = "0x00401580", unknown_segbit = "32_2164" }
{ unknown_bit = "004015a0_83_20", unknown_segment = "0x00401580", unknown_segbit = "32_2676" }
{ unknown_bit = "004015a0_99_20", unknown_segment = "0x00401580", unknown_segbit = "32_3188" }
{ unknown_bit = "004015a0_68_0", unknown_segment = "0x00401580", unknown_segbit = "32_2176" }
{ unknown_bit = "004015a0_84_0", unknown_segment = "0x00401580", unknown_segbit = "32_2688" }
{ unknown_bit = "004015a0_4_8", unknown_segment = "0x00401580", unknown_segbit = "32_136" }
{ unknown_bit = "004015a0_20_8", unknown_segment = "0x00401580", unknown_segbit = "32_648" }
{ unknown_bit = "004015a0_36_8", unknown_segment = "0x00401580", unknown_segbit = "32_1160" }
[... 160 lines cut ...]
{ unknown_bit = "004015a0_47_14", unknown_segment = "0x00401580", unknown_segbit = "32_1518" }
{ unknown_bit = "004015a0_31_20", unknown_segment = "0x00401580", unknown_segbit = "32_1012" }
{ unknown_bit = "004015a0_47_20", unknown_segment = "0x00401580", unknown_segbit = "32_1524" }
{ unknown_bit = "004015a0_31_22", unknown_segment = "0x00401580", unknown_segbit = "32_1014" }
{ unknown_bit = "004015a0_47_22", unknown_segment = "0x00401580", unknown_segbit = "32_1526" }
{ unknown_bit = "004015a0_79_20", unknown_segment = "0x00401580", unknown_segbit = "32_2548" }
{ unknown_bit = "004015a0_95_20", unknown_segment = "0x00401580", unknown_segbit = "32_3060" }
</pre>

are solved by the following bits:
<pre>
LIOB33.IOB_Y1.32_20 32_20
LIOB33.IOB_Y1.32_32 32_32
LIOB33.IOB_Y1.32_46 32_46
LIOB33.IOB_Y1.32_52 32_52
LIOB33.IOB_Y1.32_54 32_54
LIOB33.IOB_Y1.32_72 32_72
LIOB33.IOB_Y1.32_73 32_73
LIOB33.IOB_Y1.32_82 32_82
LIOB33.IOB_Y1.32_94 32_94
LIOB33.IOB_Y1.32_108 32_108
LIOB33.IOB_Y1.32_112 32_112
</pre>

They seem to be OSERDES related

litghost, 2019-07-24:
```
LIOB33.IOB_Y1.32_20 32_20
LIOB33.IOB_Y1.32_32 32_32
LIOB33.IOB_Y1.32_46 32_46
LIOB33.IOB_Y1.32_52 32_52
LIOB33.IOB_Y1.32_82 32_82
LIOB33.IOB_Y1.32_108 32_108
```

are solved by https://github.com/SymbiFlow/prjxray/pull/971

Remaining bits:
```
LIOB33.IOB_Y1.32_54 32_54
LIOB33.IOB_Y1.32_72 32_72
LIOB33.IOB_Y1.32_73 32_73
LIOB33.IOB_Y1.32_94 32_94
LIOB33.IOB_Y1.32_112 32_112
```

acomodi, 2019-07-26:
```
LIOB33.IOB_Y1.32_54 32_54
LIOB33.IOB_Y1.32_112 32_112
```

Documented with https://github.com/SymbiFlow/prjxray/pull/975

Remaining:
```
LIOB33.IOB_Y1.32_72 32_72
LIOB33.IOB_Y1.32_73 32_73
LIOB33.IOB_Y1.32_94 32_94
```

acomodi, 2019-07-26:
`LIOB33.IOB_Y1.32_73 32_73` is related to TOP IODELAYE2, most probably related to an `IN_USE` feature.

acomodi, 2019-07-26:
https://github.com/SymbiFlow/prjxray/pull/980 solves  `LIOB33.IOB_Y1.32_73 32_73`.

Remaining:
```
LIOB33.IOB_Y1.32_72 32_72
LIOB33.IOB_Y1.32_94 32_94
```

litghost, 2019-07-30:
All bits in this frame are solved by current or pending PR's.

closed 2019-07-31 by litghost