f4pga/prjxray #962: Bits for base address 0x41580, frame 33

tmichalak, 2019-07-23:
Following 233 bits reported as missing for Litex:
<pre>
# In frame 0x004015a1 233 bits were not converted.
{ unknown_bit = "004015a1_80_1", unknown_segment = "0x00401580", unknown_segbit = "33_2561" }
{ unknown_bit = "004015a1_96_1", unknown_segment = "0x00401580", unknown_segbit = "33_3073" }
{ unknown_bit = "004015a1_47_23", unknown_segment = "0x00401580", unknown_segbit = "33_1527" }
{ unknown_bit = "004015a1_0_9", unknown_segment = "0x00401580", unknown_segbit = "33_09" }
{ unknown_bit = "004015a1_16_9", unknown_segment = "0x00401580", unknown_segbit = "33_521" }
{ unknown_bit = "004015a1_0_11", unknown_segment = "0x00401580", unknown_segbit = "33_11" }
{ unknown_bit = "004015a1_16_11", unknown_segment = "0x00401580", unknown_segbit = "33_523" }
{ unknown_bit = "004015a1_32_9", unknown_segment = "0x00401580", unknown_segbit = "33_1033" }
{ unknown_bit = "004015a1_32_11", unknown_segment = "0x00401580", unknown_segbit = "33_1035" }
{ unknown_bit = "004015a1_48_9", unknown_segment = "0x00401580", unknown_segbit = "33_1545" }
{ unknown_bit = "004015a1_48_11", unknown_segment = "0x00401580", unknown_segbit = "33_1547" }
{ unknown_bit = "004015a1_0_17", unknown_segment = "0x00401580", unknown_segbit = "33_17" }
{ unknown_bit = "004015a1_16_17", unknown_segment = "0x00401580", unknown_segbit = "33_529" }
{ unknown_bit = "004015a1_32_17", unknown_segment = "0x00401580", unknown_segbit = "33_1041" }
{ unknown_bit = "004015a1_48_17", unknown_segment = "0x00401580", unknown_segbit = "33_1553" }
{ unknown_bit = "004015a1_80_13", unknown_segment = "0x00401580", unknown_segbit = "33_2573" }
{ unknown_bit = "004015a1_80_22", unknown_segment = "0x00401580", unknown_segbit = "33_2582" }
{ unknown_bit = "004015a1_80_23", unknown_segment = "0x00401580", unknown_segbit = "33_2583" }
{ unknown_bit = "004015a1_96_22", unknown_segment = "0x00401580", unknown_segbit = "33_3094" }
{ unknown_bit = "004015a1_96_23", unknown_segment = "0x00401580", unknown_segbit = "33_3095" }
{ unknown_bit = "004015a1_0_31", unknown_segment = "0x00401580", unknown_segbit = "33_31" }
{ unknown_bit = "004015a1_16_31", unknown_segment = "0x00401580", unknown_segbit = "33_543" }
{ unknown_bit = "004015a1_32_31", unknown_segment = "0x00401580", unknown_segbit = "33_1055" }
{ unknown_bit = "004015a1_48_31", unknown_segment = "0x00401580", unknown_segbit = "33_1567" }
{ unknown_bit = "004015a1_65_9", unknown_segment = "0x00401580", unknown_segbit = "33_2089" }
{ unknown_bit = "004015a1_81_9", unknown_segment = "0x00401580", unknown_segbit = "33_2601" }
{ unknown_bit = "004015a1_1_11", unknown_segment = "0x00401580", unknown_segbit = "33_43" }
{ unknown_bit = "004015a1_17_11", unknown_segment = "0x00401580", unknown_segbit = "33_555" }
{ unknown_bit = "004015a1_33_11", unknown_segment = "0x00401580", unknown_segbit = "33_1067" }
{ unknown_bit = "004015a1_49_11", unknown_segment = "0x00401580", unknown_segbit = "33_1579" }
{ unknown_bit = "004015a1_65_11", unknown_segment = "0x00401580", unknown_segbit = "33_2091" }
{ unknown_bit = "004015a1_81_11", unknown_segment = "0x00401580", unknown_segbit = "33_2603" }
{ unknown_bit = "004015a1_65_17", unknown_segment = "0x00401580", unknown_segbit = "33_2097" }
{ unknown_bit = "004015a1_81_17", unknown_segment = "0x00401580", unknown_segbit = "33_2609" }
{ unknown_bit = "004015a1_97_11", unknown_segment = "0x00401580", unknown_segbit = "33_3115" }
{ unknown_bit = "004015a1_97_17", unknown_segment = "0x00401580", unknown_segbit = "33_3121" }
{ unknown_bit = "004015a1_65_31", unknown_segment = "0x00401580", unknown_segbit = "33_2111" }
{ unknown_bit = "004015a1_81_31", unknown_segment = "0x00401580", unknown_segbit = "33_2623" }
{ unknown_bit = "004015a1_97_31", unknown_segment = "0x00401580", unknown_segbit = "33_3135" }
{ unknown_bit = "004015a1_66_11", unknown_segment = "0x00401580", unknown_segbit = "33_2123" }
{ unknown_bit = "004015a1_82_11", unknown_segment = "0x00401580", unknown_segbit = "33_2635" }
{ unknown_bit = "004015a1_98_11", unknown_segment = "0x00401580", unknown_segbit = "33_3147" }
{ unknown_bit = "004015a1_2_15", unknown_segment = "0x00401580", unknown_segbit = "33_79" }
{ unknown_bit = "004015a1_18_15", unknown_segment = "0x00401580", unknown_segbit = "33_591" }
{ unknown_bit = "004015a1_34_15", unknown_segment = "0x00401580", unknown_segbit = "33_1103" }
{ unknown_bit = "004015a1_2_19", unknown_segment = "0x00401580", unknown_segbit = "33_83" }
{ unknown_bit = "004015a1_18_19", unknown_segment = "0x00401580", unknown_segbit = "33_595" }
{ unknown_bit = "004015a1_34_19", unknown_segment = "0x00401580", unknown_segbit = "33_1107" }
{ unknown_bit = "004015a1_3_1", unknown_segment = "0x00401580", unknown_segbit = "33_97" }
{ unknown_bit = "004015a1_19_1", unknown_segment = "0x00401580", unknown_segbit = "33_609" }
{ unknown_bit = "004015a1_35_1", unknown_segment = "0x00401580", unknown_segbit = "33_1121" }
{ unknown_bit = "004015a1_3_13", unknown_segment = "0x00401580", unknown_segbit = "33_109" }
{ unknown_bit = "004015a1_19_13", unknown_segment = "0x00401580", unknown_segbit = "33_621" }
{ unknown_bit = "004015a1_35_13", unknown_segment = "0x00401580", unknown_segbit = "33_1133" }
{ unknown_bit = "004015a1_67_15", unknown_segment = "0x00401580", unknown_segbit = "33_2159" }
{ unknown_bit = "004015a1_83_15", unknown_segment = "0x00401580", unknown_segbit = "33_2671" }
{ unknown_bit = "004015a1_67_19", unknown_segment = "0x00401580", unknown_segbit = "33_2163" }
{ unknown_bit = "004015a1_83_19", unknown_segment = "0x00401580", unknown_segbit = "33_2675" }
{ unknown_bit = "004015a1_99_15", unknown_segment = "0x00401580", unknown_segbit = "33_3183" }
{ unknown_bit = "004015a1_19_23", unknown_segment = "0x00401580", unknown_segbit = "33_631" }
{ unknown_bit = "004015a1_3_23", unknown_segment = "0x00401580", unknown_segbit = "33_119" }
{ unknown_bit = "004015a1_35_23", unknown_segment = "0x00401580", unknown_segbit = "33_1143" }
{ unknown_bit = "004015a1_99_19", unknown_segment = "0x00401580", unknown_segbit = "33_3187" }
{ unknown_bit = "004015a1_68_1", unknown_segment = "0x00401580", unknown_segbit = "33_2177" }
{ unknown_bit = "004015a1_84_1", unknown_segment = "0x00401580", unknown_segbit = "33_2689" }
{ unknown_bit = "004015a1_100_1", unknown_segment = "0x00401580", unknown_segbit = "33_3201" }
{ unknown_bit = "004015a1_4_9", unknown_segment = "0x00401580", unknown_segbit = "33_137" }
[... 158 lines cut ...]
{ unknown_bit = "004015a1_47_1", unknown_segment = "0x00401580", unknown_segbit = "33_1505" }
{ unknown_bit = "004015a1_31_13", unknown_segment = "0x00401580", unknown_segbit = "33_1005" }
{ unknown_bit = "004015a1_47_13", unknown_segment = "0x00401580", unknown_segbit = "33_1517" }
{ unknown_bit = "004015a1_79_15", unknown_segment = "0x00401580", unknown_segbit = "33_2543" }
{ unknown_bit = "004015a1_95_15", unknown_segment = "0x00401580", unknown_segbit = "33_3055" }
{ unknown_bit = "004015a1_79_19", unknown_segment = "0x00401580", unknown_segbit = "33_2547" }
{ unknown_bit = "004015a1_95_19", unknown_segment = "0x00401580", unknown_segbit = "33_3059" }
{ unknown_bit = "004015a1_31_23", unknown_segment = "0x00401580", unknown_segbit = "33_1015" }
</pre>
are solved by the following bits:
<pre>
LIOB33.IOB_Y1.33_15 33_15                                                                                                                    
LIOB33.IOB_Y1.33_19 33_19                                                                                                                    
LIOB33.IOB_Y1.33_33 33_33                                                                                                                    
LIOB33.IOB_Y1.33_45 33_45                                                                                                                    
LIOB33.IOB_Y1.33_54 33_54                                                                                                                    
LIOB33.IOB_Y1.33_55 33_55                                                                                                                    
LIOB33.IOB_Y1.33_73 33_73                                                                                                                    
LIOB33.IOB_Y1.33_75 33_75                                                                                                                    
LIOB33.IOB_Y1.33_81 33_81                                                                                                                    
LIOB33.IOB_Y1.33_95 33_95
LIOB33.IOB_Y1.33_107 33_107    
</pre>

They seem to be OSERDES related

litghost, 2019-07-24:
```                                                          
LIOB33.IOB_Y1.33_19 33_19                                                                                                                    
LIOB33.IOB_Y1.33_45 33_45                                                                                                                    
LIOB33.IOB_Y1.33_75 33_75                                                                                                                       
LIOB33.IOB_Y1.33_81 33_81                                                                                                                     
LIOB33.IOB_Y1.33_95 33_95 
LIOB33.IOB_Y1.33_107 33_107  
```
Are documented in https://github.com/SymbiFlow/prjxray/pull/971

Remaining:
```
LIOB33.IOB_Y1.33_15 33_15                                                                                                                 
LIOB33.IOB_Y1.33_33 33_33                                                                                                                    
LIOB33.IOB_Y1.33_54 33_54                                                                                                                    
LIOB33.IOB_Y1.33_55 33_55                                                                                                                    
LIOB33.IOB_Y1.33_73 33_73    
```

acomodi, 2019-07-24:
I have been investigating on what features/configurations do trigger the remaining bits. Just by instantiating an OSERDES the following remaining bit showed up as unknown bits:
```
33_15
33_33
33_55
```

By trying and tweaking the configurations of the OSERDES, those bits remained unchanged.
I though maybe those are like `USE` bits. I tried to add a `OSERDESE.USE` tag and I could get some bits with the addition of unrelated bits (belonging to the IOI3 pips I believe)

litghost, 2019-07-24:
I don't believe ``33_15`` is OSERDES, because of the following command:
```
find -name "segdata*" -exec grep -nH "seg\|33_15\|tag .*OLOGIC[01].* 1" {} \;
```

Which shows at least one case where ``33_15`` is set and no bits are set.

litghost, 2019-07-24:
I'm actually inclined to believe that ``33_15`` is related to ``OCLKM`` bits:

```
./build/1/specimen_036/segdata_lioi3_tbyteterm.txt:492:seg 00020000_075
./build/1/specimen_036/segdata_lioi3_tbyteterm.txt:592:bit 33_15
./build/1/specimen_036/segdata_lioi3_tbyteterm.txt:610:tag IOI3.IOI_ILOGIC0_OCLKB.IOI_OCLKM_0 1
./build/1/specimen_036/segdata_lioi3_tbyteterm.txt:621:tag IOI3.IOI_OCLKM_0.IOI_LEAF_GCLK3 1
./build/1/specimen_036/segdata_lioi3_tbyteterm.txt:637:tag IOI3.IOI_OCLK_0.IOI_LEAF_GCLK3 1
./build/1/specimen_036/segdata_lioi3_tbyteterm.txt:638:tag IOI3.IOI_OCLK_1.IOI_LEAF_GCLK4 1
```

acomodi, 2019-07-24:
That is interesting, but the fact is that `33_15` shows up even when OCLKM or any clock sources actually is used. In my simple test design I have instantiated an OSERDES which was only connecting the output to an IOPAD output, and the IOI3_INTER did not route any clock signal.

acomodi, 2019-07-25:
I have tried to go on with the `OSERDESE.USE` strategy. I think that we can have zero OLOGIC pips in case there is no clock source. This because the pips in the `IOI3_INTER` that are not pseudo-pips are related almost only to clock sources (I need to be sure of that).

I have updated locally the `036-ologic` fuzzer starting from https://github.com/SymbiFlow/prjxray/pull/971 and removed the clock sources from the OSERDESE2 instances. This produced interesting results and I got 4 of the remaining bits listed in https://github.com/SymbiFlow/prjxray/issues/867#issuecomment-514523385.

The following is a diff between the segbits I obtained w.r.t. to the one obtained in https://github.com/SymbiFlow/prjxray/pull/971
```
diff build/segbits_xiob33.db /tmp/segbits_xiob33.db
0a1
> IOB33.IOB_Y0.IS_CLKDIV_INVERTED 30_85
25d25
< IOB33.IOB_Y0.OSERDESE.IN_USE 32_112 33_73
27a28
> IOB33.IOB_Y0.ZINV_CLK 31_90 31_92
29a31
> IOB33.IOB_Y1.IS_CLKDIV_INVERTED 31_42
54d55
< IOB33.IOB_Y1.OSERDESE.IN_USE 32_54 33_15
56a58
> IOB33.IOB_Y1.ZINV_CLK 30_35 30_37
```

I think that by making optional the insertion of the clock signals in oserdes I can correctly solve the bits for:
```
IOB33.IOB_Y0.OSERDESE.IN_USE 32_112 33_73
IOB33.IOB_Y1.OSERDESE.IN_USE 32_54 33_15
```

litghost, 2019-07-25:
``33_33``, ``33_55``  are related to ODDR SRTYPE and TSRTYPE.  I'll create a PR once #975 lands.

acomodi, 2019-07-26:
```
LIOB33.IOB_Y1.33_15 33_15                                                                                                                                                                                                                                  
LIOB33.IOB_Y1.33_73 33_73  
```
Documented in https://github.com/SymbiFlow/prjxray/pull/975

```
LIOB33.IOB_Y1.33_33 33_33                                                                                                                                                                                                                                       
LIOB33.IOB_Y1.33_55 33_55
```
Discovered, but still need 036-ologic fuzzer adjustment.

Remaining:
```
LIOB33.IOB_Y1.33_54 33_54 
```

acomodi, 2019-07-26:
`LIOB33.IOB_Y1.33_54 33_54` Is related to IDELAYE2

acomodi, 2019-07-26:
https://github.com/SymbiFlow/prjxray/pull/980 solves `LIOB33.IOB_Y1.33_54 33_54`

PR #981 (2019-07-30): Add TSRTYPE features to OLOGIC fuzzer
- Also prefix decimal features with A-Z

Fixes #962
closed 2019-07-30 by litghost