f4pga/prjxray #963: Bits for base address 0x41580, frame 35

tmichalak, 2019-07-23:
Following 8 bits reported as missing for Litex:
<pre>
{ unknown_bit = "004015a3_56_1", unknown_segment = "0x00401580", unknown_segbit = "35_1793" }
{ unknown_bit = "004015a3_60_1", unknown_segment = "0x00401580", unknown_segbit = "35_1921" }
{ unknown_bit = "004015a3_68_1", unknown_segment = "0x00401580", unknown_segbit = "35_2177" }
{ unknown_bit = "004015a3_72_1", unknown_segment = "0x00401580", unknown_segbit = "35_2305" }
{ unknown_bit = "004015a3_80_1", unknown_segment = "0x00401580", unknown_segbit = "35_2561" }
{ unknown_bit = "004015a3_84_1", unknown_segment = "0x00401580", unknown_segbit = "35_2689" }
{ unknown_bit = "004015a3_92_1", unknown_segment = "0x00401580", unknown_segbit = "35_2945" }
{ unknown_bit = "004015a3_96_1", unknown_segment = "0x00401580", unknown_segbit = "35_3073" }
</pre>
are solved by:
<pre>
LIOB33.IOB_Y1.35_33 35_33
</pre>

The bit seems to be IDELAY related

tmichalak, 2019-07-23:
@mkurc-ant Could you have a look?

mkurc-ant, 2019-07-23:
@tmichalak The 35_33 bit correlates negatively with IDELAY_Y1.IDELAY_VALUE[4] meaning that it should be cleared when the feature is set.

The fuzzer for IDELAY has already been merged, but there is an pending update by litghost https://github.com/SymbiFlow/prjxray/pull/957

tmichalak, 2019-07-24:
Solved with #957

closed 2019-07-26 by litghost