f4pga/prjxray #964: Bits for base address 0x41580, frame 39

tmichalak, 2019-07-23:
Following 88 bits reported as missing for Litex:
<pre>
{ unknown_bit = "004015a7_64_27", unknown_segment = "0x00401580", unknown_segbit = "39_2075" }
{ unknown_bit = "004015a7_64_29", unknown_segment = "0x00401580", unknown_segbit = "39_2077" }
{ unknown_bit = "004015a7_1_11", unknown_segment = "0x00401580", unknown_segbit = "39_43" }
{ unknown_bit = "004015a7_1_15", unknown_segment = "0x00401580", unknown_segbit = "39_47" }
{ unknown_bit = "004015a7_66_11", unknown_segment = "0x00401580", unknown_segbit = "39_2123" }
{ unknown_bit = "004015a7_66_15", unknown_segment = "0x00401580", unknown_segbit = "39_2127" }
{ unknown_bit = "004015a7_67_5", unknown_segment = "0x00401580", unknown_segbit = "39_2149" }
{ unknown_bit = "004015a7_67_7", unknown_segment = "0x00401580", unknown_segbit = "39_2151" }
{ unknown_bit = "004015a7_5_11", unknown_segment = "0x00401580", unknown_segbit = "39_171" }
{ unknown_bit = "004015a7_5_15", unknown_segment = "0x00401580", unknown_segbit = "39_175" }
{ unknown_bit = "004015a7_70_11", unknown_segment = "0x00401580", unknown_segbit = "39_2251" }
{ unknown_bit = "004015a7_70_15", unknown_segment = "0x00401580", unknown_segbit = "39_2255" }
{ unknown_bit = "004015a7_70_25", unknown_segment = "0x00401580", unknown_segbit = "39_2265" }
{ unknown_bit = "004015a7_70_27", unknown_segment = "0x00401580", unknown_segbit = "39_2267" }
{ unknown_bit = "004015a7_71_5", unknown_segment = "0x00401580", unknown_segbit = "39_2277" }
{ unknown_bit = "004015a7_71_7", unknown_segment = "0x00401580", unknown_segbit = "39_2279" }
{ unknown_bit = "004015a7_9_11", unknown_segment = "0x00401580", unknown_segbit = "39_299" }
{ unknown_bit = "004015a7_9_15", unknown_segment = "0x00401580", unknown_segbit = "39_303" }
{ unknown_bit = "004015a7_74_11", unknown_segment = "0x00401580", unknown_segbit = "39_2379" }
{ unknown_bit = "004015a7_74_15", unknown_segment = "0x00401580", unknown_segbit = "39_2383" }
{ unknown_bit = "004015a7_13_11", unknown_segment = "0x00401580", unknown_segbit = "39_427" }
{ unknown_bit = "004015a7_13_15", unknown_segment = "0x00401580", unknown_segbit = "39_431" }
{ unknown_bit = "004015a7_78_11", unknown_segment = "0x00401580", unknown_segbit = "39_2507" }
{ unknown_bit = "004015a7_78_15", unknown_segment = "0x00401580", unknown_segbit = "39_2511" }
{ unknown_bit = "004015a7_78_25", unknown_segment = "0x00401580", unknown_segbit = "39_2521" }
{ unknown_bit = "004015a7_78_27", unknown_segment = "0x00401580", unknown_segbit = "39_2523" }
{ unknown_bit = "004015a7_79_5", unknown_segment = "0x00401580", unknown_segbit = "39_2533" }
{ unknown_bit = "004015a7_79_7", unknown_segment = "0x00401580", unknown_segbit = "39_2535" }
{ unknown_bit = "004015a7_15_27", unknown_segment = "0x00401580", unknown_segbit = "39_507" }
{ unknown_bit = "004015a7_15_29", unknown_segment = "0x00401580", unknown_segbit = "39_509" }
{ unknown_bit = "004015a7_17_11", unknown_segment = "0x00401580", unknown_segbit = "39_555" }
{ unknown_bit = "004015a7_17_15", unknown_segment = "0x00401580", unknown_segbit = "39_559" }
{ unknown_bit = "004015a7_82_11", unknown_segment = "0x00401580", unknown_segbit = "39_2635" }
{ unknown_bit = "004015a7_82_15", unknown_segment = "0x00401580", unknown_segbit = "39_2639" }
{ unknown_bit = "004015a7_82_25", unknown_segment = "0x00401580", unknown_segbit = "39_2649" }
{ unknown_bit = "004015a7_82_27", unknown_segment = "0x00401580", unknown_segbit = "39_2651" }
{ unknown_bit = "004015a7_83_5", unknown_segment = "0x00401580", unknown_segbit = "39_2661" }
{ unknown_bit = "004015a7_83_7", unknown_segment = "0x00401580", unknown_segbit = "39_2663" }
{ unknown_bit = "004015a7_21_11", unknown_segment = "0x00401580", unknown_segbit = "39_683" }
{ unknown_bit = "004015a7_21_15", unknown_segment = "0x00401580", unknown_segbit = "39_687" }
{ unknown_bit = "004015a7_86_11", unknown_segment = "0x00401580", unknown_segbit = "39_2763" }
{ unknown_bit = "004015a7_86_15", unknown_segment = "0x00401580", unknown_segbit = "39_2767" }
{ unknown_bit = "004015a7_86_25", unknown_segment = "0x00401580", unknown_segbit = "39_2777" }
{ unknown_bit = "004015a7_86_27", unknown_segment = "0x00401580", unknown_segbit = "39_2779" }
{ unknown_bit = "004015a7_88_27", unknown_segment = "0x00401580", unknown_segbit = "39_2843" }
{ unknown_bit = "004015a7_88_29", unknown_segment = "0x00401580", unknown_segbit = "39_2845" }
{ unknown_bit = "004015a7_25_11", unknown_segment = "0x00401580", unknown_segbit = "39_811" }
{ unknown_bit = "004015a7_25_15", unknown_segment = "0x00401580", unknown_segbit = "39_815" }
{ unknown_bit = "004015a7_90_11", unknown_segment = "0x00401580", unknown_segbit = "39_2891" }
{ unknown_bit = "004015a7_90_15", unknown_segment = "0x00401580", unknown_segbit = "39_2895" }
{ unknown_bit = "004015a7_91_5", unknown_segment = "0x00401580", unknown_segbit = "39_2917" }
{ unknown_bit = "004015a7_91_7", unknown_segment = "0x00401580", unknown_segbit = "39_2919" }
{ unknown_bit = "004015a7_29_11", unknown_segment = "0x00401580", unknown_segbit = "39_939" }
{ unknown_bit = "004015a7_29_15", unknown_segment = "0x00401580", unknown_segbit = "39_943" }
{ unknown_bit = "004015a7_94_11", unknown_segment = "0x00401580", unknown_segbit = "39_3019" }
{ unknown_bit = "004015a7_94_15", unknown_segment = "0x00401580", unknown_segbit = "39_3023" }
{ unknown_bit = "004015a7_94_25", unknown_segment = "0x00401580", unknown_segbit = "39_3033" }
{ unknown_bit = "004015a7_94_27", unknown_segment = "0x00401580", unknown_segbit = "39_3035" }
{ unknown_bit = "004015a7_95_5", unknown_segment = "0x00401580", unknown_segbit = "39_3045" }
{ unknown_bit = "004015a7_95_7", unknown_segment = "0x00401580", unknown_segbit = "39_3047" }
{ unknown_bit = "004015a7_33_11", unknown_segment = "0x00401580", unknown_segbit = "39_1067" }
{ unknown_bit = "004015a7_33_15", unknown_segment = "0x00401580", unknown_segbit = "39_1071" }
{ unknown_bit = "004015a7_98_11", unknown_segment = "0x00401580", unknown_segbit = "39_3147" }
{ unknown_bit = "004015a7_98_15", unknown_segment = "0x00401580", unknown_segbit = "39_3151" }
{ unknown_bit = "004015a7_37_11", unknown_segment = "0x00401580", unknown_segbit = "39_1195" }
{ unknown_bit = "004015a7_37_15", unknown_segment = "0x00401580", unknown_segbit = "39_1199" }
{ unknown_bit = "004015a7_41_11", unknown_segment = "0x00401580", unknown_segbit = "39_1323" }
{ unknown_bit = "004015a7_41_15", unknown_segment = "0x00401580", unknown_segbit = "39_1327" }
[... 8 lines cut ...]
{ unknown_bit = "004015a7_55_5", unknown_segment = "0x00401580", unknown_segbit = "39_1765" }
{ unknown_bit = "004015a7_55_7", unknown_segment = "0x00401580", unknown_segbit = "39_1767" }
{ unknown_bit = "004015a7_58_11", unknown_segment = "0x00401580", unknown_segbit = "39_1867" }
{ unknown_bit = "004015a7_58_15", unknown_segment = "0x00401580", unknown_segbit = "39_1871" }
{ unknown_bit = "004015a7_58_25", unknown_segment = "0x00401580", unknown_segbit = "39_1881" }
{ unknown_bit = "004015a7_58_27", unknown_segment = "0x00401580", unknown_segbit = "39_1883" }
{ unknown_bit = "004015a7_59_5", unknown_segment = "0x00401580", unknown_segbit = "39_1893" }
{ unknown_bit = "004015a7_59_7", unknown_segment = "0x00401580", unknown_segbit = "39_1895" }
{ unknown_bit = "004015a7_62_11", unknown_segment = "0x00401580", unknown_segbit = "39_1995" }
{ unknown_bit = "004015a7_62_15", unknown_segment = "0x00401580", unknown_segbit = "39_1999" }
{ unknown_bit = "004015a7_62_25", unknown_segment = "0x00401580", unknown_segbit = "39_2009" }
{ unknown_bit = "004015a7_62_27", unknown_segment = "0x00401580", unknown_segbit = "39_2011" }
</pre>
are solved by:
<pre>
LIOB33.IOB_Y1.39_5 39_5                                                                                                                      
LIOB33.IOB_Y1.39_7 39_7
LIOB33.IOB_Y1.39_59 39_59
LIOB33.IOB_Y1.39_61 39_61   
LIOB33.IOB_Y1.39_107 39_107                                                                                                                  
LIOB33.IOB_Y1.39_111 39_111                                                                                                                  
LIOB33.IOB_Y1.39_121 39_121                                                                                                                  
LIOB33.IOB_Y1.39_123 39_123                                                                                                                                                                                                                               
</pre>
They seem to be IOB related.

litghost, 2019-07-24:
```
LIOB33.IOB_Y1.39_107 39_107                                                                                                                  
LIOB33.IOB_Y1.39_111 39_111 
```
These are part of SSTL135, which is documented in #970.  Other bits are not documented.

litghost, 2019-07-26:
```
LIOB33.IOB_Y1.39_59 39_59
LIOB33.IOB_Y1.39_61 39_61 
```

These are related to to differential output.

litghost, 2019-07-26:
> ```
> LIOB33.IOB_Y1.39_59 39_59
> LIOB33.IOB_Y1.39_61 39_61 
> ```
> 
> These are related to to differential output.

#983 solves:
```
LIOB33.IOB_Y1.39_59 39_59
LIOB33.IOB_Y1.39_61 39_61 
```

litghost, 2019-07-26:
Remaining bits were related to ``IN_TERM``, solved in #983.

kgugala, 2019-07-30:
The list above is incomplete - the following bits are missing:
```
{ unknown_bit = "004015a7_50_14", unknown_segment = "0x00401580", unknown_segbit = "39_1614" }                                                                                                                                                                                    
{ unknown_bit = "004015a7_50_15", unknown_segment = "0x00401580", unknown_segbit = "39_1615" }                                                                                                                                                                                    
{ unknown_bit = "004015a7_50_16", unknown_segment = "0x00401580", unknown_segbit = "39_1616" }
```

and those are solved by:
```
LIOB33.IOB_Y1.39_174 39_174 
LIOB33.IOB_Y1.39_175 39_175
LIOB33.IOB_Y1.39_176 39_176
```

litghost, 2019-07-30:
All bits in this frame are solved by current or pending PR's.

closed 2019-07-31 by litghost