f4pga/prjxray #965: Bits for base address 0x41580, frame 38

tmichalak, 2019-07-23:
Following 46 bits reported as missing for Litex:
<pre>
{ unknown_bit = "004015a6_2_16", unknown_segment = "0x00401580", unknown_segbit = "38_80" }
{ unknown_bit = "004015a6_2_20", unknown_segment = "0x00401580", unknown_segbit = "38_84" }
{ unknown_bit = "004015a6_67_16", unknown_segment = "0x00401580", unknown_segbit = "38_2160" }
{ unknown_bit = "004015a6_67_20", unknown_segment = "0x00401580", unknown_segbit = "38_2164" }
{ unknown_bit = "004015a6_6_16", unknown_segment = "0x00401580", unknown_segbit = "38_208" }
{ unknown_bit = "004015a6_6_20", unknown_segment = "0x00401580", unknown_segbit = "38_212" }
{ unknown_bit = "004015a6_71_16", unknown_segment = "0x00401580", unknown_segbit = "38_2288" }
{ unknown_bit = "004015a6_71_20", unknown_segment = "0x00401580", unknown_segbit = "38_2292" }
{ unknown_bit = "004015a6_10_16", unknown_segment = "0x00401580", unknown_segbit = "38_336" }
{ unknown_bit = "004015a6_10_20", unknown_segment = "0x00401580", unknown_segbit = "38_340" }
{ unknown_bit = "004015a6_14_16", unknown_segment = "0x00401580", unknown_segbit = "38_464" }
{ unknown_bit = "004015a6_14_20", unknown_segment = "0x00401580", unknown_segbit = "38_468" }
{ unknown_bit = "004015a6_79_16", unknown_segment = "0x00401580", unknown_segbit = "38_2544" }
{ unknown_bit = "004015a6_79_20", unknown_segment = "0x00401580", unknown_segbit = "38_2548" }
{ unknown_bit = "004015a6_18_16", unknown_segment = "0x00401580", unknown_segbit = "38_592" }
{ unknown_bit = "004015a6_18_20", unknown_segment = "0x00401580", unknown_segbit = "38_596" }
{ unknown_bit = "004015a6_83_16", unknown_segment = "0x00401580", unknown_segbit = "38_2672" }
{ unknown_bit = "004015a6_83_20", unknown_segment = "0x00401580", unknown_segbit = "38_2676" }
{ unknown_bit = "004015a6_22_16", unknown_segment = "0x00401580", unknown_segbit = "38_720" }
{ unknown_bit = "004015a6_22_20", unknown_segment = "0x00401580", unknown_segbit = "38_724" }
{ unknown_bit = "004015a6_87_16", unknown_segment = "0x00401580", unknown_segbit = "38_2800" }
{ unknown_bit = "004015a6_87_20", unknown_segment = "0x00401580", unknown_segbit = "38_2804" }
{ unknown_bit = "004015a6_26_16", unknown_segment = "0x00401580", unknown_segbit = "38_848" }
{ unknown_bit = "004015a6_26_20", unknown_segment = "0x00401580", unknown_segbit = "38_852" }
{ unknown_bit = "004015a6_91_16", unknown_segment = "0x00401580", unknown_segbit = "38_2928" }
{ unknown_bit = "004015a6_91_20", unknown_segment = "0x00401580", unknown_segbit = "38_2932" }
{ unknown_bit = "004015a6_30_16", unknown_segment = "0x00401580", unknown_segbit = "38_976" }
{ unknown_bit = "004015a6_30_20", unknown_segment = "0x00401580", unknown_segbit = "38_980" }
{ unknown_bit = "004015a6_95_16", unknown_segment = "0x00401580", unknown_segbit = "38_3056" }
{ unknown_bit = "004015a6_95_20", unknown_segment = "0x00401580", unknown_segbit = "38_3060" }
{ unknown_bit = "004015a6_34_16", unknown_segment = "0x00401580", unknown_segbit = "38_1104" }
{ unknown_bit = "004015a6_34_20", unknown_segment = "0x00401580", unknown_segbit = "38_1108" }
{ unknown_bit = "004015a6_99_16", unknown_segment = "0x00401580", unknown_segbit = "38_3184" }
{ unknown_bit = "004015a6_99_20", unknown_segment = "0x00401580", unknown_segbit = "38_3188" }
{ unknown_bit = "004015a6_38_16", unknown_segment = "0x00401580", unknown_segbit = "38_1232" }
{ unknown_bit = "004015a6_38_20", unknown_segment = "0x00401580", unknown_segbit = "38_1236" }
{ unknown_bit = "004015a6_42_16", unknown_segment = "0x00401580", unknown_segbit = "38_1360" }
{ unknown_bit = "004015a6_42_20", unknown_segment = "0x00401580", unknown_segbit = "38_1364" }
{ unknown_bit = "004015a6_46_16", unknown_segment = "0x00401580", unknown_segbit = "38_1488" }
{ unknown_bit = "004015a6_46_20", unknown_segment = "0x00401580", unknown_segbit = "38_1492" }
{ unknown_bit = "004015a6_55_16", unknown_segment = "0x00401580", unknown_segbit = "38_1776" }
{ unknown_bit = "004015a6_55_20", unknown_segment = "0x00401580", unknown_segbit = "38_1780" }
{ unknown_bit = "004015a6_59_16", unknown_segment = "0x00401580", unknown_segbit = "38_1904" }
{ unknown_bit = "004015a6_59_20", unknown_segment = "0x00401580", unknown_segbit = "38_1908" }
{ unknown_bit = "004015a6_63_16", unknown_segment = "0x00401580", unknown_segbit = "38_2032" }
{ unknown_bit = "004015a6_63_20", unknown_segment = "0x00401580", unknown_segbit = "38_2036" }
</pre>
are solved by:
<pre>
LIOB33.IOB_Y1.38_16 38_16                                                                                                                    
LIOB33.IOB_Y1.38_20 38_20
</pre>
These bits seem to be IOB related.

litghost, 2019-07-23:
These bits are the ``SLEW.FAST`` bits for IOSTANDARD ``SSTL135``.  I'm working on a fuzzer update to 030 to support ``SSTL135``.

tmichalak, 2019-07-24:
@litghost 
There is around 160 bits that can't be solved with the SLEW defined as:
<pre>
RIOB33.IOB_Y0.SLEW.SLOW 38_106 38_110 39_105 39_109
RIOB33.IOB_Y1.SLEW.SLOW 38_18 38_22 39_17 39_21
</pre>
but they solve if we have them separately:
<pre>
RIOB33.IOB_Y0.SLEW.38_106 38_106
RIOB33.IOB_Y0.SLEW.38_110 38_110
RIOB33.IOB_Y0.SLEW.39_105 39_105
RIOB33.IOB_Y0.SLEW.39_109 39_109
RIOB33.IOB_Y1.SLEW.38_18 38_18
RIOB33.IOB_Y1.SLEW.38_22 38_22
RIOB33.IOB_Y1.SLEW.39_17 39_17
RIOB33.IOB_Y1.SLEW.39_21 39_21
</pre>

It seems that the SLEW bits don't work the RIOBs because bit `39_21` is not there:
<pre>
...
RIOB33_X43Y19.IOB_Y1.SLEW.38_18
RIOB33_X43Y19.IOB_Y1.SLEW.38_22
RIOB33_X43Y19.IOB_Y1.SLEW.39_17
...
</pre>

litghost, 2019-07-24:
That is what is said in https://github.com/SymbiFlow/prjxray/issues/965#issuecomment-514235039, anyways this is fixed in #970

tmichalak, 2019-07-26:
@litghost after #970 there are still 6 bits, seems 3 per site that are or used to be part of the solution for the DRIVE, SLEW, IOSTANDARD features:
<pre>
38_04
{ unknown_bit = "004015a6_67_4", unknown_segment = "0x00401580", unknown_segbit = "38_2148" }
{ unknown_bit = "004015a6_71_4", unknown_segment = "0x00401580", unknown_segbit = "38_2276" }
{ unknown_bit = "004015a6_79_4", unknown_segment = "0x00401580", unknown_segbit = "38_2532" }
{ unknown_bit = "004015a6_83_4", unknown_segment = "0x00401580", unknown_segbit = "38_2660" }
{ unknown_bit = "004015a6_91_4", unknown_segment = "0x00401580", unknown_segbit = "38_2916" }
{ unknown_bit = "004015a6_95_4", unknown_segment = "0x00401580", unknown_segbit = "38_3044" }
{ unknown_bit = "004015a6_55_4", unknown_segment = "0x00401580", unknown_segbit = "38_1764" }
{ unknown_bit = "004015a6_59_4", unknown_segment = "0x00401580", unknown_segbit = "38_1892" }

38_06
{ unknown_bit = "004015a6_67_6", unknown_segment = "0x00401580", unknown_segbit = "38_2150" }
{ unknown_bit = "004015a6_71_6", unknown_segment = "0x00401580", unknown_segbit = "38_2278" }
{ unknown_bit = "004015a6_79_6", unknown_segment = "0x00401580", unknown_segbit = "38_2534" }
{ unknown_bit = "004015a6_83_6", unknown_segment = "0x00401580", unknown_segbit = "38_2662" }
{ unknown_bit = "004015a6_91_6", unknown_segment = "0x00401580", unknown_segbit = "38_2918" }
{ unknown_bit = "004015a6_95_6", unknown_segment = "0x00401580", unknown_segbit = "38_3046" }
{ unknown_bit = "004015a6_55_6", unknown_segment = "0x00401580", unknown_segbit = "38_1766" }
{ unknown_bit = "004015a6_59_6", unknown_segment = "0x00401580", unknown_segbit = "38_1894" }

38_120
{ unknown_bit = "004015a6_70_24", unknown_segment = "0x00401580", unknown_segbit = "38_2264" }
{ unknown_bit = "004015a6_78_24", unknown_segment = "0x00401580", unknown_segbit = "38_2520" }
{ unknown_bit = "004015a6_82_24", unknown_segment = "0x00401580", unknown_segbit = "38_2648" }
{ unknown_bit = "004015a6_86_24", unknown_segment = "0x00401580", unknown_segbit = "38_2776" }
{ unknown_bit = "004015a6_94_24", unknown_segment = "0x00401580", unknown_segbit = "38_3032" }
{ unknown_bit = "004015a6_54_24", unknown_segment = "0x00401580", unknown_segbit = "38_1752" }
{ unknown_bit = "004015a6_58_24", unknown_segment = "0x00401580", unknown_segbit = "38_1880" }
{ unknown_bit = "004015a6_62_24", unknown_segment = "0x00401580", unknown_segbit = "38_2008" }

38_122
{ unknown_bit = "004015a6_70_26", unknown_segment = "0x00401580", unknown_segbit = "38_2266" }
{ unknown_bit = "004015a6_78_26", unknown_segment = "0x00401580", unknown_segbit = "38_2522" }
{ unknown_bit = "004015a6_82_26", unknown_segment = "0x00401580", unknown_segbit = "38_2650" }
{ unknown_bit = "004015a6_86_26", unknown_segment = "0x00401580", unknown_segbit = "38_2778" }
{ unknown_bit = "004015a6_94_26", unknown_segment = "0x00401580", unknown_segbit = "38_3034" }
{ unknown_bit = "004015a6_54_26", unknown_segment = "0x00401580", unknown_segbit = "38_1754" }
{ unknown_bit = "004015a6_58_26", unknown_segment = "0x00401580", unknown_segbit = "38_1882" }
{ unknown_bit = "004015a6_62_26", unknown_segment = "0x00401580", unknown_segbit = "38_2010" }

38_14
{ unknown_bit = "004015a6_2_14", unknown_segment = "0x00401580", unknown_segbit = "38_78" } 
{ unknown_bit = "004015a6_67_14", unknown_segment = "0x00401580", unknown_segbit = "38_2158" }
{ unknown_bit = "004015a6_6_14", unknown_segment = "0x00401580", unknown_segbit = "38_206" } 
{ unknown_bit = "004015a6_71_14", unknown_segment = "0x00401580", unknown_segbit = "38_2286" }
{ unknown_bit = "004015a6_10_14", unknown_segment = "0x00401580", unknown_segbit = "38_334" }
{ unknown_bit = "004015a6_14_14", unknown_segment = "0x00401580", unknown_segbit = "38_462" }
{ unknown_bit = "004015a6_79_14", unknown_segment = "0x00401580", unknown_segbit = "38_2542" }
{ unknown_bit = "004015a6_18_14", unknown_segment = "0x00401580", unknown_segbit = "38_590" }
{ unknown_bit = "004015a6_83_14", unknown_segment = "0x00401580", unknown_segbit = "38_2670" }
{ unknown_bit = "004015a6_22_14", unknown_segment = "0x00401580", unknown_segbit = "38_718" }
{ unknown_bit = "004015a6_87_14", unknown_segment = "0x00401580", unknown_segbit = "38_2798" }
{ unknown_bit = "004015a6_26_14", unknown_segment = "0x00401580", unknown_segbit = "38_846" }
{ unknown_bit = "004015a6_91_14", unknown_segment = "0x00401580", unknown_segbit = "38_2926" }
{ unknown_bit = "004015a6_30_14", unknown_segment = "0x00401580", unknown_segbit = "38_974" }
{ unknown_bit = "004015a6_95_14", unknown_segment = "0x00401580", unknown_segbit = "38_3054" }
{ unknown_bit = "004015a6_34_14", unknown_segment = "0x00401580", unknown_segbit = "38_1102" }
{ unknown_bit = "004015a6_99_14", unknown_segment = "0x00401580", unknown_segbit = "38_3182" }
{ unknown_bit = "004015a6_38_14", unknown_segment = "0x00401580", unknown_segbit = "38_1230" }
{ unknown_bit = "004015a6_42_14", unknown_segment = "0x00401580", unknown_segbit = "38_1358" }
{ unknown_bit = "004015a6_46_14", unknown_segment = "0x00401580", unknown_segbit = "38_1486" }
{ unknown_bit = "004015a6_55_14", unknown_segment = "0x00401580", unknown_segbit = "38_1774" }
{ unknown_bit = "004015a6_59_14", unknown_segment = "0x00401580", unknown_segbit = "38_1902" }
{ unknown_bit = "004015a6_63_14", unknown_segment = "0x00401580", unknown_segbit = "38_2030" }

39_113
{ unknown_bit = "004015a7_1_17", unknown_segment = "0x00401580", unknown_segbit = "39_49" }
[... cut ...]
{ unknown_bit = "004015a7_41_17", unknown_segment = "0x00401580", unknown_segbit = "39_1329" }
{ unknown_bit = "004015a7_45_17", unknown_segment = "0x00401580", unknown_segbit = "39_1457" }
{ unknown_bit = "004015a7_49_17", unknown_segment = "0x00401580", unknown_segbit = "39_1585" }
{ unknown_bit = "004015a7_54_17", unknown_segment = "0x00401580", unknown_segbit = "39_1745" }
{ unknown_bit = "004015a7_58_17", unknown_segment = "0x00401580", unknown_segbit = "39_1873" }
{ unknown_bit = "004015a7_62_17", unknown_segment = "0x00401580", unknown_segbit = "39_2001" }
</pre>

litghost, 2019-07-26:
> @litghost after #970 there are still 6 bits, seems 3 per site that are or used to be part of the solution for the DRIVE, SLEW, IOSTANDARD features:

Ya I was debugging that yesterday.  Haven't gotten a bead on it yet.

tmichalak, 2019-07-26:
Closed? So do we have the missing bits or not yet?

litghost, 2019-07-26:
Closed wrong issue.

litghost, 2019-07-26:
Remaining bits were related to ``IN_TERM``, solved in #983.  Only remaining are in #966

litghost, 2019-07-30:
All bits in this frame are solved by current or pending PR's.

PR #970 (2019-07-26): Refactor IOB fuzzer.
 - Add SSTL135
 - Refactor process_rdb to handle varying SLEW by IOSTANDARD.
 - Remove bits that belong to OLOGIC (and are now documented in #971).

Fixes #965
closed 2019-07-31 by litghost