f4pga/prjxray #966: Remaining Litex bits for word 50

tmichalak, 2019-07-23:
The following bits are reported missing for Litex
<pre>
# In frame 0x004015a6 2 bits were not converted.
{ unknown_bit = "004015a6_50_15", unknown_segment = "0x00401580", unknown_segbit = "38_1615" }
{ unknown_bit = "004015a6_50_26", unknown_segment = "0x00401580", unknown_segbit = "38_1626" }

# In frame 0x004015a7 4 bits were not converted.
{ unknown_bit = "004015a7_50_14", unknown_segment = "0x00401580", unknown_segbit = "39_1614" }
{ unknown_bit = "004015a7_50_15", unknown_segment = "0x00401580", unknown_segbit = "39_1615" }
{ unknown_bit = "004015a7_50_16", unknown_segment = "0x00401580", unknown_segbit = "39_1616" }
{ unknown_bit = "004015a7_50_22", unknown_segment = "0x00401580", unknown_segbit = "39_1622" }
</pre>
The base address and word seems to indicate they are in the HCLK_IOI tile, however, this might be somewhere else, although definitely clock tree related.

litghost, 2019-07-23:
As part of updating the IOB fuzzer, I've realized there is an INTERNAL_VREF parameter which is IOBANK wide.  I haven't started looking yet, but is it possible these bits are the INTERNAL_VREF settings?

litghost, 2019-07-26:
I'm gonna try to see if I can solve these.

litghost, 2019-07-26:
Unknown bits likely in the HCLK_IOI:
```
  26_183 (count = 1)
  27_175 (count = 1)
  27_176 (count = 1)
  28_180 (count = 1)
  28_183 (count = 1)
  28_186 (count = 1)
  28_189 (count = 1)
  29_175 (count = 1)
  29_177 (count = 1)
  29_179 (count = 1)
  29_181 (count = 1)
  29_191 (count = 1)
  30_175 (count = 1)
  30_176 (count = 1)
  30_179 (count = 1)
  31_181 (count = 1)
  32_174 (count = 1)
  37_188 (count = 1)
  37_189 (count = 1)
  38_175 (count = 1)
  38_186 (count = 1)
  39_174 (count = 1)
  39_175 (count = 1)
  39_176 (count = 1)
  39_182 (count = 1)
```
As expect, some of the bits were the INTERNAL_VREF, specifically:
```
38_186 39_182
```

But that leaves plenty of unknown bits :(

litghost, 2019-07-26:
Other bits are likely HCLK_IOI pips, which there is no fuzzer for.

litghost, 2019-07-30:
There are still 4 bits left in ``HCLK_IOI`` that are unsolved:
```
Group HCLK_IOI (count = 4):
  38_175 (count = 1)
  39_174 (count = 1)
  39_175 (count = 1)
  39_176 (count = 1)
```

closed 2019-07-31 by litghost