f4pga/prjxray #972: Fuzzers taking +6h to run now?

mithro, 2019-07-24:
It seems like since yesterday the xray builds are taking a *very* long time.
```
2019-07-23T23:31:30.192284 - 054-pip-fan-alt          - 6h39m: + finish
2019-07-23T23:31:30.192284 - 054-pip-fan-alt          - 6h39m: + echo 'Cleaning up temp files'
2019-07-23T23:31:30.192284 - 054-pip-fan-alt          - 6h39m: + rm -f ./tmp.rOoBDE1sT9
2019-07-23T23:31:30.192284 - 054-pip-fan-alt          - 6h39m: + rm -f ./tmp.cS6RVBg60u
2019-07-23T23:31:30.192284 - 054-pip-fan-alt          - 6h39m: + '[' '' = Y ']'
2019-07-23T23:31:30.192284 - 054-pip-fan-alt          - 6h39m: + i=51
2019-07-23T23:31:30.192284 - 054-pip-fan-alt          - 6h39m: + true
2019-07-23T23:31:30.192284 - 054-pip-fan-alt          - 6h39m: + make ITER=51 cleaniter
2019-07-23T23:31:30.192284 - 054-pip-fan-alt          - 6h39m: + make ITER=51 build/todo.txt
2019-07-23T23:31:30.192284 - 054-pip-fan-alt          - 6h39m: + '[' '!' -s build/todo.txt -a 51 -eq 1 ']'
2019-07-23T23:31:30.192284 - 054-pip-fan-alt          - 6h39m: + python3 /tmpfs/src/github/symbiflow-prjxray-presubmit-db-artix7/fuzzers/int_loop_check.py --zero-entries --timeout-iters 50
2019-07-23T23:31:30.192284 - 054-pip-fan-alt          - 6h39m: + '[' -f build/todo/timeout ']'
2019-07-23T23:31:30.192284 - 054-pip-fan-alt          - 6h39m: + echo 'ERROR: timeout'
2019-07-23T23:31:30.192284 - 054-pip-fan-alt          - 6h39m: + exit 1
2019-07-23T23:31:30.192284 - 054-pip-fan-alt          - 6h39m: make[1]: *** [run] Error 1
2019-07-23T23:31:30.192284 - 054-pip-fan-alt          - 6h39m: 
2019-07-23T23:31:30.192284 - 054-pip-fan-alt          - 6h39m: --------------------------------------------------------------------------
2019-07-23T23:31:30.192284 - 054-pip-fan-alt          - 6h39m: Failed @ 2019-07-23T23:31:30.106558 with exit code: 2
2019-07-23T23:31:30.217122 - 054-pip-fan-alt          - 6h39m: WARNING: Fuzzer failed!
Makefile:107: recipe for target '054-pip-fan-alt/run.ok' failed
make: *** [054-pip-fan-alt/run.ok] Error 2
```

mithro, 2019-07-24:
The latest fast build was commit ae526981a278ffbc70b200cd6f26fc47ce171134
```
commit ae526981a278ffbc70b200cd6f26fc47ce171134 (antmicro/master)
Merge: 1752fce b659a16
Author: litghost <537074+litghost@users.noreply.github.com>
Date:   Mon Jul 22 10:04:36 2019 -0700

    Merge pull request #946 from antmicro/idelay-fuzzer
    
    Fuzzer for IDELAY
```

mithro, 2019-07-24:
```
13:38 (4h ago) build #732 2051583e-a033-4712-bcff-5fa63cb8e22b RUNNING
09:34 (8h ago) build #730 97ecb3d6-56a0-40ae-9a68-2473008f118f RUNNING
```

mithro, 2019-07-24:
Looks like pull request #969 and commit 2b93883d78f2dd9e542785257baaf1f37fabbeb5 ran successfully in 199m:49s?

mithro, 2019-07-24:
My guess is that https://github.com/SymbiFlow/prjxray/pull/954 is potentially causing this?

litghost, 2019-07-24:
> My guess is that #954 is potentially causing this?

Incorrect, #954 didn't actually modify any fuzzers.

litghost, 2019-07-24:
This appears to caused by 049 periodically timing out:

Bad run https://source.cloud.google.com/results/invocations/9d5d0280-48cd-4879-a85e-afa0427eea07/targets:
```
2019-07-24T13:03:39.711357 - 049-int-imux-gfan        - 3h35m: Finishing (User:26187s System:1549s).
2019-07-24T13:03:39.956955 - 049-int-imux-gfan        - 3h35m: Failed @ 2019-07-24T13:03:39.711569 with exit code: 2
```

Good run:
https://source.cloud.google.com/results/invocations/76cde16f-9552-46f2-ac17-d5682688ec43/targets

acomodi, 2019-07-24:
`052` presents the same issue:

```
2019-07-24T14:47:37.172523 - 052-pip-clkin            - 5h16m: + python3 /tmpfs/src/github/symbiflow-prjxray-presubmit-db-artix7/fuzzers/int_loop_check.py --zero-entries --timeout-iters 50
2019-07-24T14:47:37.172523 - 052-pip-clkin            - 5h16m: + '[' -f build/todo/timeout ']'
2019-07-24T14:47:37.172523 - 052-pip-clkin            - 5h16m: + echo 'ERROR: timeout'
2019-07-24T14:47:37.172523 - 052-pip-clkin            - 5h16m: + exit 1
2019-07-24T14:47:37.172523 - 052-pip-clkin            - 5h16m: make[1]: *** [run] Error 1
```

tmichalak, 2019-07-26:
@mithro @litghost is this still the case? It seems it resolved itself. I am wondering what could be the root cause here...

litghost, 2019-07-26:
> @mithro @litghost is this still the case? It seems it resolved itself. I am wondering what could be the root cause here...

I've been repeatedly running the continuous to see if it replicates.  So far no luck.

litghost, 2019-07-31:
I don't think we've seen this re-occur. Closing for now.

closed 2019-07-31 by litghost