f4pga/prjxray #982: Bits for base address 0x41580, frame 27

tmichalak, 2019-07-26:
These Litex bits:
<pre>
# In frame 0x0040159b 24 bits were not converted.
{ unknown_bit = "0040159b_53_6", unknown_segment = "0x00401580", unknown_segbit = "27_1702" }
{ unknown_bit = "0040159b_57_6", unknown_segment = "0x00401580", unknown_segbit = "27_1830" }
{ unknown_bit = "0040159b_61_6", unknown_segment = "0x00401580", unknown_segbit = "27_1958" }
{ unknown_bit = "0040159b_69_6", unknown_segment = "0x00401580", unknown_segbit = "27_2214" }
{ unknown_bit = "0040159b_77_6", unknown_segment = "0x00401580", unknown_segbit = "27_2470" }
{ unknown_bit = "0040159b_81_6", unknown_segment = "0x00401580", unknown_segbit = "27_2598" }
{ unknown_bit = "0040159b_85_6", unknown_segment = "0x00401580", unknown_segbit = "27_2726" }
{ unknown_bit = "0040159b_93_6", unknown_segment = "0x00401580", unknown_segbit = "27_2982" }
{ unknown_bit = "0040159b_54_6", unknown_segment = "0x00401580", unknown_segbit = "27_1734" }
{ unknown_bit = "0040159b_58_6", unknown_segment = "0x00401580", unknown_segbit = "27_1862" }
{ unknown_bit = "0040159b_62_6", unknown_segment = "0x00401580", unknown_segbit = "27_1990" }
{ unknown_bit = "0040159b_70_6", unknown_segment = "0x00401580", unknown_segbit = "27_2246" }
{ unknown_bit = "0040159b_78_6", unknown_segment = "0x00401580", unknown_segbit = "27_2502" }
{ unknown_bit = "0040159b_82_6", unknown_segment = "0x00401580", unknown_segbit = "27_2630" }
{ unknown_bit = "0040159b_86_6", unknown_segment = "0x00401580", unknown_segbit = "27_2758" }
{ unknown_bit = "0040159b_94_6", unknown_segment = "0x00401580", unknown_segbit = "27_3014" }
{ unknown_bit = "0040159b_83_18", unknown_segment = "0x00401580", unknown_segbit = "27_2674" }
{ unknown_bit = "0040159b_91_18", unknown_segment = "0x00401580", unknown_segbit = "27_2930" }
{ unknown_bit = "0040159b_95_18", unknown_segment = "0x00401580", unknown_segbit = "27_3058" }
{ unknown_bit = "0040159b_55_18", unknown_segment = "0x00401580", unknown_segbit = "27_1778" }
{ unknown_bit = "0040159b_59_18", unknown_segment = "0x00401580", unknown_segbit = "27_1906" }
{ unknown_bit = "0040159b_67_18", unknown_segment = "0x00401580", unknown_segbit = "27_2162" }
{ unknown_bit = "0040159b_71_18", unknown_segment = "0x00401580", unknown_segbit = "27_2290" }
{ unknown_bit = "0040159b_79_18", unknown_segment = "0x00401580", unknown_segbit = "27_2546" }
</pre>

are solved by 
<pre>
RIOB33.IOB_Y0.ISERDES.27_18 27_18
RIOB33.IOB_Y0.ISERDES.27_70 27_70
RIOB33.IOB_Y0.ISERDES.27_102 27_102
</pre>

tmichalak, 2019-07-26:
`27_70 is associated with ISERDES.IN_USE along with few other bits.
The ISERDES.IN_USE is spread across many of the bits we are currently missing for Litex...

kgugala, 2019-07-30:
`RIOB33.IOB_Y0.ISERDES.27_18 27_18` is solved by #991, but it is not `Y0`: `IOB33.IOB_Y1.ISERDES.DATA_WIDTH.W8 27_18`

litghost, 2019-07-30:
All bits in this frame are solved by current or pending PR's.

closed 2019-07-31 by litghost