f4pga/prjxray #996: Create ISERDES minitest

litghost, 2019-07-30:
A minitest is required for the ISERDES.  The purpose of this minitest is the following:
 - The design should work either a Basys3 or Arty
 - Should be verifiable that it works on hardware (unclear how to do this)

The purpose of this minitest is 3 fold:
 - It should demonstrate that all bits required to use the ISERDESE2 are documented
 - It should be useful in testing that fasm2v supports the ISERDESE2
 - The verilog should be usable by symbiflow-arch-defs to verify ISERDESE2 support

mkurc-ant, 2019-08-14:
> Should be verifiable that it works on hardware (unclear how to do this)

I have an idea for that: We can make a design with a transmitter part and a receiver part. The transmitter will use ordinary logic and output data on some pins. Then there will be a receiver with ISERDES on other pin(s). And we make a external short between those pins with a wire. Can be done on eg. Basys3

litghost, 2019-08-14:
I like it.  The PMod port seems like a good candidate here?

mkurc-ant, 2019-08-14:
Yes, definitely. And a similar thing can be done with OSERDES - transmitt via serdes and receive via logic.

mithro, 2019-08-14:
@mkurc-ant - It would be good to have a real design in the future too (like the big LiteX Linux design). Something based on [liteiclink](https://github.com/enjoy-digital/liteiclink) would be a great test here.

mkurc-ant, 2019-08-22:
https://github.com/SymbiFlow/prjxray/pull/1030

dlharmon, 2019-09-09:
A bidirectional pin using IOBUF can have both a normal logic output and ISERDES attached. When T=0 (output enabled), the ISERDES will see the data output to the pin. This enables you to use any unused IO pin with no hardware jumper required. Same works for OSERDES and normal logic input.

litghost, 2019-09-09:
> A bidirectional pin using IOBUF can have both a normal logic output and ISERDES attached. When T=0 (output enabled), the ISERDES will see the data output to the pin. This enables you to use any unused IO pin with no hardware jumper required. Same works for OSERDES and normal logic input.

Oooooh, that's cute!  I like it.