f4pga/prjxray #1035: Disable tests for documentation related PRs

mgielda, 2019-08-27:
We have a parts-docs label which means a PR is related to documentation. This could potentially be used to skip all or most of the CI when only pushing something into the docs - we'd like to be able to move ahead quickly with the docs and the only way to get something up on RTD is to merge it here.

Documentation changes are highly unlikely to break stuff as long as they don't really touch anything outside the docs themselves. So perhaps filtering on the directory of the files changed (`docs/`) could be a better idea, dunno which is easier to implement.

@mithro offered to look into this, so assigning him.

mithro, 2019-08-27:
Send the pull request for non `master` branch and the tests will not be automatically run.

mgielda, 2019-08-27:
Alright: course of action - we will build docs from a "docs" branch instead, and do merges between that and the master branch. I'll close this issue once I verify that this indeed works.

mgielda, 2019-08-27:
Seems to work, closing!

closed 2019-08-27 by mgielda