f4pga/prjxray #1048: make db-extras-zynq7 fails after a clean build

mithro, 2019-09-12:
```
Writing bitstream ./design.perframecrc.bit...
INFO: [Vivado 12-1842] Bitgen Completed Successfully.
INFO: [Project 1-118] WebTalk data collection is enabled (User setting is ON. Install Setting is ON.).
INFO: [Common 17-186] '/home/tansell/github/SymbiFlow/prjxray/fuzzers/001-part-yaml/build/specimen_001/design/usage_statistics_webtalk.xml' has been successfully sent to Xilinx on Wed Sep 11 17:11:25 2019. For additional details about this file, please refer to the WebTalk help file at /opt/Xilinx/Vivado/2017.2/doc/webtalk_introduction.html.
9 Infos, 1 Warnings, 0 Critical Warnings and 0 Errors encountered.
write_bitstream completed successfully
write_bitstream: Time (s): cpu = 00:00:07 ; elapsed = 00:00:07 . Memory (MB): peak = 2369.625 ; gain = 32.016 ; free physical = 1054 ; free virtual = 34293
# set_property BITSTREAM.GENERAL.PERFRAMECRC NO [current_design]
INFO: [Common 17-206] Exiting Vivado at Wed Sep 11 17:11:25 2019...
/home/tansell/github/SymbiFlow/prjxray/build/tools/gen_part_base_yaml design.perframecrc.bit -f > part.yaml
/bin/sh: /home/tansell/github/SymbiFlow/prjxray/build/tools/gen_part_base_yaml: No such file or directory
make[4]: *** [../../Makefile.specimen:2: part.yaml] Error 127
make[4]: Leaving directory '/home/tansell/github/SymbiFlow/prjxray/fuzzers/001-part-yaml/build/specimen_001'
make[3]: *** [Makefile:11: build/specimen_001] Error 2
make[3]: Leaving directory '/home/tansell/github/SymbiFlow/prjxray/fuzzers/001-part-yaml'
make[2]: *** [Makefile:19: run] Error 2
make[2]: Leaving directory '/home/tansell/github/SymbiFlow/prjxray/fuzzers/001-part-yaml'
make[1]: *** [Makefile:133: part_only] Error 2
make[1]: Leaving directory '/home/tansell/github/SymbiFlow/prjxray/fuzzers'
make: *** [Makefile:137: db-extras-zynq7] Error 2
```

mithro, 2019-09-12:
I think the issue on this one is that I didn't do a `make build` only a `make env`....

tmichalak, 2019-09-20:
@mithro Can this be closed? Or is there still an issue here?

tmichalak, 2019-10-09:
I guess this can be closed. Haven't seen this problem in the CI for a while.

closed 2019-10-09 by tmichalak