f4pga/prjxray #1050: db-extras-zynq7 writing files to wrong path

mithro, 2019-09-12:
https://github.com/SymbiFlow/prjxray/blob/6d21194b569b2e320d7b1c4fe87c8db814a823b1/Makefile#L136-L140

Should be `database/zynq7/harness/` **not** `database/artix7/harness`.

mithro, 2019-09-12:
```patch
diff --git a/Makefile b/Makefile
index 12be9a41..948fc7f0 100644
--- a/Makefile
+++ b/Makefile
@@ -137,7 +137,7 @@ db-extras-zynq7:
        +source minitests/roi_harness/zybo-swbut.sh && $(MAKE) -C fuzzers part_only
        +source minitests/roi_harness/zybo-swbut.sh && \
                $(MAKE) -C minitests/roi_harness \
-                       HARNESS_DIR=database/artix7/harness/zybo/swbut run
+                       HARNESS_DIR=database/zynq7/harness/zybo/swbut run
 
 db-check:
        @true
```

tmichalak, 2019-09-12:
@mithro 
Created a PR with a fix (#1051), however I see you have applied the required changes for Zynq7 manually in the database.

tmichalak, 2019-09-20:
The PR has been merged. The issue has been fixed.

closed 2019-09-20 by tmichalak