f4pga/prjxray #1104: IOSTANDARD to use for PUDC_B PULLUP is unknown

litghost, 2019-10-15:
An unused PUDC_B pin should be configured as `IN_ONLY` and `PULLUP`.  In order to configure `IN_ONLY`, an IOSTANDARD is required.  This appears to be somewhat variable, but it is unclear why certain IOSTANDARDs are used.

- xc7a50tfgg484-1: IOSTANDARD appears to be LVCMOS33
- xc7k70tfbg676-2: IOSTANDARD appears to be LVCMOS15
- xc7z010clg400-1: IOSTANDARD appears to be LVCMOS33

Currently [fasm2frames](https://github.com/SymbiFlow/prjxray/blob/master/utils/fasm2frames.py#L127) always emits the `IN_ONLY` as LVCMOS33 if `--emit_pudc_b_pullup` is set.

One theory to be tested is whether the IOSTANDARD is only variable based on whether the IO bank with the PUDC_B pin is `HR` or `HP`.

JakeMercer, 2019-10-19:
From [UG471](https://www.xilinx.com/support/documentation/user_guides/ug471_7Series_SelectIO.pdf)

> The 7 series FPGAs offer both high-performance (HP) and high-range (HR) I/O banks. The
> HP I/O banks are designed to meet the performance requirements of high-speed memory
> and other chip-to-chip interfaces with voltages up to 1.8V. The HR I/O banks are designed
> to support a wider range of I/O standards with voltages up to 3.3V.

So this would suggest a pin in the HP bank should never be LVCMOS33 or LVCMOS25.

litghost, 2019-10-21:
> So this would suggest a pin in the HP bank should never be LVCMOS33 or LVCMOS25.

Agreed.  The question is whether there are packages that use something other than LVCMOS33 for HR and LVCMOS15 for HP.