f4pga/prjxray #1141: Incorrect PLL IN_USE feature

litghost, 2019-11-12:
https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1150 failed as a result of a bad segbit definition:

```
FAILED: cd /tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7/build/xc7/tests/pll && /usr/bin/cmake -E env PYTHONPATH=/tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7/build/env/conda/lib/python3.7/site-packages:/tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7/third_party/prjxray:/tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7/third_party/prjxray/third_party/fasm /tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7/build/env/conda/bin/python3 /tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7/third_party/prjxray/utils/fasm2frames.py --db-root /tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7/third_party/prjxray-db/artix7 --sparse --emit_pudc_b_pullup /tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7/build/xc7/tests/pll/pll_ext_basys3_x1y0/artix7-xc7a50t-roi-virt-xc7a50t-test/top.fasm /tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7/build/xc7/tests/pll/pll_ext_basys3_x1y0/artix7-xc7a50t-roi-virt-xc7a50t-test/top.frames
Traceback (most recent call last):
  File "/tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7/third_party/prjxray/utils/fasm2frames.py", line 199, in <module>
    main()
  File "/tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7/third_party/prjxray/utils/fasm2frames.py", line 195, in main
    emit_pudc_b_pullup=args.emit_pudc_b_pullup)
  File "/tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7/third_party/prjxray/utils/fasm2frames.py", line 123, in run
    assembler.parse_fasm_filename(filename_in, extra_features=extra_features)
  File "/tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7/third_party/prjxray/prjxray/fasm_assembler.py", line 174, in parse_fasm_filename
    self.add_fasm_line(line, missing_features)
  File "/tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7/third_party/prjxray/prjxray/fasm_assembler.py", line 167, in add_fasm_line
    self.enable_feature(tile, feature, address, line_str)
  File "/tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7/third_party/prjxray/prjxray/fasm_assembler.py", line 129, in enable_feature
    update_segbit(bit)
  File "/tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7/third_party/prjxray/prjxray/fasm_assembler.py", line 115, in update_segbit
    self.frame_clear(frame_addr, word_addr, bit_index, line)
  File "/tmpfs/src/github/symbiflow-arch-defs-presubmit-xc7/third_party/prjxray/prjxray/fasm_assembler.py", line 96, in frame_clear
    self.frames_line[key],
prjxray.fasm_assembler.FasmInconsistentBits: FASM line "CMT_TOP_L_UPPER_T_X106Y44.CMT_TOP_R_UPPER_T_PLLE2_CLKIN2.CMT_TOP_L_UPPER_T_FREQ_BB0" wanted to clear bit (4199709, 75, 8) but was set by FASM line "CMT_TOP_L_UPPER_T_X106Y44.PLLE2.IN_USE"
ninja: build stopped: subcommand failed.
```

litghost, 2019-11-12:
IN_USE:
```
CMT_TOP_L_UPPER_T.PLLE2.IN_USE 28_05 28_10 28_16 28_42 28_46 28_184 28_187 28_190 28_234 28_248 28_251 28_252 28_560 28_590 28_591 28_592 28_595 28_596 28_736 28_753 28_754 28_755 28_756 28_758 28_759 28_760 28_761 28_763 28_764 28_765 28_766 28_767 28_771 28_774 28_775 28_779 28_780 28_783 28_784 28_785 28_786 28_787 28_789 28_790 28_791 28_792 28_794 28_795 28_796 28_797 28_800 28_803 28_806 28_809 28_813 28_815 29_08 29_16 29_45 29_46 29_47 29_236 29_249 29_250 29_251 29_531 29_535 29_557 29_589 29_590 29_593 29_594 29_595 29_753 29_754 29_756 29_758 29_759 29_761 29_765 29_766 29_768 29_777 29_779 29_780 29_781 29_782 29_783 29_784 29_785 29_787 29_791 29_792 29_793 29_794 29_795 29_797 29_798 29_799 29_800 29_803 29_804 29_806 29_807 29_808 29_809 29_810 29_813
```
versus
```
CMT_TOP_L_UPPER_T.CMT_TOP_R_UPPER_T_PLLE2_CLKIN2.CMT_TOP_L_UPPER_T_FREQ_BB0 !28_08 !29_07 !29_08
```

litghost, 2019-11-12:
Looks like `29_08` is in `IN_USE`

mkurc-ant, 2019-11-13:
That's strange. I've runned both `032` and `034` fuzzers locally (current master `65ccddb`) and none of them found the bit `29_08` in the `IN_USE` feature. ('034' solves the `IN_USE` as a byproduct).

Investigating...

mkurc-ant, 2019-11-13:
The solution for `IN_USE` you posted contains additional bits w.r.t. the solution I've got: `28_10` and `29_08`.

Please provide SHA of the prjxray-db you are using and if possible SHA of the prjxray repo used to generate it.

acomodi, 2019-11-13:
Even though this does not regard the bits collision issue, I have seen that, by driving the PLL directly from the IBUF_clk net lets the `CMT_TOP_L_UPPER_T_X106Y44.CMT_TOP_R_UPPER_T_PLLE2_CLKIN2.CMT_TOP_L_UPPER_T_FREQ_BB0` pip to be used.

Instead, if the CLKIN comes from a BUFG, VPR chooses a different path, avoiding the generation of this failure.

mkurc-ant, 2019-11-13:
@acomodi By "IBUF" you mean a regular input buffer, not the dedicated clock input buffer? Because we do not have bits for clock input buffers that are in HCLK_IOI3 tile.

Could you elaborate what is the failure?

acomodi, 2019-11-13:
The failure is the one related to this issue: if the `CMT_TOP_L_UPPER_T_X106Y44.CMT_TOP_R_UPPER_T_PLLE2_CLKIN2.CMT_TOP_L_UPPER_T_FREQ_BB0` pip is chosen to route to one of the clock inputs of the PLL, than there is a conflict as the bits for this pip are cleared and some of the same bits are enabled by the `PLLE_ADV.IN_USE` feature.

As regards the IBUF clk net, i mean the dedicated clock input buffer (the W5 IOPAD).

litghost, 2019-11-13:
> Please provide SHA of the prjxray-db you are using and if possible SHA of the prjxray repo used to generate it.

https://github.com/SymbiFlow/prjxray-db/commit/ca781e739d725938cb5a17b8586a5375ca263ca2
https://github.com/SymbiFlow/prjxray/commit/aa9393b199e34ce8cd8f3671c2b8c51714bf0b70

litghost, 2019-11-13:
prjxray master is still showing `29_08` in the `IN_USE`.

mkurc-ant, 2019-11-13:
Ok, I'm going to run a full build of the db locally (clean build, different machine) and see where the problem is.

mkurc-ant, 2019-11-14:
@litghost https://github.com/SymbiFlow/prjxray/pull/1142 I filtered out conflicting bits in 032.

When I ran the "fasm diff" target for the PLL, the bitstream was decoded to fasm without any error but then Vivado complained that some clock routes are "illegal" and in order to be there require CLOCK_DEDICATED_ROUTE = false.

closed 2019-11-15 by litghost