f4pga/prjxray #1182: Unknown bits for IOBs when using SSTL135

mkurc-ant, 2019-12-16:
Unknown bits were found during running of the `minitest/iostandards/features` test. Didn't make more detailed tests to which mode if IOB  (input/output/inout) those bits apply to. Here are some findings:

Part: `xc7a50tfgg484-1`

For SSTL135 single ended in region X0Y1
```
{ unknown_bit = "00000027_23_7", unknown_segment = "0x00000000", unknown_segbit = "39_743" }
{ unknown_bit = "00000027_76_7", unknown_segment = "0x00000000", unknown_segbit = "39_2439" }
```

For SSTL135 differential SLEW=SLOW in region X1Y0:
```
{ unknown_bit = "004015a6_23_8", unknown_segment = "0x00401580", unknown_segbit = "38_744" }
{ unknown_bit = "004015a6_39_8", unknown_segment = "0x00401580", unknown_segbit = "38_1256" }
{ unknown_bit = "004015a6_47_8", unknown_segment = "0x00401580", unknown_segbit = "38_1512" }
{ unknown_bit = "004015a7_23_9", unknown_segment = "0x00401580", unknown_segbit = "39_745" }
{ unknown_bit = "004015a7_39_9", unknown_segment = "0x00401580", unknown_segbit = "39_1257" }
{ unknown_bit = "004015a7_47_9", unknown_segment = "0x00401580", unknown_segbit = "39_1513" }
```

For SSTL135 differential SLEW=FAST in region X0Y1:
```
{ unknown_bit = "00000026_39_8", unknown_segment = "0x00000000", unknown_segbit = "38_1256" }
{ unknown_bit = "00000026_52_8", unknown_segment = "0x00000000", unknown_segbit = "38_1672" }
{ unknown_bit = "00000026_96_8", unknown_segment = "0x00000000", unknown_segbit = "38_3080" }
{ unknown_bit = "00000027_39_9", unknown_segment = "0x00000000", unknown_segbit = "39_1257" }
{ unknown_bit = "00000027_52_9", unknown_segment = "0x00000000", unknown_segbit = "39_1673" }
{ unknown_bit = "00000027_96_9", unknown_segment = "0x00000000", unknown_segbit = "39_3081" }
```

litghost, 2020-04-20:
I believe this has been fixed?

mkurc-ant, 2020-04-21:
I believe that it is still relevant. These bits pop up when running the `minitest/iostandards/features` for `SSTL135`. I'm surprised that they don't show for eg. LiteX which uses `SSTL135`

mkurc-ant, 2020-05-07:
It appears that once I've made the minitest to set INTERNAL_VREF in https://github.com/SymbiFlow/prjxray/pull/1324 the missing bits disappeared. These bits may not be related to SSTL135 itself but may mean something like "use external VREF" on a given IOB.

closed 2020-05-07 by mkurc-ant