f4pga/prjxray #1183: Possible instability of 047 for Zynq

mkurc-ant, 2019-12-18:
During a CI build for #1181 (`afe279e`) the following error occurred. Changes in that PR were not related to Zynq fuzzers.

```
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m: Traceback (most recent call last):
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m:   File "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-zynq7/utils/bit2fasm.py", line 116, in <module>
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m:     main()
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m:   File "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-zynq7/utils/bit2fasm.py", line 112, in main
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m:     args.db_root, bits_file.name, args.verbose, args.canonical)
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m:   File "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-zynq7/utils/bit2fasm.py", line 41, in bits_to_fasm
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m:     sort_key=grid.tile_key,
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m:   File "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-zynq7/third_party/fasm/fasm/output.py", line 375, in merge_and_sort
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m:     for line in model:
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m:   File "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-zynq7/prjxray/fasm_disassembler.py", line 121, in find_features_in_bitstream
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m:     solved_bitdata, bitdata, verbose=verbose):
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m:   File "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-zynq7/prjxray/fasm_disassembler.py", line 51, in find_features_in_tile
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m:     tile_segbits = self.grid.get_tile_segbits_at_tilename(tile_name)
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m:   File "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-zynq7/prjxray/grid.py", line 138, in get_tile_segbits_at_tilename
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m:     return self.db.get_tile_segbits(gridinfo.tile_type)
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m:   File "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-zynq7/prjxray/db.py", line 161, in get_tile_segbits
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m:     self.tile_types[tile_type.upper()])
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m:   File "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-zynq7/prjxray/tile_segbits.py", line 80, in __init__
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m:     self.segbits[BlockType.CLB_IO_CLK] = read_segbits(f)
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m:   File "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-zynq7/prjxray/tile_segbits.py", line 61, in read_segbits
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m:     assert len(parts) > 1
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m: AssertionError
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m: make[3]: *** [design_bits.ok] Error 1
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m: make[2]: *** [build/16/specimen_031/OK] Error 2
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m: make[2]: *** Waiting for unfinished jobs....
2019-12-17T23:40:22.835467 - 047-hclk-ioi-pips         - 2h59m: make[1]: *** [run] Error 2
```

mkurc-ant, 2019-12-18:
Might also be a fuzzer dependency problem.

tmichalak, 2021-01-26:
Seems outdated, let's reopen if the issue reappears. Closing for now.

closed 2021-01-26 by tmichalak