f4pga/prjxray #1219: 005-tilegrid failing on Zynq

mithro, 2020-02-04:
```
2020-01-30T07:50:57.471229 - 005-tilegrid              - 9m46s: + python3 /tmpfs/src/github/symbiflow-prjxray-continuous-db-zynq7/fuzzers/005-tilegrid/clb_int/../fuzzaddr/generate.py --oneval 0 --design params.csv --dword 1 --dframe 15
2020-01-30T07:50:57.471229 - 005-tilegrid              - 9m46s: ++ fgrep CRITICAL vivado.log
2020-01-30T07:50:57.471229 - 005-tilegrid              - 9m46s: + test -z ''
2020-01-30T07:50:57.471229 - 005-tilegrid              - 9m46s: + for x in 'design*.bit'
2020-01-30T07:50:57.471229 - 005-tilegrid              - 9m46s: + /tmpfs/src/github/symbiflow-prjxray-continuous-db-zynq7/build/tools/bitread --part_file /tmpfs/src/github/symbiflow-prjxray-continuous-db-zynq7/database/zynq7/xc7z020clg484-1/part.yaml -F 0x00000000:0xffffffff -o design.bits -z -y design.bit
2020-01-30T07:50:57.471229 - 005-tilegrid              - 9m46s: + python3 /tmpfs/src/github/symbiflow-prjxray-continuous-db-zynq7/fuzzers/005-tilegrid/clb_int/../fuzzaddr/generate.py --oneval 0 --design params.csv --dword 1 --dframe 15
2020-01-30T07:50:57.471229 - 005-tilegrid              - 9m46s: make[1]: *** [run] Error 2
2020-01-30T07:50:57.471229 - 005-tilegrid              - 9m46s: 
2020-01-30T07:50:57.471229 - 005-tilegrid              - 9m46s: --------------------------------------------------------------------------
2020-01-30T07:50:57.471229 - 005-tilegrid              - 9m46s: Failed @ 2020-01-30T07:50:57.378034 with exit code: 2
2020-01-30T07:50:57.486301 - 005-tilegrid              - 9m46s: WARNING: Fuzzer failed!
Makefile:61: recipe for target '005-tilegrid/run.ok' failed
make: *** [005-tilegrid/run.ok] Error 2
make: *** Waiting for unfinished jobs....
```

mithro, 2020-02-04:
Full logs at 
 - https://console.cloud.google.com/storage/browser/symbiflow-prjxray/artifacts/prod/foss-fpga-tools/prjxray/continuous/database/zynq7/356/20200129-233645
 - https://source.cloud.google.com/results/invocations/9801ce51-9288-457d-8a3b-32cb018f746b

acomodi, 2020-02-04:
There seems to be `segmentation fault` issue: 
```
make[3]: *** [build/specimen_006/OK] Error 139
```

I need to reproduce this locally as well, so to have more information. For sure this is affecting the 7020 part.

litghost, 2020-04-20:
I believe this has been fixed?

tmichalak, 2021-01-26:
Seems to be fixed. Closing.

closed 2021-01-26 by tmichalak