f4pga/prjxray #1232: htmlgen broken with tilegrid.json location change

mithro, 2020-02-13:
```
Loading tilegrid.
Traceback (most recent call last):
  File "/tmp/tmp.5JJ3N7HyOv/prjxray/htmlgen/htmlgen.py", line 1013, in <module>
    main()
  File "/tmp/tmp.5JJ3N7HyOv/prjxray/htmlgen/htmlgen.py", line 1008, in main
    allow_fake=args.allow_fake,
  File "/tmp/tmp.5JJ3N7HyOv/prjxray/htmlgen/htmlgen.py", line 972, in run
    grid = load_tilegrid(db_dir, verbose=verbose, allow_fake=allow_fake)
  File "/tmp/tmp.5JJ3N7HyOv/prjxray/htmlgen/htmlgen.py", line 233, in load_tilegrid
    assert allow_fake, 'No tilegrid.json found'
AssertionError: No tilegrid.json found
The command "./.travis/generate-html.sh" exited with 1.
```

mithro, 2020-02-13:
https://travis-ci.com/SymbiFlow/prjxray-db/builds/148715630

mithro, 2020-02-13:
Hrm - I thought I fixed this in https://github.com/SymbiFlow/prjxray/commit/adc297a560950c403c984cd488bfa1e8e4eceb0b has it moved yet again?

acomodi, 2020-02-13:
@mithro I know what the issue is. The Travis build script tries to generate the html for each settings. Being the artix200t.sh setting already there, it tries to generate the html for it, but the tilegrid is not present actually, until https://github.com/SymbiFlow/prjxray/pull/1227 gets merged

acomodi, 2020-02-13:
@mithro Fix here: https://github.com/SymbiFlow/prjxray/pull/1233

mithro, 2020-05-26:
```
Traceback (most recent call last):
  File "/tmp/tmp.BVba6Gau4q/prjxray/htmlgen/htmlgen.py", line 1018, in <module>
    main()
  File "/tmp/tmp.BVba6Gau4q/prjxray/htmlgen/htmlgen.py", line 993, in main
    os.path.curdir, 'html', get_setting('XRAY_DATABASE')),
  File "/home/travis/virtualenv/python3.6.7/lib/python3.6/posixpath.py", line 94, in join
    genericpath._check_arg_types('join', a, *p)
  File "/home/travis/virtualenv/python3.6.7/lib/python3.6/genericpath.py", line 149, in _check_arg_types
    (funcname, s.__class__.__name__)) from None
TypeError: join() argument must be str or bytes, not 'NoneType'
The command "./.travis/generate-html.sh" exited with 1.
```

mithro, 2020-05-26:
https://travis-ci.com/github/SymbiFlow/prjxray-db

tmichalak, 2021-01-26:
@mithro is this still an issue, or can this be closed?

mithro, 2021-01-26:
The last link shows htmlgen is still broken but possibly for different reasons;
```
Generating for artix7_200t
--------------------------------------------
settings.sh 01881ed8cf11dcd733ef34d904d2aa96bdb77632f7060f67a6daff7af44ec481  /tmp/tmp.kHu5ag7Mxt/prjxray/settings/artix7_200t.sh
# Copyright (C) 2017-2020  The Project X-Ray Authors.
#
# Use of this source code is governed by a ISC-style
# license that can be found in the LICENSE file or at
# https://opensource.org/licenses/ISC
#
# SPDX-License-Identifier: ISC
export XRAY_DATABASE="artix7"
export XRAY_PART="xc7a200tffg1156-1"
export XRAY_ROI_FRAMES="0x00000000:0xffffffff"
# All CLB's in part, all BRAM's in part, all DSP's in part.
# tcl queries IOB => don't bother adding
export XRAY_ROI_TILEGRID="SLICE_X0Y0:SLICE_X163Y249 RAMB18_X0Y0:RAMB18_X8Y99 RAMB36_X0Y0:RAMB36_X8Y49 DSP48_X0Y0:DSP48_X8Y99 IOB_X0Y0:IOB_X1Y249"
export XRAY_EXCLUDE_ROI_TILEGRID=""
# This is used by fuzzers/005-tilegrid/generate_full.py
# (special handling for frame addresses of certain IOIs -- see the script for details).
# This needs to be changed for any new device!
# If you have a FASM mismatch or unknown bits in IOIs, CHECK THIS FIRST.
export XRAY_IOI3_TILES="RIOI3_X105Y9 LIOI3_X0Y9"
source $(dirname ${BASH_SOURCE[0]})/../utils/environment.sh
eval $(python3 ${XRAY_UTILS_DIR}/create_environment.py)
--------------------------------------------
Traceback (most recent call last):
  File "/tmp/tmp.kHu5ag7Mxt/prjxray/htmlgen/htmlgen.py", line 17, in <module>
    from prjxray.util import get_fabric_for_part
ModuleNotFoundError: No module named 'prjxray'
The command "./.travis/generate-html.sh" exited with 1.
```