f4pga/prjxray #1234: IOB bits removed from recent DB update

acomodi, 2020-02-13:
There are some IOB bits which have been removed for the zynq7 part on one of the latest db updates:
https://github.com/SymbiFlow/prjxray-db/commit/4bc6385ab300b1819848371f508185f57b649a0e#diff-3cffc716e2d1b3d5811f59c39bb22de7L14.

This may due to the change of the base part from 7010 to 7020. In fact, while in 7010 the LIOB were absent, in the 7020 there is one clock region containing them.
It is possible that the current fuzzer is not able to correctly solve some of the LIOB bits for lack of the same number of IOB tiles as in the artix7 db.

litghost, 2020-02-13:
I believe that https://github.com/SymbiFlow/prjxray/blob/master/fuzzers/030-iob/generate.py#L31 needs to be updated for the PUDC to the z7020 part.

mkurc-ant, 2020-02-18:
I'll work on that.

mkurc-ant, 2020-02-18:
That should fix this issue: https://github.com/SymbiFlow/prjxray/pull/1245

closed 2020-04-20 by litghost