f4pga/prjxray #1236: New sort_db is broken

litghost, 2020-02-14:
```
  File "../../utils/sort_db.py", line 463, in <module>
    sys.exit(main(sys.argv))
  File "../../utils/sort_db.py", line 458, in main
    sort_dir('.')
  File "../../utils/sort_db.py", line 450, in sort_dir
    sort_file(n)
  File "../../utils/sort_db.py", line 422, in sort_file
    x = sort_json(n)
  File "../../utils/sort_db.py", line 385, in sort_json
    xjson.pprint(f, d)
  File "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-artix7/utils/xjson.py", line 125, in pprint
    data = sort(data)
  File "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-artix7/utils/xjson.py", line 117, in sort
    return rsorter(data)
  File "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-artix7/utils/xjson.py", line 82, in rsorter
    nitems.append((key(k), k, rsorter(v)))
  File "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-artix7/utils/xjson.py", line 108, in rsorter
    nlist.sort(key=lambda n: n[0])
TypeError: unorderable types: tuple() < NoneType()
```

mithro, 2020-02-14:
Any idea what the input to `xjson.py` was there?

litghost, 2020-02-14:
Doesn't matter?

litghost, 2020-02-14:
I'll also point out that sorting ordered constructions (tuples and list) can break the data?

mithro, 2020-02-14:
In Python 3, you can't sort something which has a None object in it...

```
In [1]: ['a', None, '1'].sort()
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-1-46c34c36c41c> in <module>()
----> 1 ['a', None, '1'].sort()

TypeError: '<' not supported between instances of 'NoneType' and 'str'
```

litghost, 2020-02-14:
> In Python 3, you can't sort something which has a None object in it...
>
> ```
> In [1]: ['a', None, '1'].sort()
> --------------------------------------------------------------------------- [...]

And?  What's you point?  Your code is attempting to modified already ordered data?  That's not a valid thing to do?

litghost, 2020-02-14:
To be clear:
`{'a': 1, 'b': 2}` is the same as `{'b': 2, 'a': 1}`, but `[1, 3, 2]` is **not** the same thing as `[1, 2, 3]`.

mithro, 2020-02-14:
The issue is that JSON only has arrays and dicts. In our JSON data structure, almost all our lists (that are not 2 in length) should be sets (and `set([1, 3, 2])` is the same thing as `set([1, 2, 3])`). Are there examples where this is not the case?

litghost, 2020-02-14:
Almost every list in the JSON is ordered.  For example, the timing data is defined as the following tuple: https://github.com/SymbiFlow/prjxray/blob/master/prjxray/timing.py#L162

litghost, 2020-02-14:
> almost all our lists should be sets

That just isn't true?

mithro, 2020-02-14:
Some examples;
 * `tileconn.json`'s top level
 * `tileconn.json: XXX['wire_paris']`
 * `design.json:  XXX['ports']`
 * `design.json:  XXX['ports']['wires_outside_roi']`
 * `site_type_XXXX.json` - No lists
 * `tile_type_XXXX['sites']`

litghost, 2020-02-14:
> Some examples;
> 
> * `tileconn.json`'s top level
> * `tileconn.json: XXX['wire_paris']`
> * `design.json:  XXX['ports']`
> * `design.json:  XXX['ports']['wires_outside_roi']`
> * `site_type_XXXX.json` - No lists
> * `tile_type_XXXX['sites']`

And I already provided a counter example?  You cannot just start reordering ordered data and expect it to work?  Your code was broken.

mithro, 2020-02-14:
We have two options;
 - Special case the ordered data.
 - Special case the unordered data.

The ordered data is currently, lists of wire pairs/coordinates and the timing data? Are there others?

litghost, 2020-02-14:
Or just don't guess and randomly break things?

mithro, 2020-02-14:
Inside the JSON `[]` objects in our current database we current have **both**;
 * (a) data which should be considered ordered, and
 * (b) data which should be considered unordered.

We can thus;
 - (1) Define [] as ordered and convert the currently unordered data into a different structure.
 - (2) Define [] as unordered and convert the currently ordered data into a different structure.

I believe you wish to do (1) -- correct?

litghost, 2020-02-14:
Neither: Define [] as always ordered.  Fuzzers that write [] should have a stable order.

litghost, 2020-02-14:
> * tileconn.json: XXX['wire_paris']

This is an example of ordered data, **not** unordered data.

mithro, 2020-02-14:
The pairs **inside** `XXX['wire_pairs']` are ordered but the actual `XXX['wire_pairs']` list itself is unordered?

litghost, 2020-02-14:
> The pairs **inside** `XXX['wire_pairs']` are ordered but the actual `XXX['wire_pairs']` list itself is unordered?

True

closed 2020-04-20 by litghost