f4pga/prjxray #1246: Instability of tilegrid for xc7z020

acomodi, 2020-02-18:
There are some instabilities regarding the tilegrid fuzzer that should be solved.

For reference, here is an error of a particular run: https://github.com/SymbiFlow/prjxray/pull/1227#issuecomment-587601068

acomodi, 2020-02-25:
There have been periodic instabilities on the zynq7 build. The ones that did show up for now are the following:

Bit collisions:
- https://github.com/SymbiFlow/prjxray/pull/1253#issuecomment-590810303

Fuzzer failures:
- https://github.com/SymbiFlow/prjxray/pull/1227#issuecomment-587601068

The latter one is most probably already solved by increasing the number of specimens.

mkurc-ant, 2020-04-28:
Again, instability: https://github.com/SymbiFlow/prjxray/pull/1307

tcal-x, 2020-04-28:
Also, [#1313](https://github.com/SymbiFlow/prjxray/pull/1313).

mithro, 2020-05-27:
```
Checking zynq7 database
============================
Checking individual .db...
segbits_*.db: 33 okay
mask_*.db: 39 okay
Checking aggregate dir...
ERROR: 1 collisions
  0042229a_044_27: had INT_R_X69Y22.INT_R.IMUX25.LOGIC_OUTS18, got CLBLM_R_X69Y22.CLBLM_R.SLICEL_X1.BLUT.INIT[16]
Traceback (most recent call last):
  File "utils/checkdb.py", line 190, in <module>
    main()
  File "utils/checkdb.py", line 186, in main
    run(args.db_root, args.part, verbose=args.verbose)
  File "utils/checkdb.py", line 172, in run
    check_tile_overlap(db, verbose=verbose)
  File "utils/checkdb.py", line 148, in check_tile_overlap
    raise ValueError("%s collisions" % len(collisions))
ValueError: 1 collisions
Makefile:172: recipe for target 'db-check-zynq7' failed
make: *** [db-check-zynq7] Error 1
```

commit 837e1605 (2020-10-02): Add fix to prevent INT features with 3 bits.

This may help fix #1246.
closed 2020-10-02 by litghost