f4pga/prjxray #1258: Document remaining bits in opentitan bitstream

litghost, 2020-02-28:
```
grep unknown opentitan.fasm  | wc
    557    6127   52651
```

557 bits isn't too many.  Most of these are likely related to the DSP, as 1 DSP is being used:

```
DSP_L_X66Y110.DSP48.DSP_0.AREG_0
DSP_L_X66Y110.DSP48.DSP_0.BREG_0
DSP_L_X66Y110.DSP48.DSP_0.MASK[45:0] = 46'b1111111111111111111111111111111111111111111111
DSP_L_X66Y110.DSP48.DSP_0.ZADREG[0]
DSP_L_X66Y110.DSP48.DSP_0.ZALUMODEREG[0]
DSP_L_X66Y110.DSP48.DSP_0.ZAREG_2_ACASCREG_1
DSP_L_X66Y110.DSP48.DSP_0.ZBREG_2_BCASCREG_1
DSP_L_X66Y110.DSP48.DSP_0.ZCARRYINREG[0]
DSP_L_X66Y110.DSP48.DSP_0.ZCARRYINSELREG[0]
DSP_L_X66Y110.DSP48.DSP_0.ZCREG[0]
DSP_L_X66Y110.DSP48.DSP_0.ZDREG[0]
DSP_L_X66Y110.DSP48.DSP_0.ZINMODEREG[0]
DSP_L_X66Y110.DSP48.DSP_0.ZIS_ALUMODE_INVERTED[3:0] = 4'b1100
DSP_L_X66Y110.DSP48.DSP_0.ZIS_INMODE_INVERTED[4:0] = 5'b11111
DSP_L_X66Y110.DSP48.DSP_0.ZIS_OPMODE_INVERTED[6:0] = 7'b1110101
DSP_L_X66Y110.DSP48.DSP_0.ZMREG[0]
DSP_L_X66Y110.DSP48.DSP_0.ZOPMODEREG[0]
DSP_L_X66Y110.DSP48.DSP_0.ZPREG[0]
```

litghost, 2020-02-28:
Unknown bits:
```
{ unknown_bit = "0042089b_5_19", unknown_segment = "0x00420880", unknown_segbit = "27_179" }
{ unknown_bit = "0042089b_5_20", unknown_segment = "0x00420880", unknown_segbit = "27_180" }
{ unknown_bit = "0042089b_5_21", unknown_segment = "0x00420880", unknown_segbit = "27_181" }
{ unknown_bit = "0042089b_5_24", unknown_segment = "0x00420880", unknown_segbit = "27_184" }
{ unknown_bit = "0044089b_76_19", unknown_segment = "0x00440880", unknown_segbit = "27_2451" }
{ unknown_bit = "0044089b_76_20", unknown_segment = "0x00440880", unknown_segbit = "27_2452" }
{ unknown_bit = "0044089b_76_21", unknown_segment = "0x00440880", unknown_segbit = "27_2453" }
{ unknown_bit = "0044089b_96_19", unknown_segment = "0x00440880", unknown_segbit = "27_3091" }
{ unknown_bit = "0044089b_96_20", unknown_segment = "0x00440880", unknown_segbit = "27_3092" }
{ unknown_bit = "0044089b_76_24", unknown_segment = "0x00440880", unknown_segbit = "27_2456" }
{ unknown_bit = "0044089b_96_21", unknown_segment = "0x00440880", unknown_segbit = "27_3093" }
{ unknown_bit = "0044089b_96_24", unknown_segment = "0x00440880", unknown_segbit = "27_3096" }
{ unknown_bit = "0044089b_56_19", unknown_segment = "0x00440880", unknown_segbit = "27_1811" }
{ unknown_bit = "0044089b_56_20", unknown_segment = "0x00440880", unknown_segbit = "27_1812" }
{ unknown_bit = "0044089b_56_21", unknown_segment = "0x00440880", unknown_segbit = "27_1813" }
{ unknown_bit = "0044089b_56_24", unknown_segment = "0x00440880", unknown_segbit = "27_1816" }
{ unknown_bit = "0044089b_66_19", unknown_segment = "0x00440880", unknown_segbit = "27_2131" }
{ unknown_bit = "0044089b_66_20", unknown_segment = "0x00440880", unknown_segbit = "27_2132" }
{ unknown_bit = "0044089b_66_21", unknown_segment = "0x00440880", unknown_segbit = "27_2133" }
{ unknown_bit = "0044089b_86_19", unknown_segment = "0x00440880", unknown_segbit = "27_2771" }
{ unknown_bit = "0044089b_86_20", unknown_segment = "0x00440880", unknown_segbit = "27_2772" }
{ unknown_bit = "0044089b_66_24", unknown_segment = "0x00440880", unknown_segbit = "27_2136" }
{ unknown_bit = "0044089b_86_21", unknown_segment = "0x00440880", unknown_segbit = "27_2773" }
{ unknown_bit = "0044089b_86_24", unknown_segment = "0x00440880", unknown_segbit = "27_2776" }
{ unknown_bit = "00401b9a_93_14", unknown_segment = "0x00401b80", unknown_segbit = "26_2990" }
{ unknown_bit = "00400e1b_5_19", unknown_segment = "0x00400e00", unknown_segbit = "27_179" }
{ unknown_bit = "00400e1b_5_20", unknown_segment = "0x00400e00", unknown_segbit = "27_180" }
{ unknown_bit = "00400e1b_5_21", unknown_segment = "0x00400e00", unknown_segbit = "27_181" }
{ unknown_bit = "00400e1b_5_24", unknown_segment = "0x00400e00", unknown_segbit = "27_184" }
{ unknown_bit = "00420e1b_5_19", unknown_segment = "0x00420e00", unknown_segbit = "27_179" }
{ unknown_bit = "00420e1b_5_20", unknown_segment = "0x00420e00", unknown_segbit = "27_180" }
{ unknown_bit = "00420e1b_5_21", unknown_segment = "0x00420e00", unknown_segbit = "27_181" }
{ unknown_bit = "00420e1b_5_24", unknown_segment = "0x00420e00", unknown_segbit = "27_184" }
{ unknown_bit = "00420e1b_15_19", unknown_segment = "0x00420e00", unknown_segbit = "27_499" }
{ unknown_bit = "00420e1b_15_20", unknown_segment = "0x00420e00", unknown_segbit = "27_500" }
{ unknown_bit = "00420e1b_15_21", unknown_segment = "0x00420e00", unknown_segbit = "27_501" }
{ unknown_bit = "00420e1b_15_24", unknown_segment = "0x00420e00", unknown_segbit = "27_504" }
{ unknown_bit = "00420e1b_45_19", unknown_segment = "0x00420e00", unknown_segbit = "27_1459" }
{ unknown_bit = "00420e1b_45_20", unknown_segment = "0x00420e00", unknown_segbit = "27_1460" }
{ unknown_bit = "00420e1b_45_21", unknown_segment = "0x00420e00", unknown_segbit = "27_1461" }
{ unknown_bit = "00420e1b_45_24", unknown_segment = "0x00420e00", unknown_segbit = "27_1464" }
{ unknown_bit = "00420e1b_56_19", unknown_segment = "0x00420e00", unknown_segbit = "27_1811" }
{ unknown_bit = "00420e1b_56_20", unknown_segment = "0x00420e00", unknown_segbit = "27_1812" }
{ unknown_bit = "00420e1b_56_21", unknown_segment = "0x00420e00", unknown_segbit = "27_1813" }
{ unknown_bit = "00420e1b_56_24", unknown_segment = "0x00420e00", unknown_segbit = "27_1816" }
{ unknown_bit = "00420e1b_66_19", unknown_segment = "0x00420e00", unknown_segbit = "27_2131" }
{ unknown_bit = "00420e1b_66_20", unknown_segment = "0x00420e00", unknown_segbit = "27_2132" }
{ unknown_bit = "00420e1b_66_21", unknown_segment = "0x00420e00", unknown_segbit = "27_2133" }
{ unknown_bit = "00420e1b_66_24", unknown_segment = "0x00420e00", unknown_segbit = "27_2136" }
{ unknown_bit = "00420e1b_76_19", unknown_segment = "0x00420e00", unknown_segbit = "27_2451" }
{ unknown_bit = "00420e1b_76_20", unknown_segment = "0x00420e00", unknown_segbit = "27_2452" }
{ unknown_bit = "00420e1b_76_21", unknown_segment = "0x00420e00", unknown_segbit = "27_2453" }
{ unknown_bit = "00420e1b_76_24", unknown_segment = "0x00420e00", unknown_segbit = "27_2456" }
{ unknown_bit = "00420e1b_96_19", unknown_segment = "0x00420e00", unknown_segbit = "27_3091" }
{ unknown_bit = "00420e1b_96_20", unknown_segment = "0x00420e00", unknown_segbit = "27_3092" }
{ unknown_bit = "00420e1b_96_21", unknown_segment = "0x00420e00", unknown_segbit = "27_3093" }
{ unknown_bit = "00420e1b_96_24", unknown_segment = "0x00420e00", unknown_segbit = "27_3096" }
{ unknown_bit = "0042199b_25_19", unknown_segment = "0x00421980", unknown_segbit = "27_819" }
[... cut ...]
{ unknown_bit = "00442c1b_86_21", unknown_segment = "0x00442c00", unknown_segbit = "27_2773" }
{ unknown_bit = "00442c1b_86_24", unknown_segment = "0x00442c00", unknown_segbit = "27_2776" }
{ unknown_bit = "00442c1b_96_19", unknown_segment = "0x00442c00", unknown_segbit = "27_3091" }
{ unknown_bit = "00442c1b_96_20", unknown_segment = "0x00442c00", unknown_segbit = "27_3092" }
{ unknown_bit = "00442c1b_96_21", unknown_segment = "0x00442c00", unknown_segbit = "27_3093" }
{ unknown_bit = "00442c1b_96_24", unknown_segment = "0x00442c00", unknown_segbit = "27_3096" }
```

tmichalak, 2021-01-26:
@litghost We would need some steps describing how to recreate the results that are pasted. 
Namelly, what design to run and how to generate the list of unknown bits.
We suspect the missing bits belong to the DSP, but we want people to be able to verify that.

e7p, 2021-04-02:
I recreated this like stated above using the current versions of opentitan as well as prjxray and as of now, only 15 bits are missing:

```
{ unknown_bit = "0042049a_63_21", unknown_segment = "0x00420480", unknown_segbit = "26_2037" }
{ unknown_bit = "0044071a_42_21", unknown_segment = "0x00440700", unknown_segbit = "26_1365" }
{ unknown_bit = "00001b9a_2_14", unknown_segment = "0x00001b80", unknown_segbit = "26_78" }
{ unknown_bit = "00001b9a_3_14", unknown_segment = "0x00001b80", unknown_segbit = "26_110" }
{ unknown_bit = "00401b9a_95_30", unknown_segment = "0x00401b80", unknown_segbit = "26_3070" }
{ unknown_bit = "0040341d_75_11", unknown_segment = "0x00403400", unknown_segbit = "29_2411" }
{ unknown_bit = "0044049a_42_21", unknown_segment = "0x00440480", unknown_segbit = "26_1365" }
{ unknown_bit = "0044049a_47_21", unknown_segment = "0x00440480", unknown_segbit = "26_1525" }
{ unknown_bit = "0044049a_53_21", unknown_segment = "0x00440480", unknown_segbit = "26_1717" }
{ unknown_bit = "0044049a_58_21", unknown_segment = "0x00440480", unknown_segbit = "26_1877" }
{ unknown_bit = "0044049a_78_21", unknown_segment = "0x00440480", unknown_segbit = "26_2517" }
{ unknown_bit = "0044049a_83_21", unknown_segment = "0x00440480", unknown_segbit = "26_2677" }
{ unknown_bit = "0044049a_88_21", unknown_segment = "0x00440480", unknown_segbit = "26_2837" }
{ unknown_bit = "0044049a_93_21", unknown_segment = "0x00440480", unknown_segbit = "26_2997" }
{ unknown_bit = "0044049a_98_21", unknown_segment = "0x00440480", unknown_segbit = "26_3157" }
```