f4pga/prjxray #1264: htmlgen - KeyError: 'XRAY_ROI'

mithro, 2020-03-06:
```
source $(dirname ${BASH_SOURCE[0]})/../utils/environment.sh
--------------------------------------------
Loading tilegrid.
Writing ./html/artix7/index.html
Traceback (most recent call last):
  File "/tmp/tmp.0LNdLtQCmx/prjxray/htmlgen/htmlgen.py", line 1017, in <module>
    main()
  File "/tmp/tmp.0LNdLtQCmx/prjxray/htmlgen/htmlgen.py", line 1012, in main
    allow_fake=args.allow_fake,
  File "/tmp/tmp.0LNdLtQCmx/prjxray/htmlgen/htmlgen.py", line 980, in run
    mk_tilegrid_page(dbstate, output, grid)
  File "/tmp/tmp.0LNdLtQCmx/prjxray/htmlgen/htmlgen.py", line 387, in mk_tilegrid_page
    get_setting("XRAY_PART"), get_setting("XRAY_ROI"),
  File "/tmp/tmp.0LNdLtQCmx/prjxray/htmlgen/htmlgen.py", line 28, in <lambda>
    return lambda name: settings[name]
KeyError: 'XRAY_ROI'
The command "./.travis/generate-html.sh" exited with 1.
```

mithro, 2020-03-06:
https://travis-ci.com/SymbiFlow/prjxray-db/builds/152196782?utm_medium=notification&utm_source=email

mithro, 2020-03-31:
This is still happening on head - https://travis-ci.com/github/SymbiFlow/prjxray-db/builds/157300050?utm_medium=notification&utm_source=email
```
source $(dirname ${BASH_SOURCE[0]})/../utils/environment.sh
--------------------------------------------
Loading tilegrid.
Writing ./html/artix7/index.html
Traceback (most recent call last):
  File "/tmp/tmp.nNvTwaA29R/prjxray/htmlgen/htmlgen.py", line 1017, in <module>
    main()
  File "/tmp/tmp.nNvTwaA29R/prjxray/htmlgen/htmlgen.py", line 1012, in main
    allow_fake=args.allow_fake,
  File "/tmp/tmp.nNvTwaA29R/prjxray/htmlgen/htmlgen.py", line 980, in run
    mk_tilegrid_page(dbstate, output, grid)
  File "/tmp/tmp.nNvTwaA29R/prjxray/htmlgen/htmlgen.py", line 387, in mk_tilegrid_page
    get_setting("XRAY_PART"), get_setting("XRAY_ROI"),
  File "/tmp/tmp.nNvTwaA29R/prjxray/htmlgen/htmlgen.py", line 28, in <lambda>
    return lambda name: settings[name]
KeyError: 'XRAY_ROI'
The command "./.travis/generate-html.sh" exited with 1.
```

acomodi, 2020-04-01:
@mithro Right, lost track of this issue. I'll look into it and fix it

closed 2020-04-06 by mithro