f4pga/prjxray #1285: Missing INIT string data in FASM for selected BRAM sites

nelsobe, 2020-03-30:
When generating a large memory in Verilog, in some cases, parts of the BRAM INIT information is missing when the .bit file is converted to FASM.

Here is an example (memory was initialized to all 1's in Verilog, BRAM INIT string should be 256 1's but some get left off):

BRAM_R_X37Y35.RAMB18_Y0.INIT_3F[127:0] = 128'b11111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111

There are 3 other such INIT strings in this design out of many that have this issue.  This is for the xc7a50 part.

The common thread is the location of the particular BRAM cells -
X37Y35 and X37Y40 are the only locations identified thus far where the information drops occur.  Have run 100's of other test cases which map BRAMs to other locations without problems (cannot say have tried every other single site, however).

Additional info:
- This memory uses 32 RAMB36's.  Only 2 of the 32 have this problem.
- Always is at BRAM_R_X37Y35 and BRAM_R_X47Y40. Other locations in that column are just fine.

To re-create:
1. Generate a bitstream for the attached .sv code for the Artix-7 50T part.

2. Run $XRAY_BIT2FASM on the resulting .bit file.

Here is the test code:

[128kb8.zip](https://github.com/SymbiFlow/prjxray/files/4405050/128kb8.zip)

litghost, 2020-03-30:
If you run $XRAY_BIT2FASM --verbose, and grep the output for "unknown", do you find anything?

nelsobe, 2020-03-30:
Yes, lots of such messages.  I just dumped everything it output to a file, it is attached.   
[128kb8.txt](https://github.com/SymbiFlow/prjxray/files/4405115/128kb8.txt)

nelsobe, 2020-04-17:
This issue (#1285) suggests that there are some holes in the prjxray bitstream db as it relates to BRAMs.  Interestingly, it only occurs for _some_ BRAMs in one column of the device - the bitstream is correct for others.  

Would like to track down and propose a fix for this, and have some new students coming on-line soon to help.  A bit of direction from the experts out there could be helpful to jump start this.

mithro, 2020-04-17:
@litghost, @nelsobe could use some advice on the best way to look into this further.

litghost, 2020-04-17:
@nelsobe Have you confirmed that the `INIT_xx` does from the FASM do **not** match the `INIT_xx` parameter from the relevant RAMBx cells in Vivado?

litghost, 2020-04-17:
To be clear, BRAM36's are exposed in FASM as 2 BRAM18's.  If you repeat the design using only BRAM18's, do you observe the same error?  How confident is your mapping between the BRAM36 and 2x BRAM18's?

nelsobe, 2020-04-17:
Yes, we have verified that the INIT in Vivado is different from the INIT in the resulting FASM (and that it gives a bunch of "unknown" messagses when it creates the FASM).

Here is some background info so maybe you can help us zero in on where in the code base things are going wrong.  The relevant files are above in the initial comment for this issue up above if you want to reproduce.

The design consists of a single memory written in behavioral Verilog.  Vivado synthesizes that to a bitstream (device is Artix-7 50T part).  That is the entire processing chain, nothing unusual about it that I can see.

When $XRAY_BIT2FASM --verbose is run on the resulting bitstream it ends with errors about unknown bits and unknown segments.  

100's of tests have been run with various sizes and shapes of memories.  The result shows that the bit2fasm errors only occur for BRAMs in specific locations on the chip - they are always for BRAMs in the same couple of locations. For memories that are mapped to different locations there are never any errors.

An error in the bitstream database for this part could be an explanation but I would assume there may be other explanations. We are new enough to the system that I don't know where to point my students.  Any suggestions appreciated.  However, I assume it is a bug that ought to be found and fixed in the tools and we are happy to dive in.

Here are the errors at the bottom of the fasm file when --verbose is included :

# In frame 0x0040129b 8 bits were not converted.
{ unknown_bit = "0040129b_76_19", unknown_segment = "0x00401280", unknown_segbit = "27_2451" }
{ unknown_bit = "0040129b_76_20", unknown_segment = "0x00401280", unknown_segbit = "27_2452" }
{ unknown_bit = "0040129b_76_21", unknown_segment = "0x00401280", unknown_segbit = "27_2453" }
{ unknown_bit = "0040129b_76_24", unknown_segment = "0x00401280", unknown_segbit = "27_2456" }
{ unknown_bit = "0040129b_86_19", unknown_segment = "0x00401280", unknown_segbit = "27_2771" }
{ unknown_bit = "0040129b_86_20", unknown_segment = "0x00401280", unknown_segbit = "27_2772" }
{ unknown_bit = "0040129b_86_21", unknown_segment = "0x00401280", unknown_segbit = "27_2773" }
{ unknown_bit = "0040129b_86_24", unknown_segment = "0x00401280", unknown_segbit = "27_2776" }

# In frame 0x0040031b 40 bits were not converted.
{ unknown_bit = "0040031b_56_19", unknown_segment = "0x00400300", unknown_segbit = "27_1811" }
{ unknown_bit = "0040031b_56_20", unknown_segment = "0x00400300", unknown_segbit = "27_1812" }
{ unknown_bit = "0040031b_56_21", unknown_segment = "0x00400300", unknown_segbit = "27_1813" }
{ unknown_bit = "0040031b_96_19", unknown_segment = "0x00400300", unknown_segbit = "27_3091" }
{ unknown_bit = "0040031b_96_20", unknown_segment = "0x00400300", unknown_segbit = "27_3092" }
{ unknown_bit = "0040031b_56_24", unknown_segment = "0x00400300", unknown_segbit = "27_1816" }
{ unknown_bit = "0040031b_96_21", unknown_segment = "0x00400300", unknown_segbit = "27_3093" }
{ unknown_bit = "0040031b_96_24", unknown_segment = "0x00400300", unknown_segbit = "27_3096" }
{ unknown_bit = "0040031b_25_19", unknown_segment = "0x00400300", unknown_segbit = "27_819" }
{ unknown_bit = "0040031b_25_20", unknown_segment = "0x00400300", unknown_segbit = "27_820" }
{ unknown_bit = "0040031b_25_21", unknown_segment = "0x00400300", unknown_segbit = "27_821" }
{ unknown_bit = "0040031b_25_24", unknown_segment = "0x00400300", unknown_segbit = "27_824" }
{ unknown_bit = "0040031b_66_19", unknown_segment = "0x00400300", unknown_segbit = "27_2131" }
{ unknown_bit = "0040031b_66_20", unknown_segment = "0x00400300", unknown_segbit = "27_2132" }
{ unknown_bit = "0040031b_66_21", unknown_segment = "0x00400300", unknown_segbit = "27_2133" }
{ unknown_bit = "0040031b_66_24", unknown_segment = "0x00400300", unknown_segbit = "27_2136" }
{ unknown_bit = "0040031b_35_19", unknown_segment = "0x00400300", unknown_segbit = "27_1139" }
{ unknown_bit = "0040031b_35_20", unknown_segment = "0x00400300", unknown_segbit = "27_1140" }
{ unknown_bit = "0040031b_35_21", unknown_segment = "0x00400300", unknown_segbit = "27_1141" }
{ unknown_bit = "0040031b_35_24", unknown_segment = "0x00400300", unknown_segbit = "27_1144" }
{ unknown_bit = "0040031b_76_19", unknown_segment = "0x00400300", unknown_segbit = "27_2451" }
{ unknown_bit = "0040031b_76_20", unknown_segment = "0x00400300", unknown_segbit = "27_2452" }
{ unknown_bit = "0040031b_76_21", unknown_segment = "0x00400300", unknown_segbit = "27_2453" }
{ unknown_bit = "0040031b_76_24", unknown_segment = "0x00400300", unknown_segbit = "27_2456" }
{ unknown_bit = "0040031b_5_19", unknown_segment = "0x00400300", unknown_segbit = "27_179" }
{ unknown_bit = "0040031b_5_20", unknown_segment = "0x00400300", unknown_segbit = "27_180" }
{ unknown_bit = "0040031b_5_21", unknown_segment = "0x00400300", unknown_segbit = "27_181" }
{ unknown_bit = "0040031b_45_19", unknown_segment = "0x00400300", unknown_segbit = "27_1459" }
{ unknown_bit = "0040031b_45_20", unknown_segment = "0x00400300", unknown_segbit = "27_1460" }
{ unknown_bit = "0040031b_5_24", unknown_segment = "0x00400300", unknown_segbit = "27_184" }
{ unknown_bit = "0040031b_45_21", unknown_segment = "0x00400300", unknown_segbit = "27_1461" }
{ unknown_bit = "0040031b_45_24", unknown_segment = "0x00400300", unknown_segbit = "27_1464" }
{ unknown_bit = "0040031b_86_19", unknown_segment = "0x00400300", unknown_segbit = "27_2771" }
[... cut ...]
{ unknown_bit = "0000031b_66_21", unknown_segment = "0x00000300", unknown_segbit = "27_2133" }
{ unknown_bit = "0000031b_66_24", unknown_segment = "0x00000300", unknown_segbit = "27_2136" }
{ unknown_bit = "0000031b_76_19", unknown_segment = "0x00000300", unknown_segbit = "27_2451" }
{ unknown_bit = "0000031b_76_20", unknown_segment = "0x00000300", unknown_segbit = "27_2452" }
{ unknown_bit = "0000031b_76_21", unknown_segment = "0x00000300", unknown_segbit = "27_2453" }
{ unknown_bit = "0000031b_76_24", unknown_segment = "0x00000300", unknown_segbit = "27_2456" }

litghost, 2020-04-17:
I need the DCP from your replication, or your exact TCL script and Vivado version used.

litghost, 2020-04-17:
Tile `BRAM_R_X47Y40` is not part of the 50T fabric?  Exactly which part are you testing on?

litghost, 2020-04-17:
I assume `BRAM_R_X47Y40` is a typo, and you meant `BRAM_R_X37Y40`?

nelsobe, 2020-04-17:
Yes, the 47 was a typo, they are both in X37.

Files attached, including .tcl file which generates .bit file (also included).   I am running Vivado 2017.2 on Ubuntu 16.04.

[128kb8.zip](https://github.com/SymbiFlow/prjxray/files/4495501/128kb8.zip)

litghost, 2020-04-17:
The problems appears to soley about *all* BRAM_R tiles. BRAM_L tiles operate per the database.  As a short term solution, you can set prohibit on sites located within BRAM_R tiles, something akin:
```
foreach site [get_sites -of [get_tiles -filter {TYPE == BRAM_R}]] {
  set_property PROHIBIT true $site
}
```

nelsobe, 2020-04-17:
Thanks.  So, are you saying the BRAM_R are not supported yet?  Or, that there is a bug in their handling?  

A curious thing, however, is that from our experiments it is clear that the brams at BRAM_R_X37Y50 and BRAM_R_X37Y55 work just fine.  That is, the resulting FASM file has exactly the same INIT contents as Vivado does.  This goes for random data as well as all 1's data.  That may be due to luck in how the frames line up or something else in the code, but it is true nonetheless.  That is why I only originally reported it for just two of the locations.  I haven't done an exhaustive test for all BRAM_R locations on the chip, but I do know that some work (X37Y50 and X37Y55) and some do not (X37Y35 and X37Y40).

litghost, 2020-04-17:
At a minimum, the top 128 bits INIT_3F of BRAM_R (and the top bits of the high INITP) appear to have problems.  If I had to guess, the problem lies in bitread, but I don't have hard data yet.

litghost, 2020-04-17:
> So, are you saying the BRAM_R are not supported yet?

If you can avoid needing to set INIT_3F[255:128] or INITP_07[255:128], then I expect things work as expected.

nelsobe, 2020-04-18:
That is exactly what we noticed - always the top half of INIT_3F was all 0's.

litghost, 2020-04-20:
@tmichalak I believe that bitread is missing some bits at the end of the bitstream.

Replication, using Vivado's rbt output (with the `-raw_bitfile` flag):
```
# diff build_xc7a50tfgg484-1/specimen_001/design.rbt build_xc7a50tfgg484-1/specimen_002/design.rbt 
6c6
< Date:        	Fri Apr 17 16:51:42 2020
---
> Date:        	Fri Apr 17 16:51:41 2020
573773c573773
< 00000000000000000000000000000000
---
> 00000000000000001000000000000000
573873c573873
< 00100010000001110101000101010000
---
> 11111000101000010111101101110000
```

So the RBT version of the bitstream shows two differences.  The first change in the high bit of `INIT_3F` changing from a 0 to a 1.  The second change is the CRC changes, so that can be ignored.

However the `bits` file generate from bitread doesn't show any differences at all!:

```
diff build_xc7a50tfgg484-1/specimen_001/design.bits build_xc7a50tfgg484-1/specimen_002/design.bits
```

Is this a latent bug in bitread or an error in the `part.yaml` file?

tmichalak, 2020-04-20:
The `part.yaml` is fine. This is definitely a bitread bug - seems to be an off-by-one kind.

KKtiandao, 2022-05-07:
@litghost Has this bug been resolved now?