f4pga/prjxray #1346: Collision between INT_R_X69Y22.INT_R.IMUX25.LOGIC_OUTS18, got CLBLM_R_X69Y22.CLBLM_R.SLICEL_X1.BLUT.INIT[16] on Zynq

mithro, 2020-05-27:
```
Checking zynq7 database
============================
Checking individual .db...
segbits_*.db: 33 okay
mask_*.db: 39 okay
Checking aggregate dir...
ERROR: 1 collisions
  0042229a_044_27: had INT_R_X69Y22.INT_R.IMUX25.LOGIC_OUTS18, got CLBLM_R_X69Y22.CLBLM_R.SLICEL_X1.BLUT.INIT[16]
Traceback (most recent call last):
  File "utils/checkdb.py", line 190, in <module>
    main()
  File "utils/checkdb.py", line 186, in main
    run(args.db_root, args.part, verbose=args.verbose)
  File "utils/checkdb.py", line 172, in run
    check_tile_overlap(db, verbose=verbose)
  File "utils/checkdb.py", line 148, in check_tile_overlap
    raise ValueError("%s collisions" % len(collisions))
ValueError: 1 collisions
Makefile:172: recipe for target 'db-check-zynq7' failed
make: *** [db-check-zynq7] Error 1
```

mithro, 2020-05-27:
This seem to start happening after 1f410829371e774e0cfa0e01bd3c46d85dbb49b6

mithro, 2020-05-27:
Actually, it seems a long running bug?

litghost, 2020-05-27:
This issue is a dup: https://github.com/SymbiFlow/prjxray/issues/1246

closed 2020-05-27 by litghost