f4pga/prjxray #1384: Export fasm2frames and bitstream generation in a separate package

acomodi, 2020-07-02:
### Problem statement

The current implementation of fasm2frames and frames2bit does not allow for an easy portability of the bitstream generation step.

There are two main issues:
1. fasm2frames needs to be installed via pip, and it is currently located in the prjxray/utils directory. Given that this is a relevant step in the SymbiFlow toolchain, it should deserve a standalone location.
2. there is no direct fasm2bit tool as the only tool to write bitstreams is frames2bit and it is currently written in C++. For an end-user point of view, the frames file is actually not needed apart from debugging reasons, therefore the fasm2frames and frames2bit steps could be merged in one step only, also making it easier to perform the bitstream generation step. 
   - The problem is that we have only a C++ implementation of the frames2bit tool. Given that most of the runtime spent by the tool should be a Disk I/O time, it would be convenient to port everything to python and easily merge the fasm2frames and frames2bit step.

### Goal

Have a standalone bitstream generation tool that, taken a fasm file as input, generates a bitstream, all wrapped in one command only.
This would also be compatible with the way QuickLogic bitstream generation is performed.

### Solution

This goal can be achieved in two separate steps:
1. Extract the fasm2frames script into a separate repository (e.g. xc7-fasm)  with all the relative tests and add a step that calls frames2bit as a subprocess. (This requires having frames2bit binary on the local machine).
   - All the tools that rely on fasm2frames will just install it via pip.
2. Convert frames2bit into python and integrate that into the existing fasm2frames code.

acomodi, 2020-07-02:
@litghost @mithro @kgugala FYI

kgugala, 2020-07-02:
the first step (from the proposed ones) seems to relatively easy to do and will simplify the flow a bit.
Reimplementing frames2bit into python will remove the necessity of writing and reading back the frames file.

We have a pure python bitstream writer (fasm -> bit) in the QuickLogic flow.

acomodi, 2020-07-06:
I have created a repository here https://github.com/SymbiFlow/xc-fasm.

What is needed is a wrapper around fasm2frames and frames2bit, so we can have a single command invocation to get from fasm to bit

litghost, 2020-09-30:
This has been done, yes?

acomodi, 2020-10-01:
Yes, this is done. Closing

closed 2020-10-01 by acomodi