f4pga/prjxray #1402: Zybo and basys harnesses do not contain explicit type

andrewb1999, 2020-07-23:
https://github.com/SymbiFlow/prjxray-db/blob/master/zynq7/harness/zybo/swbut/design.json#L13

Code was added in #1383 to add an explicit type to the harness.  Not sure why this only adds the type for arty harnesses and not basys3 or zybo.

Needs to be fixed for https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1565 now that pin map csv generation has been merged.

litghost, 2020-07-23:
https://github.com/mithro/prjxray-db/blob/002032fa463ab11026afde28c7a91e8a2156bb10/artix7/harness/basys3/swbut/design.json#L13 looks good?  The basys3 50 MHz is old and can be deleted.  Let me check if it is used anywhere.

I believe the story is the same for the zybo harness?

litghost, 2020-07-23:
@mithro Can you delete https://github.com/SymbiFlow/prjxray-db/tree/master/zynq7/harness/zybo/swbut and https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/harness/basys3/swbut_50

If you believe they should still exist, we need to update prjxray to regenerate those harness, as they are currently very old.

mithro, 2020-07-23:
Shouldn't these be built by https://github.com/SymbiFlow/prjxray/blob/master/Makefile#L197-L231 ?

I think we might want to keep the zybo one, but the `swbut_50` should be deleted as there is a normal `swbut` right?

andrewb1999, 2020-07-23:
> https://github.com/mithro/prjxray-db/blob/002032fa463ab11026afde28c7a91e8a2156bb10/artix7/harness/basys3/swbut/design.json#L13 looks good? The basys3 50 MHz is old and can be deleted. Let me check if it is used anywhere.
> 
> I believe the story is the same for the zybo harness?

Yes that basys3 harness looks good.

litghost, 2020-07-23:
> Shouldn't these be built by https://github.com/SymbiFlow/prjxray/blob/master/Makefile#L197-L231 ?

The zybo and swbut_50 harnesses are not specified there

> I think we might want to keep the zybo one, but the `swbut_50` should be deleted as there is a normal `swbut` right?

Pretty much.  The zybo harness never got incorporated into arch-defs, right now we only have the full graph supported.

andrewb1999, 2020-07-24:
This is no longer required for https://github.com/SymbiFlow/symbiflow-arch-defs/pull/1565, but is probably still a good change to make.

litghost, 2020-12-14:
Fixed in https://github.com/SymbiFlow/prjxray-db/pull/13

closed 2020-12-14 by litghost