f4pga/prjxray #1476: Add part data for all parts

litghost, 2020-10-29:
Once #1475 is complete, populate part data for all parts that fabric documentation exists for.  #1475 should be done first to avoid duplicating all of the fabric data repeatedly.

dnltz, 2021-01-14:
@litghost - Before I will start I have some questions:
1) Where do I know the fabric documentation exists? Because the script in settings/ exists?
2) Where do I know devices share the same fabric?
3) Any restrictions on the parts or do you want all available packages, devices and speed grades?

litghost, 2021-01-14:
I would start with a family, e.g. "artix7".  Example:
```
get_parts -filter "FAMILY == artix7"
```

litghost, 2021-01-14:
> * Where do I know devices share the same fabric?

I think the default is each device is a different fabric, and add a file that specifies which devices are the same fabric.  Does that make sense?

dnltz, 2021-01-14:
Ah, that's a brilliant way to fetch all available parts, thanks!

Okay. I understood different devices share the same fabric as well. Makes absolutely sense to me now. That's also the reason for: https://github.com/SymbiFlow/prjxray/pull/1549

dnltz, 2021-01-16:
hey @litghost, I wrote a script to populate all supported parts. Should I also extend the Makefile to roi_only all parts during the CI runs?

litghost, 2021-01-17:
> hey @litghost, I wrote a script to populate all supported parts. Should I also extend the Makefile to roi_only all parts during the CI runs?

Yep, exactly!

PR #1562 (2021-03-18): Populate all parts
These changes allow to easily generate all part information during a full db run.

closes #1476
closed 2021-03-18 by litghost