f4pga/prjxray #1478: MMCM bit conflitct

mkurc-ant, 2020-10-30:
In the current database FASM features:
```
CMT_TOP_L_LOWER_B.CMT_LR_LOWER_B_MMCM_CLKIN1.CMT_L_LOWER_B_CLK_FREQ_BB0
CMT_TOP_L_LOWER_B.CMT_LR_LOWER_B_MMCM_CLKIN1.CMT_L_LOWER_B_CLK_FREQ_BB1
CMT_TOP_L_LOWER_B.CMT_LR_LOWER_B_MMCM_CLKIN1.CMT_L_LOWER_B_CLK_FREQ_BB2
CMT_TOP_L_LOWER_B.CMT_LR_LOWER_B_MMCM_CLKIN1.CMT_L_LOWER_B_CLK_FREQ_BB3
```
and
```
CMT_TOP_L_LOWER_B.CMT_LR_LOWER_B_MMCM_CLKIN2.CMT_L_LOWER_B_CLK_FREQ_BB0
CMT_TOP_L_LOWER_B.CMT_LR_LOWER_B_MMCM_CLKIN2.CMT_L_LOWER_B_CLK_FREQ_BB1
CMT_TOP_L_LOWER_B.CMT_LR_LOWER_B_MMCM_CLKIN2.CMT_L_LOWER_B_CLK_FREQ_BB2
CMT_TOP_L_LOWER_B.CMT_LR_LOWER_B_MMCM_CLKIN2.CMT_L_LOWER_B_CLK_FREQ_BB3
```
require clearing of some bits that belong to the `CMT_TOP_L_LOWER_B.MMCME2.IN_USE` feature.

litghost, 2020-10-30:
The MMCM pip fuzzer needed to be adjusted to get all of the FREQ_BBx features, and the MMCM feature fuzzer did not get those changes.  I believe the easiest fix here is to the get the IN_USE feature from the MMCM pip fuzzer.

mkurc-ant, 2020-11-03:
Should solve this issue: https://github.com/SymbiFlow/prjxray/pull/1483

closed 2020-11-04 by litghost