f4pga/prjxray #1485: where the ignored wires in 074 are from

the-centry, 2020-11-05:
if I want to fuzzer a new device such as k7 325t,how could I get the ignored wires in 074?

tcal-x, 2020-11-05:
Hi @the-centry , I have notes from the process that I performed for a7 100t that may be useful; see under "Step 4":

https://github.com/tcal-x/prjxray/blob/doc/NEWPART.md#step-4

Compare your ignored-wire results with existing ignored-wires files to make sure it seems reasonable, or even post it here for me or @litghost or someone else to take a look at.

Thanks for your contribution!

mithro, 2020-11-05:
@tcal-x - Any reason this documentation hasn't been added to the upstream prjxray repo? I don't think anyone would likely think to look in your fork to find info like this :-)

tcal-x, 2020-11-05:
Yeah, let me address the remaining "FIXME"s and send a PR.

the-centry, 2020-11-09:
> Yeah, let me address the remaining "FIXME"s and send a PR.

thinks so much!

closed 2021-02-26 by the-centry