f4pga/prjxray #1521: make env fails in symbiflow-arch-pkgs (invalid command 'bdist_wheel') on Ubuntu 20.04.1 LTS

hansfbaier, 2020-12-17:
```
~/HDL/symbiflow-arch-defs/third_party/prjxray$ make env
python3 -mvenv env
# Install project dependencies
if [ -e env/bin/activate ]; then . env/bin/activate; fi; source utils/environment.python.sh; python -mpip install -r requirements.txt
Obtaining file:///home/jack/HDL/symbiflow-arch-defs/third_party/prjxray/third_party/fasm (from -r requirements.txt (line 1))
Obtaining file:///home/jack/HDL/symbiflow-arch-defs/third_party/prjxray/third_party/python-sdf-timing (from -r requirements.txt (line 2))
Obtaining file:///home/jack/HDL/symbiflow-arch-defs/third_party/prjxray (from -r requirements.txt (line 3))
Requirement already satisfied: intervaltree in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 4)) (3.1.0)
Requirement already satisfied: junit-xml in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 5)) (1.9)
Requirement already satisfied: numpy in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 6)) (1.19.4)
Requirement already satisfied: openpyxl in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 7)) (3.0.5)
Requirement already satisfied: ordered-set in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 8)) (4.0.2)
Requirement already satisfied: parse in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 9)) (1.18.0)
Requirement already satisfied: progressbar2 in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 10)) (3.53.1)
Requirement already satisfied: pyjson5 in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 11)) (1.4.7)
Requirement already satisfied: pytest in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 12)) (6.2.1)
Requirement already satisfied: pyyaml in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 13)) (5.3.1)
Requirement already satisfied: scipy>=1.2.1 in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 14)) (1.5.4)
Requirement already satisfied: simplejson in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 15)) (3.17.2)
Requirement already satisfied: sympy in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 16)) (1.7.1)
Requirement already satisfied: textx in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 17)) (2.3.0)
Requirement already satisfied: yapf==0.24.0 in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 18)) (0.24.0)
Requirement already satisfied: ply in ./env/lib/python3.8/site-packages (from sdf-timing==0.0.1->-r requirements.txt (line 2)) (3.11)
Requirement already satisfied: pyjson in ./env/lib/python3.8/site-packages (from sdf-timing==0.0.1->-r requirements.txt (line 2)) (1.3.0)
Requirement already satisfied: sortedcontainers<3.0,>=2.0 in ./env/lib/python3.8/site-packages (from intervaltree->-r requirements.txt (line 4)) (2.3.0)
Requirement already satisfied: six in ./env/lib/python3.8/site-packages (from junit-xml->-r requirements.txt (line 5)) (1.15.0)
Requirement already satisfied: jdcal in ./env/lib/python3.8/site-packages (from openpyxl->-r requirements.txt (line 7)) (1.4.1)
Requirement already satisfied: et-xmlfile in ./env/lib/python3.8/site-packages (from openpyxl->-r requirements.txt (line 7)) (1.0.1)
Requirement already satisfied: python-utils>=2.3.0 in ./env/lib/python3.8/site-packages (from progressbar2->-r requirements.txt (line 10)) (2.4.0)
Requirement already satisfied: pluggy<1.0.0a1,>=0.12 in ./env/lib/python3.8/site-packages (from pytest->-r requirements.txt (line 12)) (0.13.1)
Requirement already satisfied: iniconfig in ./env/lib/python3.8/site-packages (from pytest->-r requirements.txt (line 12)) (1.1.1)
Requirement already satisfied: attrs>=19.2.0 in ./env/lib/python3.8/site-packages (from pytest->-r requirements.txt (line 12)) (20.3.0)
Requirement already satisfied: packaging in ./env/lib/python3.8/site-packages (from pytest->-r requirements.txt (line 12)) (20.8)
Requirement already satisfied: toml in ./env/lib/python3.8/site-packages (from pytest->-r requirements.txt (line 12)) (0.10.2)
Requirement already satisfied: py>=1.8.2 in ./env/lib/python3.8/site-packages (from pytest->-r requirements.txt (line 12)) (1.10.0)
Requirement already satisfied: mpmath>=0.19 in ./env/lib/python3.8/site-packages (from sympy->-r requirements.txt (line 16)) (1.1.0)
Requirement already satisfied: Arpeggio>=1.9.0 in ./env/lib/python3.8/site-packages (from textx->-r requirements.txt (line 17)) (1.10.1)
Requirement already satisfied: pyparsing>=2.0.2 in ./env/lib/python3.8/site-packages (from packaging->pytest->-r requirements.txt (line 12)) (2.4.7)
Installing collected packages: fasm, sdf-timing, prjxray
  Attempting uninstall: fasm
    Found existing installation: fasm 0.0.1
    Uninstalling fasm-0.0.1:
      Successfully uninstalled fasm-0.0.1
  Running setup.py develop for fasm
  Attempting uninstall: sdf-timing
    Found existing installation: sdf-timing 0.0.1
    Uninstalling sdf-timing-0.0.1:
      Successfully uninstalled sdf-timing-0.0.1
  Running setup.py develop for sdf-timing
  Attempting uninstall: prjxray
    Found existing installation: prjxray 0.0.1
    Uninstalling prjxray-0.0.1:
      Successfully uninstalled prjxray-0.0.1
  Running setup.py develop for prjxray
Successfully installed fasm prjxray sdf-timing
# Install project's documentation dependencies
if [ -e env/bin/activate ]; then . env/bin/activate; fi; source utils/environment.python.sh; python -mpip install -r docs/requirements.txt
Collecting git+https://github.com/SymbiFlow/sphinxcontrib-markdown-symlinks (from -r docs/requirements.txt (line 13))
  Cloning https://github.com/SymbiFlow/sphinxcontrib-markdown-symlinks to /tmp/pip-req-build-kn_zcp4u
  Running command git clone -q https://github.com/SymbiFlow/sphinxcontrib-markdown-symlinks /tmp/pip-req-build-kn_zcp4u
Requirement already satisfied (use --upgrade to upgrade): markdown-code-symlinks==0.0.2 from git+https://github.com/SymbiFlow/sphinxcontrib-markdown-symlinks in ./env/lib/python3.8/site-packages (from -r docs/requirements.txt (line 13))
Requirement already satisfied: sphinx_materialdesign_theme in ./env/lib/python3.8/site-packages (from -r docs/requirements.txt (line 1)) (0.1.11)
Requirement already satisfied: docutils in ./env/lib/python3.8/site-packages (from -r docs/requirements.txt (line 3)) (0.16)
Requirement already satisfied: sphinx>=3.0 in ./env/lib/python3.8/site-packages (from -r docs/requirements.txt (line 4)) (3.3.1)
Requirement already satisfied: sphinx-autobuild in ./env/lib/python3.8/site-packages (from -r docs/requirements.txt (line 5)) (2020.9.1)
Requirement already satisfied: breathe in ./env/lib/python3.8/site-packages (from -r docs/requirements.txt (line 7)) (4.25.1)
Requirement already satisfied: recommonmark in ./env/lib/python3.8/site-packages (from -r docs/requirements.txt (line 8)) (0.6.0)
Requirement already satisfied: sphinx-markdown-tables in ./env/lib/python3.8/site-packages (from -r docs/requirements.txt (line 9)) (0.0.15)
Requirement already satisfied: sphinxcontrib-napoleon in ./env/lib/python3.8/site-packages (from -r docs/requirements.txt (line 10)) (0.7)
[... 31 lines cut ...]
   command: /home/jack/HDL/symbiflow-arch-defs/third_party/prjxray/env/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-kn_zcp4u/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-kn_zcp4u/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-et3d7eou
       cwd: /tmp/pip-req-build-kn_zcp4u/
  Complete output (6 lines):
  usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: setup.py --help [cmd1 cmd2 ...]
     or: setup.py --help-commands
     or: setup.py cmd --help
  
  error: invalid command 'bdist_wheel'
  ----------------------------------------
  ERROR: Failed building wheel for markdown-code-symlinks
  Running setup.py clean for markdown-code-symlinks
Failed to build markdown-code-symlinks
# Check that prjxray are available
if [ -e env/bin/activate ]; then . env/bin/activate; fi; source utils/environment.python.sh; python -c "import prjxray"
# Check fasm library was installed
if [ -e env/bin/activate ]; then . env/bin/activate; fi; source utils/environment.python.sh; python -c "import fasm"
[... cut ...]  fasm.output"
# Check sdfparse lib was installed
if [ -e env/bin/activate ]; then . env/bin/activate; fi; source utils/environment.python.sh; python -c "import sdf_timing"
if [ -e env/bin/activate ]; then . env/bin/activate; fi; source utils/environment.python.sh; python -c "import sdf_timing.sdfparse"
# Check YAML is installed
if [ -e env/bin/activate ]; then . env/bin/activate; fi; source utils/environment.python.sh; python -c "import yaml" || (echo "Unable to find python-yaml" && exit 1)

```

hansfbaier, 2020-12-17:
I found a way to make it work:
`source ./env/bin/activate`
and then make env ran without errors

hansfbaier, 2020-12-17:
This ideally should be in the quickstart doc.

hansfbaier, 2020-12-17:
Should this issue be closed?

litghost, 2020-12-17:
The makefile should be updated to do the right thing.  The idea was that the makefile should enter the environment as needed without additional user action.

MidsummerNight, 2021-11-29:
I just ran into this issue as well. I'm on Ubuntu 20.04.3, and I encountered no issues at all when using `prjxray` a few months ago  on the same machine. Decided to fetch the latest code, and this happened. Unfortunately running `source ./env/bin/activate` before `make env` as suggested above doesn't fix the problem, `make env` would still give the same error.

What is `markdown-code-symlinks` supposed to be anyway?

Error output is as follows:
```
~/prjxray$ make env
python3 -mvenv env
# Install project dependencies
if [ -e env/bin/activate ]; then . env/bin/activate; fi; source utils/environment.python.sh; python -mpip install -r requirements.txt
Obtaining file:///home/lbh/prjxray/third_party/fasm (from -r requirements.txt (line 1))
Obtaining file:///home/lbh/prjxray/third_party/python-sdf-timing (from -r requirements.txt (line 2))
Obtaining file:///home/lbh/prjxray (from -r requirements.txt (line 3))
Requirement already satisfied: intervaltree in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 4)) (3.1.0)
Requirement already satisfied: junit-xml in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 5)) (1.9)
Requirement already satisfied: numpy in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 6)) (1.21.4)
Requirement already satisfied: openpyxl in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 7)) (3.0.9)
Requirement already satisfied: ordered-set in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 8)) (4.0.2)
Requirement already satisfied: parse in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 9)) (1.19.0)
Requirement already satisfied: progressbar2 in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 10)) (3.55.0)
Requirement already satisfied: pyjson5 in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 11)) (1.6.0)
Requirement already satisfied: pytest in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 12)) (6.2.5)
Requirement already satisfied: pyyaml in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 13)) (6.0)
Requirement already satisfied: scipy>=1.2.1 in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 14)) (1.7.3)
Requirement already satisfied: simplejson in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 15)) (3.17.6)
Requirement already satisfied: sympy in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 16)) (1.9)
Requirement already satisfied: textx in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 17)) (2.3.0)
Requirement already satisfied: yapf==0.24.0 in ./env/lib/python3.8/site-packages (from -r requirements.txt (line 18)) (0.24.0)
Requirement already satisfied: ply in ./env/lib/python3.8/site-packages (from sdf-timing==0.0.1->-r requirements.txt (line 2)) (3.11)
Requirement already satisfied: pyjson in ./env/lib/python3.8/site-packages (from sdf-timing==0.0.1->-r requirements.txt (line 2)) (1.3.0)
Requirement already satisfied: sortedcontainers<3.0,>=2.0 in ./env/lib/python3.8/site-packages (from intervaltree->-r requirements.txt (line 4)) (2.4.0)
Requirement already satisfied: six in ./env/lib/python3.8/site-packages (from junit-xml->-r requirements.txt (line 5)) (1.16.0)
Requirement already satisfied: et-xmlfile in ./env/lib/python3.8/site-packages (from openpyxl->-r requirements.txt (line 7)) (1.1.0)
Requirement already satisfied: python-utils>=2.3.0 in ./env/lib/python3.8/site-packages (from progressbar2->-r requirements.txt (line 10)) (2.5.6)
Requirement already satisfied: packaging in ./env/lib/python3.8/site-packages (from pytest->-r requirements.txt (line 12)) (21.3)
Requirement already satisfied: pluggy<2.0,>=0.12 in ./env/lib/python3.8/site-packages (from pytest->-r requirements.txt (line 12)) (1.0.0)
Requirement already satisfied: attrs>=19.2.0 in ./env/lib/python3.8/site-packages (from pytest->-r requirements.txt (line 12)) (21.2.0)
Requirement already satisfied: py>=1.8.2 in ./env/lib/python3.8/site-packages (from pytest->-r requirements.txt (line 12)) (1.11.0)
Requirement already satisfied: iniconfig in ./env/lib/python3.8/site-packages (from pytest->-r requirements.txt (line 12)) (1.1.1)
Requirement already satisfied: toml in ./env/lib/python3.8/site-packages (from pytest->-r requirements.txt (line 12)) (0.10.2)
Requirement already satisfied: mpmath>=0.19 in ./env/lib/python3.8/site-packages (from sympy->-r requirements.txt (line 16)) (1.2.1)
Requirement already satisfied: Arpeggio>=1.9.0 in ./env/lib/python3.8/site-packages (from textx->-r requirements.txt (line 17)) (1.10.2)
Requirement already satisfied: pyparsing!=3.0.5,>=2.0.2 in ./env/lib/python3.8/site-packages (from packaging->pytest->-r requirements.txt (line 12)) (3.0.6)
Installing collected packages: fasm, sdf-timing, prjxray
  Attempting uninstall: fasm
    Found existing installation: fasm 0.0.1
    Uninstalling fasm-0.0.1:
      Successfully uninstalled fasm-0.0.1
  Running setup.py develop for fasm
  Attempting uninstall: sdf-timing
    Found existing installation: sdf-timing 0.0.1
    Uninstalling sdf-timing-0.0.1:
      Successfully uninstalled sdf-timing-0.0.1
  Running setup.py develop for sdf-timing
  Attempting uninstall: prjxray
    Found existing installation: prjxray 0.0.1
    Uninstalling prjxray-0.0.1:
      Successfully uninstalled prjxray-0.0.1
  Running setup.py develop for prjxray
Successfully installed fasm prjxray sdf-timing
[... cut ...]
if [ -e env/bin/activate ]; then . env/bin/activate; fi; source utils/environment.python.sh; python -c "import fasm.output"
# Check sdfparse lib was installed
if [ -e env/bin/activate ]; then . env/bin/activate; fi; source utils/environment.python.sh; python -c "import sdf_timing"
if [ -e env/bin/activate ]; then . env/bin/activate; fi; source utils/environment.python.sh; python -c "import sdf_timing.sdfparse"
# Check YAML is installed
if [ -e env/bin/activate ]; then . env/bin/activate; fi; source utils/environment.python.sh; python -c "import yaml" || (echo "Unable to find python-yaml" && exit 1)
~/prjxray$ 
```

trayres, 2023-05-24:
I just ran into this issue, but:
`source ./env/bin/activate`
seemed to work in my case (on Ubuntu 20.02 LTS)