f4pga/prjxray #1540: Investigate PCIE_2_1 PIPs

acomodi, 2021-01-13:
This issue is to keep track of the information on the PIPs around the PCIE_2_1 primitives for xc7 parts.

### Description

The PCIE_2_1 primitive is a hard block that is half-clock region tall and has many connections with the fabric. By investigating the device view, it results that there are some pips in the `PCIE_INT_INTERFACE` tiles which work a bit like the `BOUNCE` pips in the INT tiles.

[image]

The image above represents a section of the `PCIE_INT_INTERFACE` that displays said pips.

By further investigation, these pips seem to occupy the 27th frame of the segment in which the PCIE_2_1 primitive is, and that,  by default, the pip allows a direct connection, without *bouncing* to the additional wire before getting to the destination port.

These PIPs should likely require a pce-int-pips fuzzer.

litghost, 2021-03-08:
Is this complete?

acomodi, 2021-03-09:
Yes, this was done. Closing.

closed 2021-03-09 by acomodi