f4pga/prjxray #1557: New parts setting cause fuzzer to fail if ran from the fuzzer directory

acomodi, 2021-01-19:
With the new way of generalizing the various parts, all the required environment variables to correctly run the fuzzers are set in the `run_fuzzer.py` script (e.g. `XRAY_FABRIC`, `XRAY_PIN_XX`).

This causes issues during fuzzer developement, as the environment is incomplete when running the fuzzer without invoking the `run_fuzzer.py` script, resulting in fuzzer failures.

litghost, 2021-01-19:
@acomodi We should be able to refactor the setenv portion of `run_fuzzer.py`.  If we write a tool that outputs bash "export ...=..." then something like:

```
source $(python3 ${XRAY_UTILS}/create_environment.py --part ${XRAY_PART})
```

Where `create_environment.py --part ${XRAY_PART}` outputs:
```
export XRAY_FABRIC=...
... etc ...
```

That make sense?

acomodi, 2021-01-19:
@litghost Sure, I think this is a good solution, thanks

closed 2021-01-20 by litghost