f4pga/prjxray #1579: xc7s15 / Spartan Edge Accelerator Board

cederom, 2021-02-05:
Hello world :-)

Is there any chance to make XRAY work with Spartan Edge Accelerator Board based on Xilinx XC7S15 FPGA? :-)

https://wiki.seeedstudio.com/Spartan-Edge-Accelerator-Board/

Any hints welcome :-)
Tomek

andrewb1999, 2021-02-05:
@cederom #1534 has some initial work on supporting Spartan-7 parts in prjxray, but it has not yet been completed. Also, support would have to be added to symbiflow-arch-defs if you wanted an open source verilog to bitstream flow.

cederom, 2021-02-05:
Thank you @andrewb1999 ! Good news :-)

mithro, 2021-12-14:
A new prjxray-db was published with this initial spartan7 support @ https://github.com/SymbiFlow/prjxray-db/tree/master/spartan7

The HTMLified version can be found at https://symbiflow.github.io/prjxray-db/#database-for-spartan7 with the HTML rendered tile graph found at [spartan7](https://symbiflow.github.io/prjxray-db/spartan7/).

bl0x, 2023-02-24:
Is it known, if the xc7s25 fabric is the same as for xc7s50?

marzoul, 2023-02-25:
Hi @bl0x, according do Xilinx doc ug470 "7 Series FPGAs Configuration User Guide", the bitstream length is not the same for s25 and s50. So these are certainly not the same fabrics.
However probably s6=s15 and s75=s100 according to bitstream length. To be confirmed by also checking AC characteristics and floorplan view in Vivado (see related #1871 about artix).

bl0x, 2023-02-25:
Dear @marzoul , thanks for the hints. I'll take a look.

cfib, 2023-02-25:
According to DS189 table 5, quiescent VCCINT current is the same for s6 and s15 (36 mA), as well as s75 and s100 (148 mA).
However, s25 (48 mA) and  s50 (96 mA) seem to be different from either.