f4pga/prjxray #1626: zynq 7015 | MYD-C7Z015 board

vladoshub, 2021-03-17:
Hello everyone!
can you add support for this facility to the project?

litghost, 2021-03-17:
We have documentation on how to do this: https://github.com/SymbiFlow/prjxray/blob/master/docs/db_dev_process/newpart.rst

Feel free to take a shot!

vladoshub, 2021-03-17:
> We have documentation on how to do this: https://github.com/SymbiFlow/prjxray/blob/master/docs/db_dev_process/newpart.rst
> 
> Feel free to take a shot!
> 
> > [image][image] GitHub**[SymbiFlow/prjxray](https://github.com/SymbiFlow/prjxray)**Documenting the Xilinx 7-series bit-stream format. - SymbiFlow/prjxray

yes, I saw it. but I really don’t understand where to get the data for "XRAY_IOI3_TILES" for the settings file. Thanks in advance