f4pga/prjxray #1639: AssertionError: Mapping file /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/settings/artix7/resources.yaml does not exists

mithro, 2021-03-30:
```
========================================
Preparing database
----------------------------------------

Preparing artix7 files
============================
Traceback (most recent call last):
  File "/tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/utils/create_environment.py", line 59, in <module>
    main()
  File "/tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/utils/create_environment.py", line 52, in main
    environment = get_environment_variables()
  File "/tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/utils/create_environment.py", line 39, in get_environment_variables
    resources = get_part_resources(res_path, os.environ['XRAY_PART'])
  File "/tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/prjxray/util.py", line 62, in get_part_resources
    assert os.path.isfile(filename), \
AssertionError: Mapping file /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/settings/artix7/resources.yaml does not exists
Part xc7a35tiftg256-1L has an unsupported device xc7a35ti
Part xc7a35tifgg484-1L has an unsupported device xc7a35ti
Part xc7a35ticsg325-1L has an unsupported device xc7a35ti
Part xc7a35ticsg324-1L has an unsupported device xc7a35ti
Part xc7a35ticpg236-1L has an unsupported device xc7a35ti
Part xc7a50ticpg236-1L has an unsupported device xc7a50ti
Part xc7a50tiftg256-1L has an unsupported device xc7a50ti
Part xc7a50tifgg484-1L has an unsupported device xc7a50ti
Part xc7a50ticsg325-1L has an unsupported device xc7a50ti
Part xc7a50ticsg324-1L has an unsupported device xc7a50ti
Part xc7a15tcpg236-1 has an unsupported device xc7a15t
Part xc7a15tcpg236-2 has an unsupported device xc7a15t
Part xc7a15tcpg236-2L has an unsupported device xc7a15t
Part xc7a15tcpg236-3 has an unsupported device xc7a15t
Part xc7a15tcsg324-1 has an unsupported device xc7a15t
Part xc7a15tcsg324-2 has an unsupported device xc7a15t
Part xc7a15tcsg324-2L has an unsupported device xc7a15t
Part xc7a15tcsg324-3 has an unsupported device xc7a15t
Part xc7a15tcsg325-1 has an unsupported device xc7a15t
Part xc7a15tcsg325-2 has an unsupported device xc7a15t
Part xc7a15tcsg325-2L has an unsupported device xc7a15t
Part xc7a15tcsg325-3 has an unsupported device xc7a15t
Part xc7a15tfgg484-1 has an unsupported device xc7a15t
Part xc7a15tfgg484-2 has an unsupported device xc7a15t
Part xc7a15tfgg484-2L has an unsupported device xc7a15t
Part xc7a15tfgg484-3 has an unsupported device xc7a15t
Part xc7a15tftg256-1 has an unsupported device xc7a15t
Part xc7a15tftg256-2 has an unsupported device xc7a15t
Part xc7a15tftg256-2L has an unsupported device xc7a15t
Part xc7a15tftg256-3 has an unsupported device xc7a15t
Part xc7a15ticpg236-1L has an unsupported device xc7a15ti
Part xc7a15ticsg324-1L has an unsupported device xc7a15ti
Part xc7a15ticsg325-1L has an unsupported device xc7a15ti
Part xc7a15tifgg484-1L has an unsupported device xc7a15ti
Part xc7a15tiftg256-1L has an unsupported device xc7a15ti
Part xc7a75tfgg484-1 has an unsupported device xc7a75t
Part xc7a75tfgg484-2 has an unsupported device xc7a75t
Part xc7a75tfgg484-2L has an unsupported device xc7a75t
Part xc7a75tfgg484-3 has an unsupported device xc7a75t
Part xc7a75tfgg676-1 has an unsupported device xc7a75t
Part xc7a75tfgg676-2 has an unsupported device xc7a75t
Part xc7a75tfgg676-2L has an unsupported device xc7a75t
Part xc7a75tfgg676-3 has an unsupported device xc7a75t
Part xc7a75tftg256-1 has an unsupported device xc7a75t
Part xc7a75tftg256-2 has an unsupported device xc7a75t
Part xc7a75tftg256-2L has an unsupported device xc7a75t
Part xc7a75tftg256-3 has an unsupported device xc7a75t
Part xc7a75tcsg324-1 has an unsupported device xc7a75t
Part xc7a75tcsg324-2 has an unsupported device xc7a75t
Part xc7a75tcsg324-2L has an unsupported device xc7a75t
Part xc7a75tcsg324-3 has an unsupported device xc7a75t
Part xc7a75ticsg324-1L has an unsupported device xc7a75ti
Part xc7a75tifgg484-1L has an unsupported device xc7a75ti
Part xc7a75tifgg676-1L has an unsupported device xc7a75ti
Part xc7a75tiftg256-1L has an unsupported device xc7a75ti
Part xc7a100tiftg256-1L has an unsupported device xc7a100ti
Part xc7a100tifgg676-1L has an unsupported device xc7a100ti
Part xc7a100tifgg484-1L has an unsupported device xc7a100ti
Part xc7a100ticsg324-1L has an unsupported device xc7a100ti
Part xc7a200tisbv484-1L has an unsupported device xc7a200ti
Part xc7a200tisbg484-1L has an unsupported device xc7a200ti
Part xc7a200tiffv1156-1L has an unsupported device xc7a200ti
Part xc7a200tiffg1156-1L has an unsupported device xc7a200ti
Part xc7a200tifbv676-1L has an unsupported device xc7a200ti
Part xc7a200tifbv484-1L has an unsupported device xc7a200ti
Part xc7a200tifbg676-1L has an unsupported device xc7a200ti
Part xc7a200tifbg484-1L has an unsupported device xc7a200ti
Traceback (most recent call last):
  File "/tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/utils/create_environment.py", line 59, in <module>
    main()
  File "/tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/utils/create_environment.py", line 52, in main
    environment = get_environment_variables()
  File "/tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/utils/create_environment.py", line 39, in get_environment_variables
    resources = get_part_resources(res_path, os.environ['XRAY_PART'])
  File "/tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/prjxray/util.py", line 62, in get_part_resources
    assert os.path.isfile(filename), \
AssertionError: Mapping file /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/settings/artix7/resources.yaml does not exists
```

litghost, 2021-03-30:
@mithro What's the context of this failure?

dnltz, 2021-03-31:
https://github.com/SymbiFlow/prjxray/pull/1627 should fix this AssertionError.

litghost, 2021-04-01:
@mithro I never did get the context for this issue, but #1627 is merged.  Can this be closed?

mithro, 2021-04-01:
It was an error I saw on the CI system while debugging other issues.