f4pga/prjxray #1658: parts.yaml file not found when running settings script for board family

benglines, 2021-04-29:
When trying to install Project X-Ray for the first time, I get the following error when running `source settings/artix7.sh`

```
Traceback (most recent call last):
  File "/home/benglines/prjxray_test/utils/create_environment.py", line 66, in <module>
    main()
  File "/home/benglines/prjxray_test/utils/create_environment.py", line 59, in main
    environment = get_environment_variables()
  File "/home/benglines/prjxray_test/utils/create_environment.py", line 37, in get_environment_variables
    part_info = get_part_information(db_root, part)
  File "/home/benglines/prjxray_test/prjxray/util.py", line 44, in get_part_information
    "Mapping file {} does not exists".format(filename)
AssertionError: Mapping file /home/benglines/prjxray_test/database/artix7/mapping/parts.yaml does not exists
```
I have also seen it complain about not finding a resources.yaml file (just like #1639), but I am not exactly sure why that happened nor how to recreate the error. Either way, I solved the issue in both cases by running `make db-prepare-artix7`, and then `source settings/artix7.sh`. The quick start guide suggest you do them in the opposite order. 

This issue could be resolved by simply switching Step 6 and Step 7 in the quick start guide, so that the shell scripts run in the correct order i.e. running `source/settings/artix7.sh` _after_ running `make db-prepare-parts` or `make db-prepare-artix7`.

mithro, 2021-04-29:
Seems like we should set up the CI to use [`tuttest`](https://github.com/antmicro/tuttest) to run the commands in the README [like we do for the symbiflow-examples repository](https://github.com/SymbiFlow/symbiflow-examples/runs/2460999808?check_suite_focus=true#step:8:7).

dnltz, 2021-04-29:
@benglines - Thanks for reporting. Indeed, the script should be sourced after preparing the db.
@mithro - If I have a little bit time soon, I can try to add such a simple test.