f4pga/prjxray #1663: Pretty diff is broken on CI

mithro, 2021-04-30:
```
----------------------------------------
 Generating pretty diff output
----------------------------------------
/usr/local/lib/node_modules/diff2html-cli/node_modules/yargs-parser/build/index.cjs:997
        throw Error(`yargs parser supports a minimum Node.js version of ${minNodeVersion}. Read our version support policy: https://github.com/yargs/yargs-parser#supported-nodejs-versions`);
        ^

Error: yargs parser supports a minimum Node.js version of 10. Read our version support policy: https://github.com/yargs/yargs-parser#supported-nodejs-versions
    at Object.<anonymous> (/usr/local/lib/node_modules/diff2html-cli/node_modules/yargs-parser/build/index.cjs:997:15)
    at Module._compile (module.js:573:30)
    at Object.Module._extensions..js (module.js:584:10)
    at Module.load (module.js:507:32)
    at tryModuleLoad (module.js:470:12)
    at Function.Module._load (module.js:462:3)
    at Module.require (module.js:517:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/diff2html-cli/node_modules/yargs/build/index.cjs:2855:16)
    at Module._compile (module.js:573:30)
/usr/local/lib/node_modules/diff2html-cli/node_modules/yargs-parser/build/index.cjs:997
        throw Error(`yargs parser supports a minimum Node.js version of ${minNodeVersion}. Read our version support policy: https://github.com/yargs/yargs-parser#supported-nodejs-versions`);
        ^

Error: yargs parser supports a minimum Node.js version of 10. Read our version support policy: https://github.com/yargs/yargs-parser#supported-nodejs-versions
    at Object.<anonymous> (/usr/local/lib/node_modules/diff2html-cli/node_modules/yargs-parser/build/index.cjs:997:15)
    at Module._compile (module.js:573:30)
    at Object.Module._extensions..js (module.js:584:10)
    at Module.load (module.js:507:32)
    at tryModuleLoad (module.js:470:12)
    at Function.Module._load (module.js:462:3)
    at Module.require (module.js:517:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/diff2html-cli/node_modules/yargs/build/index.cjs:2855:16)
    at Module._compile (module.js:573:30)
----------------------------------------
```

Sweetie7, 2022-02-02:
I am also facing this issue. Could someone please help me with this?

Ragunath, 2024-02-19:
I am facing this issue after upgrading my angular hybrid from 11 application to 12 angular version and run "npm run test" .
I tried node version from Node version is 12.15.0 to 14.20.0, no luck.
Currently, **Karma Version : 6.4.2**(it was update to latest automatically on running "ng update @angular/core@12 @angular/cli@12 –force")

Then , I downgraded my **karma version to 5.2.3, T**his time the error message is different .

ERROR [karma-server]: UncaughtException:: Unable to resolve module [set] from [C:/Users/Project/lib.js]
{
  "basedir": "C:/Users/project",
  "extensions": [
  "extensions": [
    ".js",
    ".json",
    ".ts",
    ".tsx"
  ],
  "moduleDirectory": [
    "node_modules"
  ],
  "modules": {
    "assert": "C:\\Users\\Project\\\node_modules\\karma-typescript\\node_modules\\assert\\build\\assert.js",
    "buffer": "C:\\Users\\Project\\\node_modules\\buffer\\index.js",
    "child_process": "C:\\Users\\Project\\\node_modules\\karma-typescript\\dist\\bundler\\resolve\\empty.js",
    "cluster": "C:\\Users\\Project\\\node_modules\\karma-typescript\\dist\\bundler\\resolve\\empty.js",
    "console": "C:\\Users\\Project\\\node_modules\\console-browserify\\index.js",
    "constants": "C:\\Users\\Project\\\node_modules\\constants-browserify\\constants.json",
    "crypto": "C:\\Users\\Project\\\node_modules\\crypto-browserify\\index.js",
    "dgram": "C:\\Users\\Project\\\node_modules\\karma-typescript\\dist\\bundler\\resolve\\empty.js",
    "dns": "C:\\Users\\Project\\\node_modules\\karma-typescript\\dist\\bundler\\resolve\\empty.js",
    "domain": "C:\\Users\\Project\\\node_modules\\karma-typescript\\node_modules\\domain-browser\\source\\index.js",
    "events": "C:\\Users\\Project\\\node_modules\\karma-typescript\\node_modules\\events\\events.js",
    "fs": "C:\\Users\\Project\\\node_modules\\karma-typescript\\dist\\bundler\\resolve\\empty.js",
    "http": "C:\\Users\\Project\\\node_modules\\karma-typescript\\node_modules\\stream-http\\index.js",
    "https": "C:\\Users\\Project\\\node_modules\\https-browserify\\index.js",
    "module": "C:\\Users\\Project\\\node_modules\\karma-typescript\\dist\\bundler\\resolve\\empty.js",
    "net": "C:\\Users\\Project\\\node_modules\\karma-typescript\\dist\\bundler\\resolve\\empty.js",
    "os": "C:\\Users\\Project\\\node_modules\\os-browserify\\browser.js",
    "path": "C:\\Users\\Project\\\node_modules\\karma-typescript\\node_modules\\path-browserify\\index.js",
    "punycode": "C:\\Users\\Project\\\node_modules\\karma-typescript\\node_modules\\punycode\\punycode.js",
    "querystring": "C:\\Users\\Project\\\node_modules\\querystring-es3\\index.js",
    "readline": "C:\\Users\\Project\\\node_modules\\karma-typescript\\dist\\bundler\\resolve\\empty.js",
    "repl": "C:\\Users\\Project\\\node_modules\\karma-typescript\\dist\\bundler\\resolve\\empty.js",
    "stream": "C:\\Users\\Project\\\node_modules\\stream-browserify\\index.js",
    "_stream_duplex": "C:\\Users\\Project\\\node_modules\\karma-typescript\\node_modules\\readable-stream\\lib\\_stream_duplex.js",
    "_stream_passthrough": "C:\\Users\\Project\\\node_modules\\karma-typescript\\node_modules\\readable-stream\\lib\\_stream_passthrough.js",
    "_stream_readable": "C:\\Users\\Project\\\node_modules\\karma-typescript\\node_modules\\readable-stream\\lib\\_stream_readable.js",
    "_stream_transform": "C:\\Users\\Project\\\node_modules\\karma-typescript\\node_modules\\readable-stream\\lib\\_stream_transform.js",
    "_stream_writable": "C:\\Users\\Project\\\node_modules\\karma-typescript\\node_modules\\readable-stream\\lib\\_stream_writable.js",
    "string_decoder": "C:\\Users\\Project\\\node_modules\\karma-typescript\\node_modules\\string_decoder\\lib\\string_decoder.js",
    "sys": "C:\\Users\\Project\\\node_modules\\karma-typescript\\node_modules\\util\\util.js",
    "timers": "C:\\Users\\Project\\\node_modules\\karma-typescript\\node_modules\\timers-browserify\\main.js",
    "tls": "C:\\Users\\Project\\\node_modules\\karma-typescript\\dist\\bundler\\resolve\\empty.js",
    "tty": "C:\\Users\\Project\\\node_modules\\tty-browserify\\index.js",
    "url": "C:\\Users\\Project\\\node_modules\\url\\url.js",
    "util": "C:\\Users\\Project\\\node_modules\\karma-typescript\\node_modules\\util\\util.js",
    "vm": "C:\\Users\\Project\\\node_modules\\karma-typescript\\node_modules\\vm-browserify\\index.js",
    "zlib": "C:\\Users\\Project\\\node_modules\\browserify-zlib\\lib\\index.js",
    "_process": "C:\\Users\\Project\\\node_modules\\process\\browser.js"
  },
  "filename": "",
  "paths": []
}
Error: Cannot find module 'set' from 'C:/Users/Project'
19 02 2024 13:06:56.945:ERROR [karma-server]: Error: Unable to resolve module [set] from [C:/Users/Project/lib.js]
{
  "basedir": "C:/Users/Project",
  "extensions": [
    ".js",
    ".json",
    ".ts",
[... 41 lines cut ...]
    "vm": "C:\\Users\\Project\\node_modules\\karma-typescript\\node_modules\\vm-browserify\\index.js",
    "zlib": "C:\\Users\\Project\\node_modules\\browserify-zlib\\lib\\index.js",
    "_process": "C:\\Users\\Project\\node_modules\\process\\browser.js"
  },
  "filename": "",
  "paths": []
}
Error: Cannot find module 'set' from 'C:/Users/Project'
    at C:/Users/Project\node_modules\karma-typescript\dist\bundler\resolve\resolver.js:178:27
    at C:/Users/Project\node_modules\browser-resolve\index.js:265:24
    at C:/Users/Project\node_modules\browser-resolve\node_modules\resolve\lib\async.js:46:14
[... cut ...]
(node:27540) [log4js-node-DEP0004] DeprecationWarning: Pattern %d{DATE} is deprecated due to the confusion it causes when used. Please use %d{DATETIME} instead.
(node:27540) [log4js-node-DEP0004] DeprecationWarning: Pattern %d{DATE} is deprecated due to the confusion it causes when used. Please use %d{DATETIME} instead.
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @Project test: `karma start karma.conf.js --single-run=true`
npm ERR! Exit status 1
npm ERR!
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.