f4pga/prjxray #1674: Documentation Issue

AjinkyaRaghuwanshi, 2021-05-21:
The step 6 in the Quick start guide should be done after completing the step 7 and 8. Otherwise it throws an error.
Also the documentation does not specifically provide the directory the command should be ran in for ex: after the step 9 we will have to move to a different directory for step 10 and then follow the C++ commands step, so please take care of these issue,

mithro, 2021-05-24:
@acomodi - Who is the correct person to fix this?

acomodi, 2021-05-24:
I can fix this, it is quite straightforward. Long term we might use tuttest to verify the readme commands are working properly

benglines, 2021-06-03:
I saw that the wording was fixed to show that for Step 10 you need to be in a different directory, but I didn't see any changes to the ordering of the quick start guide steps, so I linked a pull request that puts step 6 after steps 7 and 8

closed 2021-06-07 by acomodi