f4pga/prjxray #1686: fuzzer does not run

KiranKanchi, 2021-06-06:
Dear team,

I am setting up the symbiflow for the first time and I have issue in step 9.

After step 8: ./download-latest-db.sh

Pick a fuzzer (or write your own) and run:

cd fuzzers/010-clb-lutinit
make -j$(nproc) run

--------------------------------------------------------------------------------
make run results in :

make clean
make[1]: Entering directory '/home/kirankanchi/xilinx/prjxray/prjxray/fuzzers/010-clb-lutinit'
rm -rf build run.ok
make[1]: Leaving directory '/home/kirankanchi/xilinx/prjxray/prjxray/fuzzers/010-clb-lutinit'
make database
make[1]: Entering directory '/home/kirankanchi/xilinx/prjxray/prjxray/fuzzers/010-clb-lutinit'
mkdir -p build
bash generate.sh build/specimen_001
generate.sh: line 11: source: filename argument required
source: usage: source filename [arguments]
generate.sh: line 15: -mode: command not found
generate.sh: line 18: -F: command not found
generate.sh: line 18: -F: command not found
generate.sh: line 18: -F: command not found
Traceback (most recent call last):
  File "/home/kirankanchi/xilinx/prjxray/prjxray/fuzzers/010-clb-lutinit/generate.py", line 16, in <module>
    segmk = Segmaker("design_%s.bits" % sys.argv[1])
  File "/home/kirankanchi/xilinx/prjxray/prjxray/prjxray/segmaker.py", line 94, in __init__
    assert self.fabric, "No fabric specified."
AssertionError: No fabric specified.
Traceback (most recent call last):
  File "/home/kirankanchi/xilinx/prjxray/prjxray/fuzzers/010-clb-lutinit/generate.py", line 16, in <module>
    segmk = Segmaker("design_%s.bits" % sys.argv[1])
  File "/home/kirankanchi/xilinx/prjxray/prjxray/prjxray/segmaker.py", line 94, in __init__
    assert self.fabric, "No fabric specified."
AssertionError: No fabric specified.
Traceback (most recent call last):
  File "/home/kirankanchi/xilinx/prjxray/prjxray/fuzzers/010-clb-lutinit/generate.py", line 16, in <module>
    segmk = Segmaker("design_%s.bits" % sys.argv[1])
  File "/home/kirankanchi/xilinx/prjxray/prjxray/prjxray/segmaker.py", line 94, in __init__
    assert self.fabric, "No fabric specified."
AssertionError: No fabric specified.
make[1]: *** [Makefile:27: build/specimen_001/OK] Error 1
make[1]: Leaving directory '/home/kirankanchi/xilinx/prjxray/prjxray/fuzzers/010-clb-lutinit'
make: *** [Makefile:32: run] Error 2

Any pointers on what could be wrong ? Thank you very much in advance.

Kind regards,
KiranKanchi

Xiretza, 2021-06-06:
Look like you skipped step 6 (sourcing the environment). Nevertheless, these scripts should probably be `set -eu` so they don't accidentally cause [major problems](https://github.com/ValveSoftware/steam-for-linux/issues/3671) in cases like this.

KiranKanchi, 2021-06-06:
Hello Xiretza,

Thank you for your kind reply, i repeated the steps and see the same error. Here is my history:

  838  sudo make build
  839  source settings/artix7.sh
  840  make db-prepare-artix7
  841  sudo make db-prepare-artix7
  842  sudo ./download-latest-db.sh
  843  cd fuzzers/010-clb-lutinit/
  844  sudo make -j12 run

I am not sure where am i going wrong.

Kind regards,
KiranKanchi

Xiretza, 2021-06-06:
Why are you trying to do any of this as root? `sudo` strips the environment for security reasons.

KiranKanchi, 2021-06-06:
ah! I bypassed your root check earlier because I could not figure out how to proceed as non root user. i will restore it back and figure out to run without root. thank you very much again!

KiranKanchi, 2021-06-06:
I created a new user and followed the same steps* and there are no problems.
*Note: I had to go to step 7 first and then do step 6.

Kind regards,
KiranKanchi

closed 2021-06-06 by KiranKanchi