f4pga/prjxray #1701: How to get in touch about a security issue?

zidingz, 2021-07-01:
Hey there,

As there isn't a `SECURITY.md` with an email on your repository, I am unsure how to contact you regarding a potential security issue.

Would you kindly add a `SECURITY.md` file with an e-mail to your repository? GitHub [recommends](https://docs.github.com/en/code-security/getting-started/adding-a-security-policy-to-your-repository) this as the best way to ensure security issues are responsibly disclosed, and it would massively help security researchers get in touch next time.

Thank you so much and I look forward to hearing from you!

mithro, 2021-07-01:
Send me an email at tansell@google.com and I'll look into it.

zidingz, 2021-07-01:
Thanks for the swift response!

An email with be with you shortly.

closed 2021-07-01 by zidingz