f4pga/prjxray #1740: LiteX/LiteDRAM behaviour inconsistencies between Symbiflow/Vivado.

enjoy-digital, 2021-09-08:
As discussed in https://github.com/enjoy-digital/litex/issues/751, it seems the LiteDRAM calibration behavior  is different between Symbiflow and Vivado. I'm closing the issue in LiteX and think this should be tracked here. 

My guess (from previous bring up issues I had with LiteDRAM/Vivado on Xilinx boards) would be that the IOStandard or Internal VREF constraints are not correctly applied with Symbiflow.

kgugala, 2021-09-09:
I just ran the whole flow following the guide in symbiflow-examples and got working design (including RAM) on Arty. Here is a bootlog:

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Sep  9 2021 11:48:08
 BIOS CRC passed (04460035)

 Migen git sha1: 9a37a58
 LiteX git sha1: afbac26e

--=============== SoC ==================--
CPU:            VexRiscv @ 80MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            32KiB
SRAM:           8KiB
L2:             8KiB
SDRAM:          262144KiB 16-bit @ 640MT/s (CL-6 CWL-5)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0 
Read leveling:
  m0, b0: |00000000000000000000000000000000| delays: -
  m0, b1: |00000000000000000000000000011111| delays: 29+-02
  m0, b2: |00000000000000000000000000000000| delays: -
  m0, b3: |00000000000000000000000000000000| delays: -
  m0, b4: |00000000000000000000000000000000| delays: -
  m0, b5: |00000000000000000000000000000000| delays: -
  m0, b6: |00000000000000000000000000000000| delays: -
  m0, b7: |00000000000000000000000000000000| delays: -
  best: m0, b01 delays: 29+-02
  m1, b0: |00000000000000000000000000000000| delays: -
  m1, b1: |00000000000000000000000000001111| delays: 30+-02
  m1, b2: |00000000000000000000000000000000| delays: -
  m1, b3: |00000000000000000000000000000000| delays: -
  m1, b4: |00000000000000000000000000000000| delays: -
  m1, b5: |00000000000000000000000000000000| delays: -
  m1, b6: |00000000000000000000000000000000| delays: -
  m1, b7: |00000000000000000000000000000000| delays: -
  best: m1, b01 delays: 30+-02
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2MiB)...
  Write: 0x40000000-0x40200000 2MiB     
   Read: 0x40000000-0x40200000 2MiB     
Memtest OK
Memspeed at 0x40000000 (2MiB)...
  Write speed: 21MiB/s
   Read speed: 17MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
No boot medium found

--============= Console ================--

litex> mem_
mem_speed    mem_test     mem_copy     mem_write    mem_read
mem_list     
litex> mem_test
mem_test <addr> [<maxsize>]
litex> mem_test 0x40000000 0x10000
Memtest at 0x40000000 (64KiB)...
  Write: 0x40000000-0x40010000 64KiB   
   Read: 0x40000000-0x40010000 64KiB   
Memtest OK

litex> 
```

@enjoy-digital which LiteX/Symbiflow version you're using (LiteX in symbiflow-examples is not the latest one)

kgugala, 2021-09-09:
I did the test with the latest LiteX and it still works:
```        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2021 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Sep  9 2021 12:27:19
 BIOS CRC passed (fe9bc101)

 Migen git sha1: 27dbf03
 LiteX git sha1: e0e9311c

--=============== SoC ==================--
CPU:            VexRiscv @ 80MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128KiB
SRAM:           8KiB
L2:             8KiB
SDRAM:          262144KiB 16-bit @ 640MT/s (CL-6 CWL-5)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0 
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000111111| delays: 29+-03
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b01 delays: 29+-03
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000011111| delays: 29+-02
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |00000000000000000000000000000000| delays: -
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b01 delays: 29+-02
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 21.4MiB/s
   Read speed: 21.4MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
No boot medium found

--============= Console ================--

litex> mem_test 0x40000000 0x100000
Memtest at 0x40000000 (1.0MiB)...
  Write: 0x40000000-0x40100000 1.0MiB     
   Read: 0x40000000-0x40100000 1.0MiB     
Memtest OK

litex> 
```

I built the design with:
```
litex-boards/litex_boards/targets/digilent_arty.py --toolchain=symbiflow --cpu-type=vexriscv --sys-clk-freq 80e6  --variant a7-35 --build
```

enjoy-digital, 2021-09-09:
@kgugala: Thanks for the test. I haven't run the test myself sorry but was just moving the issue :) 

It seems we were seeing that the read validity windows was larger with Vivado than with Symbiflow which could cause calibration issues. That's something I already saw with Vivado with wrong IO/Internal VRef constraints, so that's why I was suggesting to check that IO constraints and Internal VRef constraints were correctly applied with Symbiflow. If that's the case and if things are working for you, we can probably close this issue.

kgugala, 2021-09-09:
with 100MHz sys_clk I got wider windows:

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2021 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Sep  9 2021 12:46:20
 BIOS CRC passed (20b6f314)

 Migen git sha1: 27dbf03
 LiteX git sha1: e0e9311c

--=============== SoC ==================--
CPU:            VexRiscv @ 100MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128KiB
SRAM:           8KiB
L2:             8KiB
SDRAM:          262144KiB 16-bit @ 800MT/s (CL-6 CWL-5)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0 
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000111111111111100000| delays: 20+-06
  m0, b02: |00000000000000000000000000000011| delays: 31+-01
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b01 delays: 20+-06
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000011111111111100| delays: 24+-06
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |00000000000000000000000000000000| delays: -
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b01 delays: 24+-06
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 27.1MiB/s
   Read speed: 27.3MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
No boot medium found

--============= Console ================--

litex> 
litex> mem_test 0x40000000 0x10000000
Memtest at 0x40000000 (256.0MiB)...
  Write: 0x40000000-0x50000000 256.0MiB   
   Read: 0x40000000-0x50000000 256.0MiB   
Memtest OK

litex>
```

enjoy-digital, 2021-09-09:
OK, this looks good. I didn't realized the frequency has been reduced to 80MHz. So the remaining issue seems to be related to LiteX and already logged in https://github.com/litex-hub/litex-boards/issues/166, https://github.com/litex-hub/litex-boards/issues/263, so we can probably close this issue.

kgugala, 2021-09-09:
ok, I'm closing this one then

closed 2021-09-09 by kgugala