f4pga/prjxray #1833: CI currently broken

mithro, 2022-01-26:
[image]

mithro, 2022-01-26:
@acomodi - Any idea what is going on here?

mithro, 2022-01-26:
```
========================================
Host install packages
----------------------------------------
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package python3.8
E: Couldn't find any package by glob 'python3.8'
E: Couldn't find any package by regex 'python3.8'
E: Unable to locate package python3.8-dev
E: Couldn't find any package by glob 'python3.8-dev'
E: Couldn't find any package by regex 'python3.8-dev'
E: Unable to locate package python3.8-venv
E: Couldn't find any package by glob 'python3.8-venv'
E: Couldn't find any package by regex 'python3.8-venv'

[ID: 6605029] Build finished after 20 secs, exit value: 100
```

umarcor, 2022-05-22:
@mithro I guess this was related to Kokoro, so it can be closed?

closed 2022-08-18 by umarcor