f4pga/prjxray #1834: Possibly a wrong parameter in prjxray/util.py

MidsummerNight, 2022-01-27:
Should `part_mapping` here be `res_mapping` instead? Because `get_part_resources()` does not have a variable called `part_mapping`.
https://github.com/SymbiFlow/prjxray/blob/5349556bc2c230801d6df0cf11bccb9cfd171639/prjxray/util.py#L67
At the moment I don't know what tool calls this function though, so I can't test out yet.

tcal-x, 2022-01-27:
Yes, it looks like a cut-and-paste error from line 48.   @acomodi what do you think?

MidsummerNight, 2022-01-27:
Update:

Yep, it's definitely wrong. I altered `get_part_resources()` to look like this (manually set `part` to an invalid value `"blah"`):
```
def get_part_resources(file_path, part):
    filename = os.path.join(file_path, "resources.yaml")
    assert os.path.isfile(filename), \
        "Mapping file {} does not exists".format(filename)
    with open(filename, 'r') as stream:
        res_mapping = yaml.load(stream, Loader=yaml.FullLoader)
    part="blah"
    res = res_mapping.get(part, None)
    assert res, "Part {} not found in {}".format(part, part_mapping)
    return res
```
Then ran `source settings/spartan7.sh` and got this:
```
Traceback (most recent call last):
  File "/home/test/prjxray/utils/create_environment.py", line 66, in <module>
    main()
  File "/home/test/prjxray/utils/create_environment.py", line 59, in main
    environment = get_environment_variables()
  File "/home/test/prjxray/utils/create_environment.py", line 39, in get_environment_variables
    resources = get_part_resources(res_path, os.environ['XRAY_PART'])
  File "/home/test/prjxray/prjxray/util.py", line 68, in get_part_resources
    assert res, "Part {} not found in {}".format(part, part_mapping)
NameError: name 'part_mapping' is not defined
```
By changing `part_mapping` to `res_mapping`, I got the expected assertion error:
```
Traceback (most recent call last):
  File "/home/test/prjxray/utils/create_environment.py", line 66, in <module>
    main()
  File "/home/test/prjxray/utils/create_environment.py", line 59, in main
    environment = get_environment_variables()
  File "/home/test/prjxray/utils/create_environment.py", line 39, in get_environment_variables
    resources = get_part_resources(res_path, os.environ['XRAY_PART'])
  File "/home/test/prjxray/prjxray/util.py", line 68, in get_part_resources
    assert res, "Part {} not found in {}".format(part, res_mapping)
AssertionError: Part blah not found in {'xc7s50csga324-1': {'pins': {0: 'P14', 1: 'L13', 2: 'C12', 3: 'J5'}}, 'xc7s50csga324-1IL': {'pins': {0: 'P14', 1: 'L13', 2: 'C12', 3: 'J5'}}, 'xc7s50csga324-2': {'pins': {0: 'P14', 1: 'L13', 2: 'C12', 3: 'J5'}}, 'xc7s50fgga484-1': {'pins': {0: 'U20', 1: 'N15', 2: 'C15', 3: 'R3'}}, 'xc7s50fgga484-1IL': {'pins': {0: 'U20', 1: 'N15', 2: 'C15', 3: 'R3'}}, 'xc7s50fgga484-2': {'pins': {0: 'U20', 1: 'N15', 2: 'C15', 3: 'R3'}}, 'xc7s50ftgb196-1': {'pins': {0: 'D14', 1: 'E11', 2: 'B6', 3: 'L5'}}, 'xc7s50ftgb196-1IL': {'pins': {0: 'D14', 1: 'E11', 2: 'B6', 3: 'L5'}}, 'xc7s50ftgb196-2': {'pins': {0: 'D14', 1: 'E11', 2: 'B6', 3: 'L5'}}}
```
I guess nobody reported this before because the `resources.yaml` file generated by `update_resources.py` is always correct, which makes the assertion always true, so the `part_mapping` mistake never shows. On the bright side, this proves that `update_resources.py` is very robust!